mobizt/ESP-Mail-Client #1: send html example fails to compile

gratefulfrog, 2020-12-08:
HI,
Although your previous mail library worked well, I cannot get the example "Send HTML" to compile.

I don't know why the SD library is involved?

Thanks for your work!
Cheers,
Bob

```
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp: In member function 'bool IMAPSession::closeSession()':
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7202:35: error: no matching function for call to 'ESP_Mail_Client::imapLogout(IMAPSession&)'
   if (!MailClient.imapLogout(*this))
                                   ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:1049:6: note: candidate: bool ESP_Mail_Client::imapLogout(IMAPSession*)
 bool ESP_Mail_Client::imapLogout(IMAPSession *imap)
      ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:1049:6: note:   no known conversion for argument 1 from 'IMAPSession' to 'IMAPSession*'
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp: In member function 'bool SMTPSession::closeSession()':
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7910:51: error: no matching function for call to 'ESP_Mail_Client::smtpSendP(SMTPSession&, const char [5], bool)'
   MailClient.smtpSendP(*this, esp_mail_str_7, true);
                                                   ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:3029:8: note: candidate: size_t ESP_Mail_Client::smtpSendP(SMTPSession*, const char*, bool)
 size_t ESP_Mail_Client::smtpSendP(SMTPSession *smtp, PGM_P v, bool newline)
        ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:3029:8: note:   no known conversion for argument 1 from 'SMTPSession' to 'SMTPSession*'
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7912:105: error: no matching function for call to 'ESP_Mail_Client::handleSMTPResponse(SMTPSession&, esp_mail_smtp_status_code, int)'
   ret = MailClient.handleSMTPResponse(*this, esp_mail_smtp_status_code_221, SMTP_STATUS_SEND_BODY_FAILED);
                                                                                                         ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:4716:6: note: candidate: bool ESP_Mail_Client::handleSMTPResponse(SMTPSession*, esp_mail_smtp_status_code, int)
 bool ESP_Mail_Client::handleSMTPResponse(SMTPSession *smtp, esp_mail_smtp_status_code respCode, int errCode)
      ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:4716:6: note:   no known conversion for argument 1 from 'SMTPSession' to 'SMTPSession*'
Multiple libraries were found for "WiFi.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /opt/arduino-1.8.13/libraries/WiFi
 Not used: /home/bob/Arduino/libraries/WiFiNINA
Multiple libraries were found for "WiFiClientSecure.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFiClientSecure
 Not used: /home/bob/Arduino/libraries/_WiFiClientSecure
 Not used: /home/bob/Arduino/libraries/ESP8266WiFi
Multiple libraries were found for "SPIFFS.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPIFFS
 Not used: /home/bob/Arduino/libraries/_SPIFFS
Multiple libraries were found for "SD.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /opt/arduino-1.8.13/libraries/SD
 Not used: /home/bob/Arduino/libraries/SD
exit status 1
Error compiling for board ESP32 Dev Module.
```

mobizt, 2020-12-08:
Do not install third party library for SPIFFS and SD, they are in the core libraries already.

Are you sure you select the ESP8266 or ESP32 not Arduino boards.

I see not relevant libraries SD, WiFiNINA, _WiFiClientSecure, _SPIFFS. Why you install those bad clone and ambiguous name libraries?

Remove all above I mentioned from libraries folder. This causes the error for every libraries  that use SD.h, SPIFFS.h, WiFIClientSecure.h.

You need to clean up your libraries folder that contains many libraries which use the same class name and compiler does not know the right one is used then it select the wrong one and compile then errors.

mobizt, 2020-12-08:
This cannot compare to ESP32 MailClient libraries because it total different in the code internally.

closed 2020-12-08 by mobizt

gratefulfrog, 2020-12-08:
Hi
Thanks for your reply.
Those are not "bad clone" libraries, but all originals.
Anyway, I removed them all and get the same error.

It looks to me like a small type mismatch?

I am using the ESP32 Dev Module board which is the correct one for my hardware.

Thanks again for any thoughts.
Bob

```
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp: In member function 'bool IMAPSession::closeSession()':
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7202:35: error: no matching function for call to 'ESP_Mail_Client::imapLogout(IMAPSession&)'
   if (!MailClient.imapLogout(*this))
                                   ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:1049:6: note: candidate: bool ESP_Mail_Client::imapLogout(IMAPSession*)
 bool ESP_Mail_Client::imapLogout(IMAPSession *imap)
      ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:1049:6: note:   no known conversion for argument 1 from 'IMAPSession' to 'IMAPSession*'
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp: In member function 'bool SMTPSession::closeSession()':
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7910:51: error: no matching function for call to 'ESP_Mail_Client::smtpSendP(SMTPSession&, const char [5], bool)'
   MailClient.smtpSendP(*this, esp_mail_str_7, true);
                                                   ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:3029:8: note: candidate: size_t ESP_Mail_Client::smtpSendP(SMTPSession*, const char*, bool)
 size_t ESP_Mail_Client::smtpSendP(SMTPSession *smtp, PGM_P v, bool newline)
        ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:3029:8: note:   no known conversion for argument 1 from 'SMTPSession' to 'SMTPSession*'
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:7912:105: error: no matching function for call to 'ESP_Mail_Client::handleSMTPResponse(SMTPSession&, esp_mail_smtp_status_code, int)'
   ret = MailClient.handleSMTPResponse(*this, esp_mail_smtp_status_code_221, SMTP_STATUS_SEND_BODY_FAILED);
                                                                                                         ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:4716:6: note: candidate: bool ESP_Mail_Client::handleSMTPResponse(SMTPSession*, esp_mail_smtp_status_code, int)
 bool ESP_Mail_Client::handleSMTPResponse(SMTPSession *smtp, esp_mail_smtp_status_code respCode, int errCode)
      ^
/home/bob/Arduino/libraries/ESP-Mail-Client/src/ESP_Mail_Client.cpp:4716:6: note:   no known conversion for argument 1 from 'SMTPSession' to 'SMTPSession*'
Multiple libraries were found for "SD.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
 Not used: /opt/arduino-1.8.13/libraries/SD
Multiple libraries were found for "WiFi.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
 Not used: /opt/arduino-1.8.13/libraries/WiFi
Multiple libraries were found for "WiFiClientSecure.h"
 Used: /home/bob/.arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFiClientSecure
 Not used: /home/bob/Arduino/libraries/ESP8266WiFi
exit status 1
Error compiling for board ESP32 Dev Module.

```

mobizt, 2020-12-08:
Are you sure, the error tells in that way. They are user installed libraries instead of core libraries.

Remove all libraries in the libraries folder and test it you'll see it can pass the compilation.

mobizt, 2020-12-08:
This library ESP8266WiFi should not in the libraries folder because it is part of core lib.

mobizt, 2020-12-08:
You will see multiple line of **not use**, that's the problem.

mobizt, 2020-12-08:
I recommend you to read all the error message every time you've got the error, you will find answered yourself.

gratefulfrog, 2020-12-08:
Thanks,
The only library in the libraries folder is ESP-Mail-Client from github

The error message does not change and comes from a library file "ESP_Mail_Client.cpp"

If you look at your code, you see that all the errors come from sections protected by 
#if defined(ESP32)
and all are dereference errors on (*this

I corrected them to (this
and the code compiles.
Corrected file is attached
[ESP_Mail_Client.cpp.zip](https://github.com/mobizt/ESP-Mail-Client/files/5659264/ESP_Mail_Client.cpp.zip)

However, then connecting to smtp.gmail.com fails, I tried port 25 465 and 587

```
WiFi connected.
IP address: 
192.168.1.112

Connecting to SMTP server...
> C: connect to SMTP server
> C: host > smtp.gmail.com
> C: port > 465
> C: starting socket
! E: could not get ip from host
> C: cleaning SSL connection
Error, unable to connect to server
> E: unable to connect to server

```

mobizt, 2020-12-08:
The compilation for ESP32 bugs was already fixed in latest update.

About your debugging message, check the line at **#define SMTP_HOST "smtp.gmail.com"**. 

The SSL client does not find the server smtp.gmail.com to get the host ip address. I guess may be spaces around the host name 
e.g.  **" smtp.gmail.com "** which SSL client can't solve it.

gratefulfrog, 2020-12-08:
Hi!
Thanks for that last tip.

I have assigned all the #define values to constants and used them instead in the code.

I can now send gmail with your code! It's a pleasure !

Would you be interested in porting this to SAMD (arduino MKR boards)?  That would be a big help to me and to the entire maker community.  I would be happy to help, although I don't know exactly what to do...

Again, thanks for all your code and help!
Ciao,
Bob

mobizt, 2020-12-08:
Sure if I have time.  The WiFiNiNa and WiFi101 libraries should be modified to be able to upgrade the connection from plain to TLS which is for SRARTTLS for SMTP port 25 and 587 and may use the time.  

This hacks is done in WiFi client for ESP9266 and ESP32 included in this library.

mobizt, 2020-12-08:
One important thing I forgot to mention is the config data takes the pointer of the string instead of copy to internal buffer. 

You should declare the structured data e.g. IMAPSession and SMTPSession in global scope if you want to use it in any function.

gratefulfrog, 2020-12-08:
> Sure if I have time. The WiFiNiNa and WiFi101 libraries should be modified to be able to upgrade the connection from plain to TLS which is for SRARTTLS for SMTP port 25 and 587 and may use the time.
> 
> This hack is done in WiFi client for ESP9266 and ESP32 included in this library.

I don't know enough to do it myself.. But if you do work on it let me know!

Cheers,
Bob

mobizt, 2020-12-09:
Thanks for interesting in that topic and I will inform you the update about this.

gratefulfrog, 2020-12-09:
In the meantime, I made it work on Arduino SAMD MKR WIFI 1010 using the WIFINina library.

Full details are in my repo: https://gitlab.com/gratefulfrog/arduino-wifinina-gmail-smtp

Ciao,
B

mobizt, 2020-12-09:
Good lib

For SMTP, to connect to port 25 and 587, it requires the connection upgradable under the same socket connection from plain (nonencrypted) to secured (.TLS).  
There are many things to do follow the RFC standards both request and response.  The SMTP extension enhancements which clients need to be worked for it. To make a library you should provide it all not just to make it partial work.

gratefulfrog, 2020-12-10:
Thank you again for the help.

I had no intention of making a full library, I was just trying to send
gmail from an Arduino SAMD board.

As you can see, I know very little about this area and simply don't have
the time to learn all about it.

I'm sorry for this partial solution, but it already represents weeks of my
time, sigh.

You are welcome to use my code and make it better or different.

Cheers,
Bob