mobizt/ESP-Mail-Client #3: guru meditation error

tibochina, 2020-12-10:
Hi Mobizt,

Thank you for this very complete library.
I am using your example code : Read_Single_Email.ino
I did not make any changes in the programm, I only changed the Wifi and email address credentials.
The ESP32 successfully connects to the mailbox but this error appears when fetching a message: 

"Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled."

As a side note I also tried your example code in "Search_Emails.ino". It works perfect and there is no "guru meditation error" so it looks like the error is somewhat linked with fetching the email body.  

I imagine this code worked well with your esp32, could it be that my ESP32 has a problem ?
Do I need to make any changes?
thank you very much in advance for your help.

Tibo from china

------------------------------------------------------------------------------------------
Bellow a copy of the terminal to show you where this error appears.

Info of the selected folder
Total Messages: 13
Predicted next UID: 71
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent

Fetching message 1, UID: 68
> C: send smtp command, FETCH
> C: fetch message header
> C: fetch body part header, 1
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dab60  PS      : 0x00060830  A0      : 0x800dc339  A1      : 0x3ffb1860  
A2      : 0x3ffc1858  A3      : 0x3ffc16a4  A4      : 0xfffffd44  A5      : 0x3ffb1b20  
A6      : 0xfffffd44  A7      : 0x1bcb564f  A8      : 0x00000000  A9      : 0x3ffb1830  
A10     : 0x00000000  A11     : 0x3ffd4f28  A12     : 0x3ffb1b94  A13     : 0x3ffb1b94  
A14     : 0x00000000  A15     : 0x00000020  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffed0  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400dab60:0x3ffb1860 0x400dc336:0x3ffb1e40 0x400d29e9:0x3ffb1f20 0x400dfbdf:0x3ffb1fb0 0x40088ddd:0x3ffb1fd0

Rebooting...

mobizt, 2020-12-10:
Hi tibo,

Can you set debug to  **imap.debug(222);**

And post the error again.

tibochina, 2020-12-10:
thank you for your help!

I changed imap.debug(222);
here is the error message:
------------------------------------------------------------------------

Connecting to IMAP server...
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
> C: setting hostname for TLS session
> C: performing the SSL handshaking
> C: verifying peer X.509 certificate
> C: certificate verified
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
IMAP server connected
> C: server connected

Get the capability...
> C: get the capability
* OK The Microsoft Exchange IMAP4 service is ready. [SABLADIAUABSADAAMwAwADIAQwBBADAAMAAwADkALgBhAHAAYwBwAHIAZAAwADMALgBwAHIAbwBkAC4AbwB1AHQAbABvAG8AawAuAGMAbwBtAA==]
* CAPABILITY IMAP4 IMAP4rev1 AUTH=PLAIN AUTH=XOAUTH2 SASL-IR UIDPLUS MOVE ID UNSELECT CHILDREN IDLE NAMESPACE LITERAL+
$ OK CAPABILITY completed.

Logging in...
> C: send imap command, LOGIN
$ OK LOGIN completed.

Reading the list of mailboxes...
> C: send imap command, LIST
$ BAD User is authenticated but not connected.
Error, fail to list the mailboxes
> E: fail to list the mailboxes
> C: cleaning SSL connection

Info of the selected folder
Total Messages: 0
Predicted next UID: 0
Connecting to IMAP server...
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
> C: setting hostname for TLS session
> C: performing the SSL handshaking
> C: verifying peer X.509 certificate
> C: certificate verified
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
IMAP server connected
> C: server connected

Get the capability...
> C: get the capability
* OK The Microsoft Exchange IMAP4 service is ready. [SABLADIAUABSADAAMwAwADIAQwBBADAAMAAyADQALgBhAHAAYwBwAHIAZAAwADMALgBwAHIAbwBkAC4AbwB1AHQAbABvAG8AawAuAGMAbwBtAA==]
* CAPABILITY IMAP4 IMAP4rev1 AUTH=PLAIN AUTH=XOAUTH2 SASL-IR UIDPLUS MOVE ID UNSELECT CHILDREN IDLE NAMESPACE LITERAL+
$ OK CAPABILITY completed.

Logging in...
> C: send imap command, LOGIN
$ OK LOGIN completed.

Selecting the INBOX folder...
> C: open the mailbox folder
$ BAD User is authenticated but not connected.
Error, fail to open the mailbox
> E: fail to open the mailbox
> C: cleaning SSL connection
Error, fail to open the mailbox
> E: fail to open the mailbox

mobizt, 2020-12-10:
Ok now Microsoft is temporary hold the request (can login but no command accepted), please wait 10 - 20 minutes.

This often happen to me too when device crash and tying login to many attempts.

After everything resumed, when device crashed, please remove usb cable suddenly. I'm apologize for the error.

tibochina, 2020-12-10:
okay, no problem, so I did it again and this time the message is different . 

Connecting to IMAP server...
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
> C: [...]
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
IMAP server connected
> C: server connected

Get the capability...
> C: get the capability
* OK The Microsoft Exchange IMAP4 service is ready. [SABLADIAUABSADAAMgBDAEEAMAAyADEANQAuAGEAcABjAHAAcgBkADAAMgAuAHAAcgBvAGQALgBvAHUAdABsAG8AbwBrAC4AYwBvAG0A]
* CAPABILITY IMAP4 IMAP4rev1 AUTH=PLAIN AUTH=XOAUTH2 SASL-IR UIDPLUS MOVE ID UNSELECT CHILDREN IDLE NAMESPACE LITERAL+
$ OK CAPABILITY completed.

Logging in...
> C: send imap command, LOGIN
$ OK LOGIN completed.

Reading the list of mailboxes...
> C: send imap command, LIST
* LIST (\HasNoChildren) "/" Archive
* LIST (\HasNoChildren \Trash) "/" Deleted
* LIST (\HasNoChildren \Drafts) "/" Drafts
* LIST (\Marked \HasNoChildren) "/" Inbox
* LIST (\HasNoChildren \Junk) "/" Junk
* LIST (\HasNoChildren) "/" Notes
* LIST (\HasNoChildren) "/" Outbox
* LIST (\HasNoChildren \Sent) "/" Sent
$ OK LIST completed.

Available folders: Archive, Deleted, Drafts, Inbox, Junk, Notes, Outbox, Sent

Selecting the INBOX folder...
> C: open the mailbox folder
* 13 EXISTS
* 13 RECENT
* FLAGS (\Seen \Answered \Flagged \Deleted \Draft $MDNSent)
* OK [PERMANENTFLAGS ()] Permanent flags
* OK [UNSEEN 4] Is the first unseen message
* OK [UIDVALIDITY 14] UIDVALIDITY value
* OK [UIDNEXT 71] The next unique identifier value
$ OK [READ-ONLY] EXAMINE completed.

Info of the selected folder
Total Messages: 13
Predicted next UID: 71
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent

Fetching message 1, UID: 68
> C: send smtp command, FETCH
> C: fetch message header
* 13 FETCH (BODY[HEADER.FIELDS (SUBJECT FROM TO DATE CC Message-ID Accept-Language content-type Content-transfer-encoding Content-Language)] {392}
From: T B <enchine@hotmail.fr>
To: "aotugoutong@outlook.com" <aotugoutong@outlook.com>
Subject: Test new
Date: Thu, 10 Dec 2020 07:17:01 +0000
Message-ID:
 <VE1PR02MB5680DC6FF533F868331F21B8D8CB0@VE1PR02MB5680.eurprd02.prod.outlook.com>
Accept-Language: fr-FR, en-US
Content-Language: en-GB
Content-Type: text/plain; charset="us-ascii"
Content-Transfer-Encoding: quoted-printable
 UID 68)
$ OK FETCH completed.
> C: fetch body part header, 1
* 13 FETCH (BODY[HEADER.FIELDS (content-type Content-transfer-encoding)] {93}
Content-Type: text/plain; charset="us-ascii"
Content-Transfer-Encoding: quoted-printable
 UID 68)
$ OK FETCH completed.
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dab60  PS      : 0x00060830  A0      : 0x800dc339  A1      : 0x3ffb1860  
A2      : 0x3ffc1858  A3      : 0x3ffc16a4  A4      : 0xfffffd44  A5      : 0x3ffb1b20  
A6      : 0xfffffd44  A7      : 0x1bcb564f  A8      : 0x00000000  A9      : 0x3ffb1830  
A10     : 0x00000000  A11     : 0x3ffd7440  A12     : 0x3ffb1b94  A13     : 0x3ffb1b94  
A14     : 0x00000000  A15     : 0x00000020  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xfffffed0  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400dab60:0x3ffb1860 0x400dc336:0x3ffb1e40 0x400d29e9:0x3ffb1f20 0x400dfbdf:0x3ffb1fb0 0x40088ddd:0x3ffb1fd0

Rebooting...

mobizt, 2020-12-10:
Is this happen to message uid 68 only or other message too? 
Can you try other message uid?

mobizt, 2020-12-10:
Please change debug to  **imap.debug(333);**

tibochina, 2020-12-10:
Yes I just tried with another email, uid 66 and I get the same error.

I got blocked again by microsoft, I'll try again in a few minutes and show you the result with imap.debug(333);

mobizt, 2020-12-10:
With message subject **Test new**.

Can you send me the full message source via Email (don't post it here). please remove any info about email address and  something  you want.

You can view the message source by open email on the web browser.

[image]

mobizt, 2020-12-10:
I jus wonder the message structure that cause the error.

With [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/Read_Single_Email_Loop/Read_Single_Email_Loop.ino), I can fetch many hundred messages with no errors.

mobizt, 2020-12-10:
Update, you are not need to send me a message source as I requested.

I also found the bugs in ESP32 and trying to fix it.

tibochina, 2020-12-10:
oh okay, I was about to send it to you...
man you are fast! can't keep up with you :)
Thank you again Mobizt for your help !

mobizt, 2020-12-10:
This error only happen in ESP32 and I'm checking now.

mobizt, 2020-12-10:
Ok the issue was fixed now. The library updated to v1.0.5, please try it. 
For platformio, the update takes time.

closed 2020-12-10 by mobizt

tibochina, 2020-12-11:
Mobizt you're the best!
It works perfectly now :)
Thanks for your hard work