mobizt/ESP-Mail-Client #4: Crash "CORRUPT HEAP: Bad tail..." when using an Outlook account as sender

blackhack, 2020-12-15:
Grettings,
I'm getting this crash when using Outlook as sender, this happens with both ports 25 and 587. When the sender account is from Gmail it works flawless.

I can reproduce it using the "Send_Text" example, just filling the appropriate fields(#######) and no touching anything else.

By the way I'm with master in both ESP-Mail-Client and arduino-esp32
https://github.com/mobizt/ESP-Mail-Client/commit/4d3f5adecd9dfb3719538c196c82dcbf8f3cd828
https://github.com/espressif/arduino-esp32/commit/b05bdf690418efd28c8468b5cb8345d95ba77e9f

This is the serial log using smtp.debug(333) and debug level verbose for esp32.
````
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10920
ho 0 tail 12 room 4
load:0x40080400,len:6388
entry 0x400806b4

Connecting to AP[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 2 - STA_START
.[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
.....[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:418] _eventCallback(): STA IP: 192.168.1.97, MASK: 255.255.255.0, GW: 192.168.1.254

WiFi connected.
IP address: 
192.168.1.97

Connecting to SMTP server...
> C: connect to SMTP server
> C: host > smtp.office365.com
> C: port > 587
> C: starting socket
[V][esp_mail_ssl_client32.cpp:84] start_socket(): Free internal heap before TLS 277224
[V][esp_mail_ssl_client32.cpp:86] start_socket(): Starting socket
> C: connecting to Server
> C: smtp server connected

SMTP server connected, wait for greeting...
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
< S: 220 BN8PR03CA0016.outlook.office365.com Microsoft ESMTP MAIL Service ready at Tue, 15 Dec 2020 03:03:08 +0000

Sending greeting response...
> C: send smtp command, HELO
EHLO mydomain.net
250-BN8PR03CA0016.outlook.office365.com Hello [181.128.redacted]
< S: 250-BN8PR03CA0016.outlook.office365.com Hello [181.128.redacted]
250-SIZE 157286400
< S: 250-SIZE 157286400
250-PIPELINING
< S: 250-PIPELINING
250-DSN
< S: 250-DSN
250-ENHANCEDSTATUSCODES
< S: 250-ENHANCEDSTATUSCODES
250-STARTTLS
< S: 250-STARTTLS
250-8BITMIME
< S: 250-8BITMIME
250-BINARYMIME
< S: 250-BINARYMIME
250-CHUNKING
< S: 250-CHUNKING
250 SMTPUTF8
< S: 250 SMTPUTF8

Send command, STARTTLS
> C: send STARTTLS command
STARTTLS
220 2.0.0 SMTP server ready
< S: 220 2.0.0 SMTP server ready
> C: performing the SSL handshaking
> C: seeding the random number generator
[V][esp_mail_ssl_client32.cpp:147] start_ssl_client(): Seeding the random number generator
> C: setting up the SSL/TLS structure
[V][esp_mail_ssl_client32.cpp:171] start_ssl_client(): Setting up the SSL/TLS structure...
[I][esp_mail_ssl_client32.cpp:282] start_ssl_client(): WARNING: Use certificates for a more secure communication!
[V][esp_mail_ssl_client32.cpp:329] start_ssl_client(): Setting hostname for TLS session...
> C: performing the SSL handshaking
[V][esp_mail_ssl_client32.cpp:370] start_ssl_client(): Performing the SSL/TLS handshake...
> C: verifying peer X.509 certificate
[V][esp_mail_ssl_client32.cpp:410] start_ssl_client(): Verifying peer X.509 certificate...
> C: certificate verified
[V][esp_mail_ssl_client32.cpp:426] start_ssl_client(): Certificate verified.
[V][esp_mail_ssl_client32.cpp:444] start_ssl_client(): Free internal heap after TLS 234068
Sending greeting response...
> C: send smtp command, HELO
EHLO mydomain.net
[V][esp_mail_ssl_client32.cpp:497] send_ssl_data(): Writing HTTP request...
[V][esp_mail_ssl_client32.cpp:497] send_ssl_data(): Writing HTTP request...
250-BN8PR03CA0016.outlook.office365.com Hello [181.128.84.123]
< S: 250-BN8PR03CA0016.outlook.office365.com Hello [181.128.84.123]
250-SIZE 157286400
< S: 250-SIZE 157286400
250-PIPELINING
< S: 250-PIPELINING
250-DSN
< S: 250-DSN
250-ENHANCEDSTATUSCODES
< S: 250-ENHANCEDSTATUSCODES
250-AUTH LOGIN XOAUTH2
< S: 250-AUTH LOGIN XOAUTH2
250-8BITMIME
< S: 250-8BITMIME
250-BINARYMIME
< S: 250-BINARYMIME
250-CHUNKING
< S: 250-CHUNKING
250 SMTPUTF8
< S: 250 SMTPUTF8

Logging in...
> C: send smtp command, AUTH LOGIN
AUTH LOGIN
[V][esp_mail_ssl_client32.cpp:497] send_ssl_data(): Writing HTTP request...
[V][esp_mail_ssl_client32.cpp:497] send_ssl_data(): Writing HTTP request...
334 VXNlcm5hbWU6
CORRUPT HEAP: Bad tail at 0x3ffb31e1. Expected 0xbaad5678 got 0xbaad5600
abort() was called at PC 0x400820a5 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40088490:0x3ffb17e0 0x4008870d:0x3ffb1800 0x400820a5:0x3ffb1820 0x400821d1:0x3ffb1850 0x4014dbb7:0x3ffb1870 0x40149499:0x3ffb1b30 0x401493a1:0x3ffb1b80 0x4008cdc5:0x3ffb1bb0 0x400815c2:0x3ffb1bd0 0x40081fe9:0x3ffb1bf0 0x4000bec7:0x3ffb1c10 0x4016ed6d:0x3ffb1c30 0x4016f301:0x3ffb1c50 0x400d426e:0x3ffb1c70 0x400d5dee:0x3ffb1c90 0x400d6a2d:0x3ffb1d30 0x400d8522:0x3ffb1de0 0x400d23fd:0x3ffb1e10 0x400e02c6:0x3ffb1fb0 0x4008970e:0x3ffb1fd0

Rebooting...
````
This is the decoded crashlog
````
Decoding stack results
0x40088490: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x4008870d: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x400820a5: lock_acquire_generic at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 143
0x400821d1: _lock_acquire_recursive at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/locks.c line 171
0x4014dbb7: _vfiprintf_r at ../../../.././newlib/libc/stdio/vfprintf.c line 860
0x40149499: fiprintf at ../../../.././newlib/libc/stdio/fiprintf.c line 50
0x401493a1: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 59
0x4008cdc5: multi_heap_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c line 218
0x400815c2: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 268
0x40081fe9: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x4016ed6d: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x4016f301: operator delete[](void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_opv.cc line 32
0x400d426e: ESP_Mail_Client::delS(char*) at D:\Arduino\libraries\ESP-Mail-Client\src\ESP_Mail_Client.cpp line 5107
0x400d5dee: ESP_Mail_Client::handleSMTPResponse(SMTPSession*, esp_mail_smtp_status_code, int) at D:\Arduino\libraries\ESP-Mail-Client\src\ESP_Mail_Client.cpp line 4840
0x400d6a2d: ESP_Mail_Client::smtpAuth(SMTPSession*) at D:\Arduino\libraries\ESP-Mail-Client\src\ESP_Mail_Client.cpp line 1681
0x400d8522: SMTPSession::connect(esp_mail_session_config_t*) at D:\Arduino\libraries\ESP-Mail-Client\src\ESP_Mail_Client.cpp line 7798
0x400d23fd: setup() at C:\Users\david\AppData\Local\Temp\arduino_modified_sketch_959128/Send_Text.ino line 139
0x400e02c6: loopTask(void*) at D:\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 18
0x4008970e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
````

Thank You.

mobizt, 2020-12-15:
Thanks for the issue. It is likely a bug about the redundant of memory release issue. 

I will check and update.

mobizt, 2020-12-15:
The issue was fixed and update. Please update the library to version 1.0.6.

The library was updated in PlatformIO library registry already. For Arduino Board Manager, it still waiting  after open the request 12 days ago.

closed 2020-12-15 by mobizt