mobizt/ESP-Mail-Client #5: Unwanted restarts with ESP-Mail-Client library

rolandwidmer, 2020-12-18:
I wrote a small program, that checks frequently an IMAP server for error messages of my NAS and switches on an LED in case of an error using the ESP-Mail_client library 1.06. The program works fine, however, the ESP8266 restarts every 30 to 60 minutes unwantedly. Using the ESP.getFreeHeap()-function I found out, that with every connection to the IMAP server (every roung od loop()) to read mails the free heap shrinks a little bit (0.5 to 1 kB). Starting from about 38 kB it goes down gradually to about 3 kB, then the restart occures. When I removed all "extras" from the program, the restarts occured a bit rarer, but nevertheless occured. So there must be an error with memory management either in my program or in the library. 

This is the code:

#include <Arduino.h>
#if defined(ESP32)
  #include <WiFi.h>
#elif defined(ESP8266)
  #include <ESP8266WiFi.h>
#endif
#include <ESP_Mail_Client.h>

#define WIFI_SSID "XXX"
#define WIFI_PASSWORD "XXX"
#define IMAP_HOST "XXX"
#define IMAP_PORT 993
#define AUTHOR_EMAIL "XXX"
#define AUTHOR_PASSWORD "XXX"

//***************************************************************************

int fehler = 0;
int altfehler = 0;

// Setup the configuration for searching or fetching operation and its result
IMAP_Config config;
ESP_Mail_Session session;
IMAPSession imap;

//*************************INTERRUPT*********************************************
// Interrupt when button is pressed
void IRAM_ATTR buttonPressed() {
      fehler = 0;
      digitalWrite(14, LOW);
      }

//**************************SETUP************************************************
void setup() {
    pinMode(14, OUTPUT);
    digitalWrite(14, LOW);
    pinMode(12, INPUT_PULLUP);
    
    attachInterrupt(digitalPinToInterrupt(12), buttonPressed, RISING);    
    
    Serial.begin(115200);
    Serial.println();
    Serial.print("Connecting to AP");

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
        Serial.print(".");
        delay(200);
        }         
    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    config.fetch.uid = "";
    config.search.criteria = "UID SEARCH NOT SEEN";
    config.search.unseen_msg = true;
    config.storage.saved_path = "/email_data";
    config.storage.type = esp_mail_file_storage_type_none;
    config.download.header = true;
    config.download.text = true;
    config.download.html = true;
    config.download.attachment = false;
    config.download.inlineImg = false;
    config.enable.html = true;
    config.enable.text = true;
[... 220 lines cut ...]
123400.526 - 
123400.526 - load 0x4010f000, len 3584, room 16 
123400.526 - tail 0
123400.526 - chksum 0xb0
123400.526 - csum 0xb0
123400.526 - v2843a5ac
123400.526 - ~ld
123400.619 - 
123400.619 - Connecting to AP...
123401.226 - WiFi connected.
123401.226 - IP address 
123401.226 - 192.168.179.89
123402.204 - Free Heap before fetching mails 32896
123402.251 - Fetching mails
123432.295 - Free Heap after fetching mails 38112
123432.295 - 10s delay
123443.281 - Free Heap before fetching mails 31976
123443.281 - Fetching mails
123513.338 - Free Heap after fetching mails 37288
123513.338 - 10s delay
123524.327 - Free Heap before fetching mails 31152
123524.327 - Fetching mails
123554.397 - Free Heap after fetching mails 36464

Does anybody have an idea, what the problem might be?

mobizt, 2020-12-18:
Do not modify what you unsure what it is.

mobizt, 2020-12-18:
It is better to clean your code that ready for me to test it easier.  I will check what's the causes of memory leaks.

mobizt, 2020-12-18:
The error was fixed, please update the library.

closed 2020-12-18 by mobizt