mobizt/ESP-Mail-Client #8: ESP32 Reboot

xfdr0805, 2020-12-19:
Connecting to SMTP server...
> C: connect to SMTP server
> C: host > smtp.126.com
> C: port > 465
> C: smtp server connected

SMTP server connected, wait for greeting...
< S: 220 126.com Anti-spam GT for Coremail System (126com[20140526])

Sending greeting response...
> C: send smtp command, HELO
< S: 250-mail
< S: 250-PIPELINING
< S: 250-AUTH LOGIN PLAIN 
< S: 250-AUTH=LOGIN PLAIN
< S: 250-coremail 1Uxr2xKj7kG0xkI17xGrU7I0s8FY2U3Uj8Cz28x1UUUUU7Ic2I0Y2Ur9ozT7UCa0xDrUUUUj
< S: 250-STARTTLS
< S: 250 8BITMIME

Send command, STARTTLS
> C: send STARTTLS command
< S: 454 Command not permitted when TLS active
Error, SMTP server greeting failed
> E: SMTP server greeting failed
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x4015bbd7 on core 1

Backtrace: 0x4008cf08:0x3ffd66c0 0x4008d139:0x3ffd66e0 0x4015bbd7:0x3ffd6700 0x40084d9f:0x3ffd6730 0x400851ad:0x3ffd6750 0x4000bec7:0x3ffd6770 0x40193889:0x3ffd6790 0x4019da56:0x3ffd67b0 0x400d6137:0x3ffd67d0 0x400d664a:0x3ffd67f0 0x400e0481:0x3ffd6a70 0x400e0509:0x3ffd6ce0 0x400eb565:0x3ffd6d00 0x400e8f49:0x3ffd6d40 0x400e8fd9:0x3ffd6d80 0x401af8e1:0x3ffd6da0 0x401af971:0x3ffd6de0 0x401b0016:0x3ffd6e00 0x4008961d:0x3ffd6e30

Rebooting...

mobizt, 2020-12-19:
Did you know what STRATTLS? Why enable it for port 465?

closed 2020-12-19 by mobizt

mobizt, 2020-12-19:
Ok there is some strange response with smtp server that return improper response  250-STARTTLS with the port 465 which client connected which it should not return this to present it supports TLS on port 465 and when the client send the starttls command, they can't accept.

mobizt, 2020-12-19:
Change the smtp server that comply to rfc standard.

mobizt, 2020-12-19:
Ok the server promotes the starttls capability even the current connection is already secured and library tries to send the starttls command based on their capabilities and that is the issue. I will fix this in the next update.

mobizt, 2020-12-19:
The issue was fixed, please update.

xfdr0805, 2020-12-21:
###Thanks a lot ! but the esp32 still rebooting

Connecting to SMTP server...
> C: ESP Mail Client v1.0.8
> C: connect to SMTP server
> C: host > smtp.126.com
> C: port > 465
> C: smtp server connected

SMTP server connected, wait for greeting...
< S: 220 126.com Anti-spam GT for Coremail System (126com[20140526])

Sending greeting response...
> C: send smtp command, HELO
< S: 250-mail
< S: 250-PIPELINING
< S: 250-AUTH LOGIN PLAIN 
< S: 250-AUTH=LOGIN PLAIN
< S: 250-coremail 1Uxr2xKj7kG0xkI17xGrU7I0s8FY2U3Uj8Cz28x1UUUUU7Ic2I0Y2UrQdResUCa0xDrUUUUj
< S: 250-STARTTLS
< S: 250 8BITMIME

Logging in...
> C: send smtp command, AUTH PLAIN
> C: xfdr0805@126.com
> C: xfdr0805@126.com ********
< S: 235 Authentication successful
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014fd  PS      : 0x00060230  A0      : 0x800fb801  A1      : 0x3ffd6690  
A2      : 0x918fe5a8  A3      : 0x918fe5a4  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x00000001  
A10     : 0x3ffdf9c4  A11     : 0x3ffdd8bc  A12     : 0x00000022  A13     : 0x0000ff00  
A14     : 0x0000000a  A15     : 0x3ffd6a40  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x918fe5a8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

Backtrace: 0x400014fd:0x3ffd6690 0x400fb7fe:0x3ffd66a0 0x400fefac:0x3ffd66d0 0x400ff019:0x3ffd6710 0x401011ed:0x3ffd6740 0x40101b88:0x3ffd67c0 0x400d62c2:0x3ffd67e0 0x400df9fd:0x3ffd6a60 0x400dfa85:0x3ffd6cd0 0x400ea8ed:0x3ffd6cf0 0x400e82d1:0x3ffd6d30 0x400e8361:0x3ffd6d70 0x401aecad:0x3ffd6d90 0x401aed3d:0x3ffd6dd0 0x401af3e2:0x3ffd6df0 0x4008961d:0x3ffd6e20

Rebooting...

mobizt, 2020-12-21:
Please install and use [this extension](https://github.com/me-no-dev/EspExceptionDecoder) in Arduino to debug the stack trace and post the debug result here.

mobizt, 2020-12-21:
This is possible unstable power supply issue as I also test with this Chinese SMTP server and did not found any bug.
Please change your usb cable and try to connect large capacitor (> 220 uF) between 3v3 and gnd on the esp32 board.

mobizt, 2020-12-21:
This is the part of my test

Connecting to SMTP server...
> C: ESP Mail Client v1.0.8
> C: connect to SMTP server
> C: host > smtp.163.com
> C: port > 465
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
> C: setting hostname for TLS session
> C: performing the SSL handshaking
> C: verifying peer X.509 certificate
> C: certificate verified
> C: smtp server connected

SMTP server connected, wait for greeting...
[E][WiFiClient.cpp:288] setSocketOption(): 1006 : 9
< S: 220 163.com Anti-spam GT for Coremail System (163com[20141201])

Sending greeting response...
> C: send smtp command, HELO
< S: 250-mail
< S: 250-PIPELINING
< S: 250-AUTH LOGIN PLAIN
< S: 250-AUTH=LOGIN PLAIN
< S: 250-coremail 1Uxr2xKj7kG0xkI17xGrU7I0s8FY2U3Uj8Cz28x1UUUUU7Ic2I0Y2UF5Nt75UCa0xDrUUUUj
< S: 250-STARTTLS
< S: 250 8BITMIME

Logging in...
> C: send smtp command, AUTH PLAIN
> C: _______@163.com
> C: _______@163.com ****************
< S: 235 Authentication successful

Sending Email...
> C: send Email

Sending message header...
> C: send message header
< S: 250 Mail OK
< S: 250 Mail OK

Sending message body...
> C: send message body
< S: 354 End data with <CR><LF>.<CR><LF>

Finishing the message sending...
> C: Finish the message sending

xfdr0805, 2020-12-23:
> Please install and use [this extension](https://github.com/me-no-dev/EspExceptionDecoder) in Arduino to debug the stack trace and post the debug result here.

this is Decoding stack results

`0x4008cf08: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008d139: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4015bd47: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40084d9f: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x400851ad: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x401939f9: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x4019dbc6: std::__cxx11::basic_string  , std::allocator  >::_M_dispose() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/ext/new_allocator.h line 110
0x400d6143: SMTP_Message::~SMTP_Message() at c:\users\xfdr0\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/basic_string.h line 544
0x400d655d: send_email(String, String) at src\email.cpp line 135
0x400d832e: loop() at src\main.cpp line 430
0x40108339: loopTask(void*) at C:\users\xfdr0\.platformio\packages\framework-arduinoespressif32\cores\esp32\main.cpp line 19
0x4008961d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

mobizt, 2020-12-23:
Look like something wrong in your code.

Post your code and remove private data before post.

xfdr0805, 2020-12-23:
> Look like something wrong in your code.
> 
> Post your code and remove private data before post.

this is my code:
[email.zip](https://github.com/mobizt/ESP-Mail-Client/files/5735775/email.zip)

mobizt, 2020-12-23:
Don't zip it. should be in text (.txt)

mobizt, 2020-12-23:
Please post the full code included setup and loop.

The error is about the variable that pass to the function was disposed due to the its scope.

You need to define the config, message that available in the usage scope of sendMail and calback functions because of these vars passed as the pointer to these function and no data copied internally.

You may define these in the global scope (on the top of sketch).
```cpp
#include <Arduino.h>

ESP_Mail_Session session;
SMTP_Message message;
```

xfdr0805, 2020-12-23:
> Please post the full code included setup and loop.
> 
> The error is about the variable that pass to the function was disposed due to its scope.
> 
> You need to define the session and message variables that available in the usage scope of sendMail and calback functions because of these vars passed as the pointer to these functions and no data copied internally.
> 
> You may define these in the global scope (on the top of sketch).
> 
> ```c++
> #include <Arduino.h>
> 
> ESP_Mail_Session session;
> SMTP_Message message;
> ```
> 
> Pkease be careful. If you see the functions that use the pointer as parameters, you need to keep that variable alive for all usages.

thank you very much,but still failed   I will check it again carefully
[image]

mobizt, 2020-12-23:
Because you use the vars in other cpp file which can't be access

You need to define these vars in header file e.g. **email.h** like this

Content of  **email.h**

```cpp
#ifndef EMAIL_H
#define EMAIL_H

#include <Arduino.h>
#include <ESP_Mail_Client.h>

SMTPSession smtp;
ESP_Mail_Session session;
SMTP_Message message;

#endif
```

On the top of main.cpp and email.cpp (and any other cpp files) that you want to call the mail function, include this **email.h** file on the top like this. 

```cpp
//assume that email.h is in the same location of your cpp files.
#include "email.h"
```

Now you can use variables i.e. **session**, **message** and **smtp** in all your cpp files.