mobizt/ESP-Mail-Client #11: Problem with SPIFFS

tld70, 2021-01-02:
Hello,
if I try to compile this:
#include <WebServer.h>
#include <ArduinoOTA.h>
#include <SPIFFS.h>
#include <ESP_Mail_Client.h>
//#include <ESP32_MailClient.h>
#include <detail/RequestHandlersImpl.h>
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
I get much alarms.
*********************************************************************************************************

If I try this:
#include <WebServer.h>
#include <ArduinoOTA.h>
#include <SPIFFS.h>
//#include <ESP_Mail_Client.h>
#include <ESP32_MailClient.h>
#include <detail/RequestHandlersImpl.h>
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
this will compile.
What I do wrong?
I use a DOIT ESP32 DEVKIT V1

mobizt, 2021-01-02:
Post the error from serial monitor.
Don't just ask without info.

tld70, 2021-01-02:
The serial monitor does not run in this moment.
I just want compile this view lines and the compiler throw this errors:
Arduino: 1.8.13 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 115200, None"

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:146:0,

                 from sketch\sketch_jan02b.ino.cpp:1:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h: In member function 'virtual bool StaticRequestHandler::handle(WebServer&, HTTPMethod, String)':

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:145:44: error: reference to 'gz' is ambiguous

         if (!path.endsWith(FPSTR(mimeTable[gz].endsWith)) && !_fs.exists(path))  {

                                            ^

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/WString.h:39:76: note: in definition of macro 'FPSTR'

 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))

                                                                            ^

In file included from C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_Client.h:61:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:4:

C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/extras/MIMEInfo.h:27:3: note: candidates are: FileExtension gz

   gz,

   ^

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:5:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:6:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/mimetable.h:30:3: note:                 mime::type gz

   gz,

   ^

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:146:0,

                 from sketch\sketch_jan02b.ino.cpp:1:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:146:56: error: reference to 'gz' is ambiguous

             String pathWithGz = path + FPSTR(mimeTable[gz].endsWith);

                                                        ^

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/WString.h:39:76: note: in definition of macro 'FPSTR'

 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))

                                                                            ^

In file included from C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_Client.h:61:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:4:

C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/extras/MIMEInfo.h:27:3: note: candidates are: FileExtension gz

   gz,

   ^

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:5:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:6:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/mimetable.h:30:3: note:                 mime::type gz

   gz,

   ^

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:146:0,

                 from sketch\sketch_jan02b.ino.cpp:1:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:148:41: error: reference to 'gz' is ambiguous

                 path += FPSTR(mimeTable[gz].endsWith);

                                         ^

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\cores\esp32/WString.h:39:76: note: in definition of macro 'FPSTR'

 #define FPSTR(pstr_pointer) (reinterpret_cast<const __FlashStringHelper *>(pstr_pointer))

                                                                            ^

In file included from C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_Client.h:61:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:4:

C:\Users\Thomas\Documents\Arduino\libraries\ESP-Mail-Client-master\src/extras/MIMEInfo.h:27:3: note: candidates are: FileExtension gz

   gz,

   ^

In file included from C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/RequestHandlersImpl.h:5:0,

                 from C:\Users\Thomas\Documents\Arduino\sketch_jan02b\sketch_jan02b.ino:6:

C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WebServer\src/detail/mimetable.h:30:3: note:                 mime::type gz

   gz,

   ^

Mehrere Bibliotheken wurden für "SD.h" gefunden

 Benutzt: C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD

 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\SD

Mehrere Bibliotheken wurden für "WiFi.h" gefunden

 Benutzt: C:\Users\Thomas\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

 Nicht benutzt: C:\Program Files (x86)\Arduino\libraries\WiFi

exit status 1

Fehler beim Kompilieren für das Board DOIT ESP32 DEVKIT V1.

Dieser Bericht wäre detaillierter, wenn die Option
"Ausführliche Ausgabe während der Kompilierung"
in Datei -> Voreinstellungen aktiviert wäre.

mobizt, 2021-01-02:
Thanks for info.

The error is due to ambiguous enumeration in Webserver lib and ESP Mail Client lib.

Now I fixed the issue and library just updated. Please update the library to v 1.0.10.

closed 2021-01-02 by mobizt