mobizt/ESP-Mail-Client #12: Destroying const char * variables after smtp.connect(&session)

saman-taghipour, 2021-01-06:
Hi,
I have several const char * variables in my code. These variables work well in the other part of code. When I want to send email, after this line: if (!smtp.connect(&session))
or line 148 of here: https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/Send_HTML/Send_HTML.ino
const char * variables value are corrupted

mobizt, 2021-01-07:
All function parameters of this library are pointer to the variable which you need to keep  that variables in the usage scope of function.

You may declare these variable out of any code block at the top of your sketch and use it everywhere without any problem.

closed 2021-01-07 by mobizt

saman-taghipour, 2021-01-07:
I declared const char * variables at the top of my sketch (global variable)
and I can access these variables in any other functions.
In this function that I want to send email, at first I can access these variables too. but after a line, any const chat * variables are corrupted
this line:

[here I can access const char * variables]
/* Connect to server with the session config */
    if (!smtp.connect(&session))
      return;
[here I can not access const char * variables]

mobizt, 2021-01-07:
No one known your error without code. 
It's better to learn more about pointer in C or C++.

saman-taghipour, 2021-01-11:
/**
 * This example will send the Email in
 * the html version.
 * 
 * 
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: suwatchai@outlook.com
 * 
 * Github: https://github.com/mobizt/ESP-Mail-Client
 * 
 * Copyright (c) 2020 mobizt
 *
*/

//To use send Email for Gmail to port 465 (SSL), less secure app option should be enabled. https://myaccount.google.com/lesssecureapps?pli=1
#include <LittleFS.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <ESP_Mail_Client.h>

WiFiClient client;

#define WIFI_SSID "mySSID"
#define WIFI_PASSWORD "myPresharedkey"

/** The smtp host name e.g. smtp.gmail.com for GMail or smtp.office365.com for Outlook or smtp.mail.yahoo.com
 * For yahoo mail, log in to your yahoo mail in web browser and generate app password by go to
 * https://login.yahoo.com/account/security/app-passwords/add/confirm?src=noSrc
 * and use the app password as password with your yahoo mail account to login.
 * The google app password signin is also available https://support.google.com/mail/answer/185833?hl=en
*/
#define SMTP_HOST "smtp.gmail.com"

/** The smtp port e.g. 
 * 25  or esp_mail_smtp_port_25
 * 465 or esp_mail_smtp_port_465
 * 587 or esp_mail_smtp_port_587
*/
#define SMTP_PORT 465

/* The log in credentials */
#define AUTHOR_EMAIL "userName@gmail.com"
#define AUTHOR_PASSWORD "appPassword"

/* The SMTP Session object used for Email sending */
SMTPSession smtp;

String publicIP;

/* Callback function to get the Email sending status */
void smtpCallback(SMTP_Status status);

const char * var;

void setup()
{
  Serial.begin(9600);
  LittleFS.begin();
  Serial.println();

  Serial.print("Connecting to AP");

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(200);
  }

[... 118 lines cut ...]
      Serial.printf("Recipient: %s\n", result.recipients);
      Serial.printf("Subject: %s\n", result.subject);
    }
    Serial.println("----------------\n");
  }
}

void readVarFromFile(){
  File myFile16 = LittleFS.open("Variables.txt", "r");
  String abc = myFile16.readString();
  Serial.print("Variable abc is: ");
  Serial.println(abc);
  DynamicJsonBuffer jsonBuffer16;
  JsonObject& json16 = jsonBuffer16.parseObject(abc);
  if(!json16.success()){
    delay(1000);
    Serial.println("ERROR in readVarFromFile()");
    return;
  }
  var = json16["var1"];
  myFile16.close();
  // here variable is correct
  Serial.print("Variable is(first read): ");
  Serial.println(var);
}

saman-taghipour, 2021-01-11:
this variable is correct in every function
but after this line:
if (!smtp.connect(&session))
return;

it is corroupted!!!