mobizt/ESP-Mail-Client #13: No auth SMTP can't connect

Rhys79, 2021-01-07:
Is there a way to disable auth to use this with an unauthenticated internal SMTP relay?  Not seeing an option for that, and it's not working.

mobizt, 2021-01-07:
Just not set password.

Rhys79, 2021-01-07:
Hmm, I'll try again, getting a failed to connect so far.

Rhys79, 2021-01-07:
Well, it's connecting now but erroring out, which is odd since the last thing is OK.

SMTP server connected, wait for greeting...
< S: 220 ********* Microsoft ESMTP MAIL Service, Version: 10.0.14393.2608 ready at  Thu, 7 Jan 2021 11:36:10 -0500 

Sending greeting response...
> C: send smtp command, HELO
EHLO *******.com
250-********* Hello [***.***.***.***]
< S: 250-*********** Hello [***.***.***.***]
250-TURN
< S: 250-TURN
250-SIZE
< S: 250-SIZE
250-ETRN
< S: 250-ETRN
250-PIPELINING
< S: 250-PIPELINING
250-DSN
< S: 250-DSN
250-ENHANCEDSTATUSCODES
< S: 250-ENHANCEDSTATUSCODES
250-8bitmime
< S: 250-8bitmime
250-BINARYMIME
< S: 250-BINARYMIME
250-CHUNKING
< S: 250-CHUNKING
250-VRFY
< S: 250-VRFY
250 OK
< S: 250 OK
Error, code: 250, text: OK
> E: code: 250, text: OK

mobizt, 2021-01-08:
The library updated with no auth allowance.

closed 2021-01-08 by mobizt