mobizt/ESP-Mail-Client #14: Search in inbox from multiple email addresses

luigimarmo, 2021-01-09:
Hi,
I'm trying to using OR condition search criteria mixed to FROM address.

1. In first test i setted in config.search.criteria = "UID SEARCH FROM (email1@example.com OR email2@example.com)";
2. In second test i setted in config.search.criteria = "FROM (email1@example.com OR email2@example.com)";

I already tried with some other criteria mixing "FROM" and "OR" condition, but with no luck.
How is it possible search in inbox mails using multiple email addresses?
Thanks

mobizt, 2021-01-09:
Please read the RFC document or this [brief available keywords](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#search-criteria).

The library does nothing with your provided search criteria unless checking for missing UID and SEARCH keywords at the beginning and prepend or insert these missing keywords in that search criteria e.g it should begin with **UID SEARCH ..........**.

closed 2021-01-09 by mobizt

mobizt, 2021-01-09:
From RFC 3501 [search command](https://tools.ietf.org/html/rfc3501#page-49). The OR keyword usage is in this form

  `OR <search-key1> <search-key2>`

Which is the [polish notation](https://en.wikipedia.org/wiki/Polish_notation) which also used in all IMAP commands.

See this [this](https://www.limilabs.com/blog/imap-search-requires-parentheses) for reference.

In your case, you can try

**UID SEARCH OR FROM email1@example.com FROM  email2@example.com**