mobizt/ESP-Mail-Client #24: could not get ip from host

RaspberryPiDude314159, 2021-02-06:
I was trying to send an email using this, and all of the variables look right, but this is the error I got while trying to upload it to y ESP32 microcontroller.
```
15:17:26.753 -> 
15:17:28.072 -> Connecting to AP..
15:17:28.564 -> WiFi connected.
15:17:28.564 -> IP address: 
15:17:28.564 -> [Redacted IP address, looks right though]
15:17:28.564 -> 
15:17:28.564 -> Connecting to SMTP server...
15:17:28.564 -> > C: ESP Mail Client v1.0.8
15:17:28.564 -> > C: connect to SMTP server
15:17:28.564 -> > C: host > smtp.gmail.com
15:17:28.564 -> > C: port > 465
15:17:28.564 -> > C: starting socket
15:17:32.583 -> ! E: could not get ip from host
15:17:32.583 -> > C: cleaning SSL connection
15:17:32.583 -> Error, unable to connect to server
15:17:32.583 -> > E: unable to connect to server

```
I've done everything I can think of.
Does anyone have a fix?

mobizt, 2021-02-07:
Your internet connection problem of your router or AP not related to the library operation.

closed 2021-02-07 by mobizt

RaspberryPiDude314159, 2021-02-07:
That's not the problem! Any other internet related script runs perfectly fine.
It can connect to my router fine.

mobizt, 2021-02-07:
Your device can't connect to the internet.

Please try the simple https example in Arduino IDE to connect to some server or use the WiFiClientSecure or HTTPClient class to connect to some server over ssl port.

You also should update ESP32 Arduino core SDK and the library.

Your issue is simple and on the user part.

mobizt, 2021-02-07:
You able to connect to AP or router but no packet sent from your router as it's not response to your device.

> That's not the problem! Any other internet related script runs perfectly fine.

It's should be ok with the library too.

mobizt, 2021-02-07:
"could not get ip from host" is the error returns from WiFiClientSecure class.

That's means it can't resolve the ip from "smtp.gmail.com" because it probes to connecting to "smtp.gmail.com" but no response in time due to the internet connection problem.

mobizt, 2021-02-07:
You also should contact the network admin to open the port 465 if it blocked by Firewall.

mobizt, 2021-02-07:
The [discussion](https://github.com/mobizt/ESP-Mail-Client/discussions) is available for this kind of problem.