mobizt/ESP-Mail-Client #28: Possibility to send a non const char

cezar-pereira, 2021-02-23:
SMTP_Message.text.content accept only const char, it happens that sometimes we want to send a dynamic message.

mobizt, 2021-02-23:
Please learn the basic programming and  use [discussion](https://github.com/mobizt/ESP-Mail-Client/discussions) instead of open such issue.

const char* is the pointer the the constant char which you can change the pointer to other const char you want.

You can use String class to create the string and point the internal constant char to to the function or class or struct like this

String mystr = "something";

SMTP_Message.text.content = mystr.c_str();

mobizt, 2021-02-23:
The email  SMTP service is not instant message or chat service, you will be blocked by the SMTP server or consider to be spam email when send the message too often.

closed 2021-02-23 by mobizt