mobizt/ESP-Mail-Client #37: debug print shows date/time always back to the past

xiaolaba, 2021-03-14:
hi, it is possible an issue or no update to the log only, how is date/time info obtained, depends on SMTP server or the ESP device ? I was used the gmail for testing, could we uses this time/date info for other purpose ? or really need RTC chip to work with ESP device together ?
[image]

mobizt, 2021-03-14:
This is not the issue. You need to set the system time by your own to get the accurate timestamp.

mobizt, 2021-03-14:
You can get time from NTP server or RTC chip. The timestamp is the milliseconds from Midnight Jan 1, 1970.

mobizt, 2021-03-14:
Please use [Discussions](https://github.com/mobizt/ESP-Mail-Client/discussions) for question.

closed 2021-03-14 by mobizt

mobizt, 2021-03-14:
The device time is not related to the Email sending and reading process as it depends on SMTP and IMAP servers to set the time of the message.

The device time is for user reporting only.

xiaolaba, 2021-03-15:
thanks, my bad as no reading those functions in detail.
for those other who may has the same query or trivial, here is what to do if want to display or reveal local-time properly, I have tried and following, it is working perfectly. thanks again.

'

#include <extras\ESPTimeHelper.h>  // enable code for the time function

/* The ESPTimeHelper object used for time function */
ESPTimeHelper esptime;

void set_mytime ()
{
  float gmtOffset=9;  // GMT+9 = Janan local time
  float daylightOffset=0;
  esptime.setClock(gmtOffset, daylightOffset);
}

/* Callback function to get the Email sending status */
void smtpCallback(SMTP_Status status)
{
  /* Print the current status */
  Serial.println(status.info());

  set_mytime(); //perhaps add to display function would be more easy for others
  //esptime.setClock(gmtOffset, daylightOffset);
`