mobizt/ESP-Mail-Client #51: always true tests and infinite loops

mMerlin, 2021-04-11:
Builds using the ESP_Mail_Client library get 4 warnings for "comparison of unsigned expression >= 0 is always true" in ESP_Mail_Client.h (lines 1863. These are all from lines like:
```c++
for (size_t i = _«something».size() - 1; i >= 0; i--)
```
Since `size_t` is an unsigned datatype, `0 - 1` gives a positive result, and the `i >= 0` test never fails. That looks like an infinite loop problem. I have not yet noticed a frozen system, so it appears none of my code has actually executed those blocks of code.

If you want to stick with `size_t`, those could be changed to:
```c++
for (size_t i = _«something».size(); i >= 1; i--)
```
or
```c++
for (size_t i = _«something».size(); i > 0; i--)
```
and use `i - 1` in the loops instead of `i`. 

or (if processing order does not matter)
```c++
for (size_t i = 0; i < _«something».size(); i++)
```
Otherwise, replaced `size_t` with a sufficient size signed datatype.

The raw source code:
```c++

  /** Clear all the inline images
  */
  void clearInlineimages()
  {
    for (size_t i = _att.size() - 1; i >= 0; i--)
    {
      if (_att[i]._int.att_type == esp_mail_att_type_inline)
        _att.erase(_att.begin() + i);
    }
  };

  /* Clear all the attachments */
  void clearAttachments()
  {
    for (size_t i = _att.size() - 1; i >= 0; i--)
    {
      if (_att[i]._int.att_type == esp_mail_att_type_attachment)
        _att.erase(_att.begin() + i);
    }

    for (size_t i = _parallel.size() - 1; i >= 0; i--)
      _parallel.erase(_parallel.begin() + i);
  };

  /** Clear all rfc822 message attachment
  */
  void clearRFC822Messages()
  {
    for (size_t i = _rfc822.size() - 1; i >= 0; i--)
    {
      _rfc822[i].clear();
      _rfc822.erase(_rfc822.begin() + i);
    }
  };
```

mobizt, 2021-04-12:
Thanks for inform this. I will fix it in the next update.

closed 2021-04-12 by mobizt

mMerlin, 2021-04-12:
While you are looking at the previous issue, here is another with a similar cause. In ESP_Mail_Client.cpp, line 7444, `fbuf` is declared with `unsigned char *fbuf = new unsigned char[3];`. However, lines 7509, 7518, 7524 are checking for `< 0`, which always fails. unsigned datatypes are never less than 0. So those errors conditions will never get set.

The raw code:

```c++
  unsigned char *fbuf = new unsigned char[3];
// …
      if (len - fbufIndex == 1)
      {
        fbuf[0] = file.read();
        if (fbuf[0] < 0)
        {
          errorStatusCB(smtp, MAIL_CLIENT_ERROR_FILE_IO_ERROR);
          break;
        }
      }
      else if (len - fbufIndex == 2)
      {
        fbuf[0] = file.read();
        if (fbuf[0] < 0)
        {
          errorStatusCB(smtp, MAIL_CLIENT_ERROR_FILE_IO_ERROR);
          break;
        }
        fbuf[1] = file.read();
        if (fbuf[1] < 0)
        {
          errorStatusCB(smtp, MAIL_CLIENT_ERROR_FILE_IO_ERROR);
          break;
        }
      }
      break;
```

Since this block has a forced break of the containing while at the end, here is a simple way to reduce the duplication of the file.read() calls and associated error checks. Assuming I got that for loop test correct. Probably needs a maximum, to prevent the loop from executing more than twice.

```c++
      for (size_t i = 0; i < 2 - (len - fbufIndex); i++)
      {
        fbuf[i] = file.read();
        if ((int)fbuf[i] < 0)
        {
          errorStatusCB(smtp, MAIL_CLIENT_ERROR_FILE_IO_ERROR);
          abortFlag = True;
          break;
        }
      }
      break;
```

That test condition expression is supposed to be equivalent to:

```c++
      size_t readLimit = 0;
      if (len - fBufIndex == 1)
      {
        readLimit = 1;
      }
      else if (len - fBufIndex == 2)
      {
        readLimit = 2;
      }
      for (size_t i = 0; i < readLimit; i++)
```