mobizt/ESP-Mail-Client #54: Enable "1-bit" SD card access (as per SD_MMC)

alastaira, 2021-04-15:
Hi,

I read some old threads where people using ESP32-CAM boards wanted to attach an image saved to SD card from the camera. Many of these examples seem to be based on a tutorial using SD_MMC rather than SD, which I understand is not compatible with ESP Mail.

However, unless I'm mistaken, one of the reasons for using SD_MMC is that, by passing "true" as the third parameter value, it can be initialized in "1 bit mode", which crucially frees up pins 12 and 13 from the SPI interface. Unfortunately, I can't see an equivalent overload for the sdBegin() method of MailClient - it seems to require the full 4-pin SPI interface, which means there are no spare GPIO pins on an ESP32-CAM board. I'd be grateful if you could verify my understanding or whether there's a workaround? Thankyou!

mobizt, 2021-04-16:
Now the library update is available (v1.1.6) with this SD_MMC config added and some fixes.

To enable SD_MMC card

In ESP_Mail_FS.h, add these lines
https://github.com/mobizt/ESP-Mail-Client/blob/5483a597c0b056828e331e9c04e8b37ec1891f1e/src/ESP_Mail_FS.h#L27-L29

And also remove these lines
https://github.com/mobizt/ESP-Mail-Client/blob/5483a597c0b056828e331e9c04e8b37ec1891f1e/src/ESP_Mail_FS.h#L32-L33

The config for SD_MMC is [here](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#initialize-the-sd_mmc-card-esp32-only).

closed 2021-04-16 by mobizt

alastaira, 2021-04-16:
Wow - that was quick, thanks! :)