mobizt/ESP-Mail-Client #55: Reboots with Ethernet.

zekageri, 2021-04-22:
**If i send email from an esp with wifi connection to router, everything is okay, but if i try it with an ethernet enabled esp32, it crashes with the following backtrace:**

```
Connecting to SMTP server...
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060530  A0      : 0x800fc8c5  A1      : 0x3ffb1aa0       
A2      : 0x3ffd81c4  A3      : 0x00000000  A4      : 0x3f405856  A5      : 0x0000001e  
A6      : 0xffffffff  A7      : 0xff000000  A8      : 0x800fc860  A9      : 0x3ffb1a90       
A10     : 0x3ffd81c4  A11     : 0x3f40586b  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014       
EXCVADDR: 0x00000000  LBEG    : 0x4008d5dd  LEND    : 0x4008d5ed  LCOUNT  : 0xffffffec  

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffb1aa0 0x400fc8c2:0x3ffb1ac0 0x400fc146:0x3ffb1b20 0x400fc34c:0x3ffb1b70 0x400fbcec:0x3ffb1b90 0x400f6f9e:0x3ffb1bb0 0x400f8ddd:0x3ffb1c60 0x400e466d:0x3ffb1c90 0x400e4ad3:0x3ffb1e80 0x400e5f95:0x3ffb1f00 0x4010cfa0:0x3ffb1fb0 0x4008f566:0x3ffb1fd0
  #0  0x00000000:0x3ffb1aa0 in ?? ??:0
  #1  0x400fc8c2:0x3ffb1ac0 in esp_mail_ssl_client32::start_socket(esp_mail_ssl_client32::esp_mail_ssl_ctx32*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool) at lib\ESP-Mail-Client-master\src\wcs\esp32/esp_mail_ssl_client32.cpp:594
  #2  0x400fc146:0x3ffb1b20 in ESP_Mail_WCS32::connect(char const*, unsigned short, char const*, char const*, char const*) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_WCS32.cpp:457
  #3  0x400fc34c:0x3ffb1b70 in ESP_Mail_WCS32::connect(char const*, unsigned short) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_WCS32.cpp:457
  #4  0x400fbcec:0x3ffb1b90 in ESP_Mail_HTTPClient32::connect(bool) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_HTTPClient32.cpp:148
  #5  0x400f6f9e:0x3ffb1bb0 in ESP_Mail_Client::smtpAuth(SMTPSession*) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:101      
  #6  0x400f8ddd:0x3ffb1c60 in SMTPSession::connect(esp_mail_session_config_t*) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:101
  #7  0x400e466d:0x3ffb1c90 in sendEmail(String, String, String, String) at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #8  0x400e4ad3:0x3ffb1e80 in sendRestartEmail() at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #9  0x400e5f95:0x3ffb1f00 in loop() at src/Own_Headers/ModBus.h:365 (discriminator 2)      
  #10 0x4010cfa0:0x3ffb1fb0 in loopTask(void*) at C:\Users\Pc\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #11 0x4008f566:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

**I'am using LAN8720 eth chip with the ETH library from espressif.**

**Email sender function is looks like this**

```C
String MAINTAINER_NAME  = "Dr.Random";
String MAINTAINER_EMAIL  = "email is good";
String MAINTAINER_PASS    = "password is good.";
String GOOGLE_SMTP_HOST = "smtp.gmail.com";
#define GOOGLE_SMTP_PORT 465
#define RESTART_EMAIL_TIME 30000

SMTPSession smtp;
void smtpCallback(SMTP_Status status);

const inline boolean sendEmail(String sub,String msg,String emailTo,String nameTo){
    if(ETH_Got_IP){
        smtp.debug(0);
        smtp.callback(smtpCallback);
        ESP_Mail_Session session;

        session.server.host_name    = GOOGLE_SMTP_HOST.c_str();
        session.server.port         = GOOGLE_SMTP_PORT;
        session.login.email         = MAINTAINER_EMAIL.c_str();
        session.login.password      = MAINTAINER_PASS.c_str();

        SMTP_Message message;
        message.sender.name     = MAINTAINER_NAME.c_str();
        message.sender.email    = MAINTAINER_EMAIL.c_str();
        message.subject         = sub.c_str();
        message.addRecipient(nameTo.c_str(), emailTo.c_str());
        message.text.flowed     = true;
        message.text.content    = msg.c_str();
        message.text.charSet    = "us-ascii";

        message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
        message.priority        = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
        message.response.notify = esp_mail_smtp_notify_success | esp_mail_smtp_notify_failure | esp_mail_smtp_notify_delay;

        if (!smtp.connect(&session)){
            errorLog("Can't estabilish email session...");
            return false;
        }

        if (!MailClient.sendMail(&smtp, &message)){
            errorLog("Error sending Email, " + smtp.errorReason());
            return false;
        }

        return true;
    }
    return false;
}
```

mobizt, 2021-04-22:
You should follow [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/ESP32_Ethernet_Send_Text/ESP32_Ethernet_Send_Text.ino) and provide the minimum and clean working code.

zekageri, 2021-04-22:
Thank you for the response. I will look into this example but for now, it seems to me that i'm doung just like this. strange

mobizt, 2021-04-22:
It's not strange if you remind about how the ISR and how cpu stored the current execution data.

You **must not** call any function or do something that takes too long time in the WiFi event callback function because of nested function called and  stack grows until overflow.

You should use the variable to keep the Ethernet status in the WiFi event callback function and check that status variable in the main app loop function as in the example which the sending Email function will not call multiple times as you directly called in the WiFi event callback.

https://github.com/mobizt/ESP-Mail-Client/blob/5483a597c0b056828e331e9c04e8b37ec1891f1e/examples/ESP32_Ethernet_Send_Text/ESP32_Ethernet_Send_Text.ino#L117-L129

And in loop, **eth_connected** is the variable that keep the Ethernet connected status in the example.

https://github.com/mobizt/ESP-Mail-Client/blob/5483a597c0b056828e331e9c04e8b37ec1891f1e/examples/ESP32_Ethernet_Send_Text/ESP32_Ethernet_Send_Text.ino#L271

zekageri, 2021-04-22:
Yes, i did exactly like that. 
On the eth connected event i set my flag to true and after 30 sec i send the email in the loop once.

The strange part is that if i skip this step, and send an email from an other part of the program later, the email with the same funciton is success.

zekageri, 2021-04-22:
**My function in loop wich is calling the sender:**

```
const inline void sendRestartEmail(){
    if( ( millis() - gotIpMillis >= RESTART_EMAIL_TIME ) && canSendRestartEmail){
        canSendRestartEmail = false;
        String emailMessage = "Smart Home restarted at: ";
        emailMessage += Time.GetCurrentTime();
        emailMessage += " with dns: ";
        emailMessage += String(hostName);
        emailMessage += "\nRestart was ";
        if(isRestartUnexpected){
            emailMessage += "unexpected.";
        }else{
            emailMessage += "expected.";
        }
        
        if( !sendMail("Smart Home",emailMessage,"drsedate@gmail.com",MAINTAINER_NAME) ){
            gotIpMillis = millis();
            canSendRestartEmail = true;
        }
    }
}
```

zekageri, 2021-04-22:
It seems to me that if i wait 50 sec after eth_gotip event instead of 30 sec, it is working.

mobizt, 2021-04-22:
It's not the issue about timing as my test, you will see it will send email immediately after the Ethernet is ready with sendMillis == 0 in the checking line in the loop.

Please try the example first. If you still get the error, the issue may be somewhere in the ETH library.

zekageri, 2021-04-22:
Thank you for your help. 
I think the problem lies within my implementation somewhere. There are a lot of things going on, maybe the stack depth is not enough for it or something like that. I close that issue and sorry for that i didn't investigate the problem deeper.
Thank you again for this amazing lib.

zekageri, 2021-04-23:
### Sorry for reopening but the problem is still presist.

I tried a few times and in different configurations but my final result is that if the `smtp.debug();` is called with debug level 0, the esp32 crashes every time. If the debug level is 1 like this: `smtp.debug(1);` no crashes whatever i do but the serial is flooded with texts that i don't need.

**This is my sketch:**

```C
String MAINTAINER_NAME  = "Dr.Random";
String MAINTAINER_EMAIL = "correctEmail@gmail.com";
String MAINTAINER_PASS  = "correctPassword_AuthIsOkay";
String GOOGLE_SMTP_HOST = "smtp.gmail.com";
#define GOOGLE_SMTP_PORT 465
#define RESTART_EMAIL_TIME 10000

SMTPSession smtp;

boolean sendMail(String sub,String msg,String emailTo,String nameTo){
    if(ETH_Got_IP){
        msg += "\nCurrent Time is: ";
        msg += Time.GetCurrentTime();
        msg += "\nMessage is from host: ";
        msg += String(hostName);
        smtp.debug(1); // Here is the problem, if the level is 0, it is crashing.
        ESP_Mail_Session session;
        session.server.host_name  = GOOGLE_SMTP_HOST.c_str();
        session.server.port       = GOOGLE_SMTP_PORT;
        session.login.email       = MAINTAINER_EMAIL.c_str();
        session.login.password    = MAINTAINER_PASS.c_str();
        SMTP_Message message;

        message.sender.name     = MAINTAINER_NAME.c_str();
        message.sender.email    = MAINTAINER_EMAIL.c_str();
        message.subject         = sub.c_str();
        message.addRecipient(nameTo.c_str(), emailTo.c_str());

        message.text.content  = msg.c_str();
        message.text.charSet  = "us-ascii";
        message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
        message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
        message.response.notify = esp_mail_smtp_notify_success | esp_mail_smtp_notify_failure | esp_mail_smtp_notify_delay;

        if (!smtp.connect(&session)){
            errorLog("Can't estabilish email session...");
            return false;
        }
        if (!MailClient.sendMail(&smtp, &message)){
            errorLog("Error sending Email, " + smtp.errorReason());
            return false;
        }
        return true;
    }else{
        return false;
    }
}

const inline void sendRestartEmail(){
    if( ( millis() - gotIpMillis >= RESTART_EMAIL_TIME ) && canSendRestartEmail){
        canSendRestartEmail = false;
        String emailMessage = "Smart Home restarted ";
        if(isRestartUnexpected){
            emailMessage += " and it was unexpected.";
        }else{
            emailMessage += " and it was expected.";
        }

        if( !sendMail("Smart Home",emailMessage,"correctEmailToSend@gmail.com",MAINTAINER_NAME) ){
            gotIpMillis = millis();
            canSendRestartEmail = true;
        }else{
            vTaskDelete(Email_Task_Handle);
        }
    }
}
```

**Serial dump from the crash:**

```
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x800fcb39  A1      : 0x3ffe90f0
A2      : 0x3ffd840c  A3      : 0x00000000  A4      : 0x3f405836  A5      : 0x0000001e
A6      : 0xffffffff  A7      : 0xff000000  A8      : 0x800fcad4  A9      : 0x3ffe90e0
A10     : 0x3ffd840c  A11     : 0x3f40584b  A12     : 0x000000ff  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014
EXCVADDR: 0x00000000  LBEG    : 0x4008d75d  LEND    : 0x4008d76d  LCOUNT  : 0xffffffec

ELF file SHA256: 0000000000000000

Backtrace: 0x00000000:0x3ffe90f0 0x400fcb36:0x3ffe9110 0x400fc3ba:0x3ffe9170 0x400fc5c0:0x3ffe91c0 0x400fbf60:0x3ffe91e0 0x400f7212:0x3ffe9200 0x400f9051:0x3ffe92b0 0x400e47f6:0x3ffe92e0 0x400e4cea:0x3ffe94c0 0x400e4dcb:0x3ffe9520 0x4008fda6:0x3ffe9540
  #0  0x00000000:0x3ffe90f0 in ?? ??:0
  #1  0x400fcb36:0x3ffe9110 in esp_mail_ssl_client32::start_socket(esp_mail_ssl_client32::esp_mail_ssl_ctx32*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool) at lib\ESP-Mail-Client-master\src\wcs\esp32/esp_mail_ssl_client32.cpp:594
  #2  0x400fc3ba:0x3ffe9170 in ESP_Mail_WCS32::connect(char const*, unsigned short, char const*, char const*, char const*) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_WCS32.cpp:457
  #3  0x400fc5c0:0x3ffe91c0 in ESP_Mail_WCS32::connect(char const*, unsigned short) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_WCS32.cpp:457
  #4  0x400fbf60:0x3ffe91e0 in ESP_Mail_HTTPClient32::connect(bool) at lib\ESP-Mail-Client-master\src\wcs\esp32/ESP_Mail_HTTPClient32.cpp:148
  #5  0x400f7212:0x3ffe9200 in ESP_Mail_Client::smtpAuth(SMTPSession*) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:101
  #6  0x400f9051:0x3ffe92b0 in SMTPSession::connect(esp_mail_session_config_t*) at c:\users\pc\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/vector.tcc:101
  #7  0x400e47f6:0x3ffe92e0 in sendMail(String, String, String, String) at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #8  0x400e4cea:0x3ffe94c0 in sendRestartEmail() at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #9  0x400e4dcb:0x3ffe9520 in Email_Task(void*) at src/Own_Headers/ModBus.h:365 (discriminator 2)
  #10 0x4008fda6:0x3ffe9540 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

mobizt, 2021-04-23:
I see that you define ESP_Mail_Session as local var, this is the issue where all data included the SSL client is in this object.

When you set the callback or even get the sending result, the data to send to the callback and the sending result are taken from this ESP_Mail_Session object which are defined out of usage scope.

You should define ESP_Mail_Session as the global object or static to make it available for all usage scope.

When the session was close, the memory used by internal SSL client are freed and only some user data are reserved until the object disposed.

zekageri, 2021-04-23:
> I just see your comment in your above code. I'll check again.

Thank you!

mobizt, 2021-04-23:
The update is available [v1.1.7](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.1.7) with the debug issue fixed, please update.

closed 2021-04-23 by mobizt