mobizt/ESP-Mail-Client #64: Not compiling

VolcanoJ25, 2021-06-14:
Trying a basic send email.  Getting this on compile

> 

In file included from C:\Users\administrator\Documents\Arduino\libraries\SmtpClient/SmtpClient.h:2:0,
                 from C:\Users\administrator\Documents\projects\data.intelligence.onl\MicroControllers\TempMonitorMK1\TempMonitorMK1.ino:20:
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:64:28: error: 'LOG_INFO' is not a type
 #define info(...) coap_log(LOG_INFO, __VA_ARGS__)
                            ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:37: note: in definition of macro 'coap_log'
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                                     ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:1787:38: note: in expansion of macro 'info'
   struct esp_mail_folder_info_item_t info(size_t index)
                                      ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:23: error: expected ';' at end of member declaration
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                       ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:64:19: note: in expansion of macro 'coap_log'
 #define info(...) coap_log(LOG_INFO, __VA_ARGS__)
                   ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:1824:15: note: in expansion of macro 'info'
   const char *info();
               ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:64:36: error: expected ')' before ',' token
 #define info(...) coap_log(LOG_INFO, __VA_ARGS__)
                                    ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:37: note: in definition of macro 'coap_log'
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                                     ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:1824:15: note: in expansion of macro 'info'
   const char *info();
               ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:23: error: expected ';' at end of member declaration
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                       ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:64:19: note: in expansion of macro 'coap_log'
 #define info(...) coap_log(LOG_INFO, __VA_ARGS__)
                   ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:2125:15: note: in expansion of macro 'info'
   const char *info();
               ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:64:36: error: expected ')' before ',' token
 #define info(...) coap_log(LOG_INFO, __VA_ARGS__)
                                    ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:37: note: in definition of macro 'coap_log'
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                                     ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:2125:15: note: in expansion of macro 'info'
   const char *info();
               ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:66:29: error: 'LOG_DEBUG' is not a type
 #define debug(...) coap_log(LOG_DEBUG, __VA_ARGS__)
                             ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:37: note: in definition of macro 'coap_log'
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                                     ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:2418:8: note: in expansion of macro 'debug'
   void debug(int level);
        ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:66:29: error: 'LOG_DEBUG' is not a type
 #define debug(...) coap_log(LOG_DEBUG, __VA_ARGS__)
                             ^
C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/coap/Debug.h:58:37: note: in definition of macro 'coap_log'
 #define coap_log(...) coap_log_impl(__VA_ARGS__)
                                     ^
C:\Users\administrator\Documents\Arduino\libraries\ESP-Mail-Client\src/ESP_Mail_Client.h:2650:8: note: in expansion of macro 'debug'
   void debug(int level);
        ^
Multiple libraries were found for "SD.h"
 Used: C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board Node32s.

mobizt, 2021-06-14:
It's OK.

[image]

If you're currently installed SD library from third party, remove it. 

Post your code here if you're not installed SD library to check other libraries inclusion issue.

mobizt, 2021-06-14:
Also the library SmtpClient.h need to be removed.

closed 2021-06-14 by mobizt

VolcanoJ25, 2021-06-14:
Smtpclient.h is my implementation of this code.  What problem is it causing?Sent from my Verizon, Samsung Galaxy smartphone

VolcanoJ25, 2021-06-14:
Not familiar with that one, i don't think I added it anywhereSent from my Verizon, Samsung Galaxy smartphone

mobizt, 2021-06-14:
Please read my above post.

You should post your clean, clear and minimized code here. 
Any third party libs should be exclude from your code.