mobizt/ESP-Mail-Client #67: ESP_Mail_Client causing exception error

AXON24, 2021-06-25:
Hi Mobitz,

let me say "Thank You" for your great work on your mail script.

However, it sometimes runs into an exception error whilst performing the SSL/TLS handshake. Please see print-out below...

It appears to happen occasionally when sending the first mail (after a few hours). After the crash I can trigger the script many times without having any issues.

I tried to run it through the exception decoder, but the information given are meaningless to me. 

FYI, I'm using a Wemos D1 R1 board with the latest ESP8266 version (2.7.4). I'm having these problems with either your version 1.0.13. or 1.3.0.. I also changed the email provider from GMail to GMX, but it didn't improve anything

Have you encountered similar problems in the past? Or can you point me in the right direction to look for?

Many Thanks,

AXON24 

09:07:47.115 -> Connecting to SMTP server...
09:07:47.115 -> > C: ESP Mail Client v1.3.0
09:07:47.115 -> > C: connect to SMTP server
09:07:47.115 -> > C: host > mail.gmx.net
09:07:47.115 -> > C: port > 587
09:07:47.253 -> > C: smtp server connected
09:07:47.253 -> 
09:07:47.253 -> SMTP server connected, wait for greeting...
09:07:47.287 -> < S: 220 gmx.net (mrgmx104) Nemesis ESMTP Service ready
09:07:47.287 -> 
09:07:47.287 -> Sending greeting response...
09:07:47.287 -> > C: send smtp command, HELO
09:07:47.370 -> < S: 250-gmx.net Hello mydomain.net [80.228.80.66]
09:07:47.370 -> < S: 250-8BITMIME
09:07:47.370 -> < S: 250-SIZE 69920427
09:07:47.370 -> < S: 250 STARTTLS
09:07:47.370 -> 
09:07:47.370 -> Send command, STARTTLS
09:07:47.370 -> > C: send STARTTLS command
09:07:47.370 -> < S: 220 OK
09:07:47.370 -> > C: performing the SSL/TLS handshake
09:07:47.628 -> 
09:07:47.628 -> User exception (panic/abort/assert)
09:07:47.628 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
09:07:47.628 -> 
09:07:47.628 -> Abort called

mobizt, 2021-06-25:
If possible share your code with credential information removed.  It's about your device is running out of memory, the SSL client fail to reserve the memory.

Read the document (README) how to config MMU if you are using ESP8266 sdk v3.0.0.

mobizt, 2021-06-25:
You should debug and check your free heap regularly at any point of your code.

closed 2021-06-25 by mobizt

AXON24, 2021-06-25:
Hi mobizt,

Thanks for your quick reply.

I had a few more test and am quite certain that the e-mail script has a conflict with the Async Web Server somehow. That doesn't mean your code is causing the problem! You could be right that it might be a memory issue. As soon as I stop HTTP request from the spiffs (by simply closing the browser) , sending emails runs smoothly. I checked the free heaps at two points:

EMail with Async Web Server @ 18544 and 28864 with only the email script.

It would be OK for me to stop the HTTP request whilst sending an email, but I'm struggeling to get it back working once the email was delivered. That's causing hell of a lot stack errors. I'm a bit helpless in the moment.   

I have attached my project and perhaps you can come up with a sparking idea to kick me ahead. In the worst case I have to split the project and run the emails for example on a separate board.

Many thanks for your support!

Cheers,

AXON24

 
[AMU_ESP_NOW_WebServer_v2_1.zip](https://github.com/mobizt/ESP-Mail-Client/files/6716005/AMU_ESP_NOW_WebServer_v2_1.zip)

mobizt, 2021-06-25:
You should optimize your code to reduce the memory usage.

Avoid using String class. Use std::string instead when you needed to use the string class.
Store string in flash and copy to char array when use.
Use int8_t, uint8_t, int16_t or uint16_t instead of int.
Use enum to enumerate something that later use in comparison.