mobizt/ESP-Mail-Client #71: error: control reaches end of non-void function in ESP32_TCP_Client::send()

joshua-8, 2021-07-11:
**Build tool used:**

- [x] Arduino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

Windows 10, Arduino 1.8.13, esp32 1.0.6

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

none

**Description of problem:**
In versions 1.3.0 and 1.3.1 of this library, I'm getting the following error when I try to compile programs which include this library.

```
\Documents\Arduino\libraries\ESP_Mail_Client\src\wcs\esp32\ESP32_TCP_Client.cpp: In member function 'int ESP32_TCP_Client::send(const char*)':
\Documents\Arduino\libraries\ESP_Mail_Client\src\wcs\esp32\ESP32_TCP_Client.cpp:80:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1plus.exe: some warnings being treated as errors

exit status 1
Error compiling for board ESP32 Dev Module.

```

**Share code snippet to reproduce the issue:**
```yaml
The Send_Text example program reproduces the issue.
```

**Additional information and things you've tried:**

My interpretation of the issue is that the [send method](https://github.com/mobizt/ESP-Mail-Client/blob/a8d8c022cbc626b0f05fd116e0d0fc3f467538ad/src/wcs/esp32/ESP32_TCP_Client.cpp#L69-L80) is supposed to return an int, but in some cases there is no return value specified.

I don't know if this is a proper fix for the issue, but adding `return 0;` to the end of the method did make the error go away and my code start working again.

I would be happy to try to provide more information.

Thank you, Mobizt, for making so many great libraries!

-Joshua

mobizt, 2021-07-11:
Thanks for using the library and report this issue. 

I fix this already in [v1.3.2](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.3.2) and you can update.

joshua-8, 2021-07-11:
That solves the issue for me. You fixed it very quickly! Thank you!

closed 2021-07-11 by mobizt