mobizt/ESP-Mail-Client #72: Error compiling Send_Camera_Image

deltazerorsan, 2021-07-11:
**Build tool used:**

Windows 10

- [X ] Arduino 1.8.13
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [ X] ESP32-based ESP_CAM

**Other Libraries That are used:**

**Description of problem:**
Trying to compile the example, only added the needed parameters as WiFi credentials (lines 34, 35), SMTP server (line 43),  AUTHOR_EMAIL and AUTHOR_PASSWORD (lines 53, 54) and message.addRecipient (line 124).

The cam/OV2640.h is present and correctly referenced.

There is something I'm doing wrong but I can't find it.

Rom

`sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z7initCamv+0x8): undefined reference to `OV2640::init(camera_config_t)'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z7initCamv+0xc): undefined reference to `OV2640::run()'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z5setupv+0x8c): undefined reference to `OV2640::getfb()'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z5setupv+0x90): undefined reference to `OV2640::getSize()'
sketch\Send_Camera_Image.ino.cpp.o: In function `initCam()':
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::init(camera_config_t)'
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::run()'
sketch\Send_Camera_Image.ino.cpp.o: In function `setup()':
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::getfb()'
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::getSize()'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "SD.h"
 Used: C:\Users\rsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
 Not used: C:\Users\rsan\Documents\Arduino\libraries\SD
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\rsan\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
exit status 1
Error compiling for board AI Thinker ESP32-CAM.
`

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

mobizt, 2021-07-11:
Because you installed a third party library "SD.h" in Arduino libraries folder which musts be uninstalled as it is not compatible with ESP32 core libraries which already have SD.h lib installed.

The compiler chooses wrong SD.h library in this case then error in compilation.

closed 2021-07-11 by mobizt

deltazerorsan, 2021-07-12:
Thanks for your pointer!

I have found multiple copies of SD and remove it; same as WiFi. But...

`sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z7initCamv+0x8): undefined reference to `OV2640::init(camera_config_t)'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z7initCamv+0xc): undefined reference to `OV2640::run()'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z5setupv+0x7c): undefined reference to `OV2640::getfb()'
sketch\Send_Camera_Image.ino.cpp.o:(.literal._Z5setupv+0x80): undefined reference to `OV2640::getSize()'
sketch\Send_Camera_Image.ino.cpp.o: In function `initCam()':
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::init(camera_config_t)'
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::run()'
sketch\Send_Camera_Image.ino.cpp.o: In function `setup()':
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::getfb()'
C:\Users\rsan\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:2713: undefined reference to `OV2640::getSize()'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board AI Thinker ESP32-CAM.
`
Still nowhere to go.

Thanks for your patience,

Rom

mobizt, 2021-07-12:
This works for PlatformIO but failed in Arduino IDE.

For Arduino IDE, copy the **cam** folder to **libraries** folder of Arduino. 
And in sketch, include the header file, `#include "OV2640.h"`

deltazerorsan, 2021-07-12:
Thanks.

I was able to compile but I have a ESP32CAM and it was not possible for me to reassign the pin-out correctly and put it to work.

Thanks a lot for your support.

Rom

mobizt, 2021-07-12:
You should make your cam works first before integrate to this library.

Why you don't use "esp_camera.h" which is in the ESP32 Arduino core which easy?
https://randomnerdtutorials.com/esp32-cam-video-streaming-face-recognition-arduino-ide/

The pins config for general OV2640 built-in ESP32 boards included ESP32-CAM.
https://randomnerdtutorials.com/esp32-cam-camera-pin-gpios/

Old ESP32-Cam repository
https://github.com/igrr/esp32-cam-demo

Your ESP32-CAM can work with the cam sources provided by the Email example too if you config it right.

deltazerorsan, 2021-07-12:
Yes, my camera is working properly with other sketches based on "esp-camera" and I use the camera-pins.h to try to translate from M5 to ESP32CAM, but no success at this time.

Will take a look to https://randomnerdtutorials.com/esp32-cam-camera-pin-gpios/ just to check again.

Thanks!

Rom