mobizt/ESP-Mail-Client #73: message body is not downloaded

rolandwidmer, 2021-08-01:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**
none

**Description of problem:**
I tried your example code "Read_Single_Email.ino" with only one minor modification: search criteria were set to "not seen" (config.search.criteria = "UID SEARCH NOT SEEN";); this is probably not relevant in this issue.

The sketch finds the one unseen message in my inbox, fetches the subject of the mail, prints it out correctly BUT does not fetch/print the text message of the mail body. The status of the mail is left "not seen".

So my questions are:
How can I fetch/print the body of the mail? 
How can I set the status to "seen" after fetching it?

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**
This ist the printout from the arduino console:

14:16:00.521 -> Message fetch cmpleted
14:16:00.521 -> > C: message fetch completed
14:16:00.521 -> > C: cleaning SSL connection
14:16:00.521 -> 
14:16:00.521 -> ################################
14:16:00.521 -> Messsage Number: 1
14:16:00.521 -> Messsage UID: 1902
14:16:00.521 -> Messsage ID:  <1345373619.101.1627808201421@localhost>
14:16:00.521 -> Accept Language: 
14:16:00.521 -> Content Language: 
14:16:00.521 -> From: NAKIVO Backup & Replication <nakivo@XXXXXXXXXXXXXXXXXXXX.de>
14:16:00.521 -> From Charset: 
14:16:00.521 -> To: admin@XXXXXXXXXXXXXXXXXXXX.de
14:16:00.521 -> To Charset: 
14:16:00.521 -> CC: 
14:16:00.521 -> CC Charset: 
14:16:00.521 -> Date: Sun, 1 Aug 2021 10:56:41 +0200 (CEST)
14:16:00.521 -> Subject:  You have 1 new alarm
14:16:00.521 -> Subject Charset: 
14:16:00.521 -> Text Message: 
14:16:00.521 -> Text Message Charset: 
14:16:00.521 -> Text Message Transfer Encoding: 
14:16:00.521 -> HTML Message: 
14:16:00.521 -> HTML Message Charset: UTF-8
14:16:00.521 -> HTML Message Transfer Encoding: 7bit

mobizt, 2021-08-01:
You should use [discussions](https://github.com/mobizt/ESP-Mail-Client/discussions) instead of open new issue.

By design, the library was not alter any user message in mailbox as it's is not Email app agent to manage the messages in this limited memory environment.

If you need to modify the message flag, copy, delete or move messages, it is your own risk.

The [example for set message flag](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/Set_Flags/Set_Flags.ino).

The only reason for no message from fetching is the message structure is not a valid RFC822 message.

You should create a valid message by this library or use standard Email client.

closed 2021-08-01 by mobizt

mobizt, 2021-08-01:
Any email client you use must comply to [RFC822](https://datatracker.ietf.org/doc/html/rfc8266).
The message headers and body can recognized and parse correctly.

If you use non-standard email client to create and send message, the junk data or unrecognized or malformed data will not shown in this library. It may show in some email app but it's not valid message.