mobizt/ESP-Mail-Client #88: Bulk deletion?

Chance92, 2021-09-04:
**Build tool used:**

- [ ] Arduino

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based

**Other Libraries That are used:**

**Description of problem:**

Hi, I am a new user of this library. I am trying to find a way to delete all email in the inbox whenever it is checked. I have tried to use the Delete_Messsages example by creating a MessageList containing as many UID as possible, but it seems there's a limit to that. I noticed that the UID of new email will keep increasing even if the previous email have been deleted, but I can't create an infinite MessageList. May I ask how I can solve this? Is there a command of bulk deletion or resetting the MessageList?

```

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2021-09-04:
You should use Discussions for question.

Your requirement is not possible not only with this library but all email client apps.

The UID of message was assigned by the mail server, not this library.

Chance92, 2021-09-04:
Sorry, I am new to Github as well. I didn't realised there's a Discussions.

closed 2021-09-04 by Chance92