mobizt/ESP-Mail-Client #91: Error Compiling Wemos D1 mini pro new Version of library 1.4.0

floatAsNeeded, 2021-09-11:
Hi !

I get this error with the new update

```
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:49:1: error: expected class-name before '{' token
 {
 ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:52:3: error: 'ESP8266_WCS::~ESP8266_WCS()' marked override, but does not override
   ~ESP8266_WCS() override;
   ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:54:7: error: 'int ESP8266_WCS::connect(const char*, uint16_t)' marked override, but does not override
   int connect(const char *name, uint16_t port) override;
       ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:55:11: error: 'uint8_t ESP8266_WCS::connected()' marked override, but does not override
   uint8_t connected() override;
           ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:56:7: error: 'int ESP8266_WCS::available()' marked override, but does not override
   int available() override;
       ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:57:7: error: 'int ESP8266_WCS::read()' marked override, but does not override
   int read() override;
       ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:58:7: error: 'int ESP8266_WCS::read(uint8_t*, size_t)' marked override, but does not override
   int read(uint8_t *buf, size_t size) override;
       ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:59:10: error: 'size_t ESP8266_WCS::write(const uint8_t*, size_t)' marked override, but does not override
   size_t write(const uint8_t *buf, size_t size) override;
          ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:60:10: error: 'size_t ESP8266_WCS::write_P(const char*, size_t)' marked override, but does not override
   size_t write_P(PGM_P buf, size_t size) override;
          ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:61:7: error: 'int ESP8266_WCS::peek()' marked override, but does not override
   int peek()override;
       ^
C:\Users\eugen\Documents\Arduino\libraries\ESP_Mail_Client\src/wcs/esp8266/ESP8266_WCS.h:62:10: error: 'size_t ESP8266_WCS::peekBytes(uint8_t*, size_t)' marked override, but does not override
   size_t peekBytes(uint8_t *buffer, size_t length) override;
          ^

```

Tested with example code 'Send HTML'
ESP8266 2.7.4
Arduino IDE 1.8.13

floatAsNeeded, 2021-09-11:
Thanks, now all working okay

closed 2021-09-11 by floatAsNeeded

mobizt, 2021-09-11:
Now I fix this in the [latest library update](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.4.1) for compilation error in ESP8266 Core SDK v2.7.4 and earlier.

mobizt, 2021-09-11:
If you update ESP8266 Arduino Core SDK to v3.x.x, you may need to config the MMU which you can read [here](https://github.com/mobizt/Firebase-ESP-Client#ide-configuaration-for-esp8266-mmu---adjust-the-ratio-of-icache-to-iram) for how to.

floatAsNeeded, 2021-09-11:
Yeah, I already done that! Thanks for the remind. That's good to have also a fix for the previous SDK versions, because I think many like me that didn't update can encounter the same problem.