mobizt/ESP-Mail-Client #92: ESP32 crashing when fetching a message(header) from an iCloud IMAP account

dkalliv, 2021-09-11:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
The ESP32 crashes when trying to fetch a message header from an iCloud IMAP account. After experiencing the issue with my own code, I was able to reproduce the same problem using the Search_Emails and Read_Single_Email examples provided with the library (I'm using Version 1.4.0). The log pasted below is produced using the unmodified (apart from wifi and imap credentials) Read_Single_Email example.

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here
Connecting to AP...........
WiFi connected.
IP address: 
XXX.XXX.XXX.XXX

Connecting to IMAP server...
> C: ESP Mail Client v1.4.0
> C: connect to IMAP server
> C: host > imap.mail.me.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
IMAP server connected
> C: server connected

Check the capability...
> C: check the capability
$ CAPABILITY
* OK [CAPABILITY XAPPLEPUSHSERVICE IMAP4 IMAP4rev1 SASL-IR AUTH=ATOKEN AUTH=PLAIN] (2117B22-978ffe0c9356) st43p00im-tygg10060301.me.com
* CAPABILITY XAPPLEPUSHSERVICE IMAP4 IMAP4rev1 SASL-IR AUTH=ATOKEN AUTH=PLAIN
$ OK Completed

Logging in...
> C: send imap command, LOGIN
$ LOGIN XXXXXXXX@icloud.com XXXXXXXXXXXX
$ OK user XXXXXXXX logged in

Reading the list of mailboxes...
> C: send imap command, LIST
$ LIST "" *
* LIST (\Noinferiors) "/" "INBOX"
* LIST () "/" "Trash"
* LIST () "/" "Sent"
* LIST () "/" "test"
$ OK LIST completed (took 39 ms)

Available folders: INBOX, Trash, Sent, test

Selecting the INBOX folder...
> C: open the mailbox folder
$ EXAMINE "INBOX"
* 1 EXISTS
* 0 RECENT
* OK [UNSEEN 1]
* OK [UIDVALIDITY 1]
* OK [UIDNEXT 3]
* FLAGS (\Answered \Flagged \Draft \Deleted \Seen \Recent $MailFlagBit0 $MailFlagBit1 $MailFlagBit2 $Forwarded Redirected $NotJunk NotJunk)
* OK [PERMANENTFLAGS (\Answered \Flagged \Draft \Deleted \Seen \Recent $MailFlagBit0 $MailFlagBit1 $MailFlagBit2 $Forwarded Redirected $NotJunk NotJunk \*)]
$ OK [READ-WRITE] EXAMINE completed (took 2 ms)

Info of the selected folder
Total Messages: 1
Predicted next UID: 3
Flags: \Answered, \Flagged, \Draft, \Deleted, \Seen, \Recent, $MailFlagBit0, $MailFlagBit1, $MailFlagBit2, $Forwarded, Redirected, $NotJunk, NotJunk

Fetching message 1, UID: 2
> C: send imap command, FETCH
> C: fetch message header
$ UID FETCH 2 BODY.PEEK[HEADER.FIELDS (SUBJECT FROM TO DATE CC Message-ID Accept-Language content-type Content-transfer-encoding Content-Language)]
* 1 FETCH (UID 2 BODY[HEADER.FIELDS (SUBJECT FROM TO DATE CC Message-ID Accept-Language content-type Content-transfer-encoding Content-Language)] {328}
TO: XXXXXXXX@icloud.com
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40089fc5  PS      : 0x00060830  A0      : 0x800d5c27  A1      : 0x3ffb2040  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb2000  
A10     : 0x3ffdd8c8  A11     : 0x3ffde5c0  A12     : 0x3ffdd8c8  A13     : 0x00000016  
A14     : 0x00000016  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40089fc5  LEND    : 0x40089fd5  LCOUNT  : 0xffffffff  

Backtrace:0x40089fc2:0x3ffb20400x400d5c24:0x3ffb2050 0x400d691e:0x3ffb2070 0x400e1071:0x3ffb20a0 0x400e34d5:0x3ffb2680 0x400d2aed:0x3ffb2790 0x400e9c83:0x3ffb2820 

ELF file SHA256: 0000000000000000

Rebooting...
```

**Additional information and things you've tried:**
I tried downgrading to the ESP32 1.0.6 version, but experienced the same issue as with version 2.0.0. I also tested with a simple message with only a subject and no body, but had the same results.

mobizt, 2021-09-11:
You should decode the stack traces and enable debug in core not only in the library.

mobizt, 2021-09-11:
You should try with other mail server e.g. outlook or gmail to identify the causes.

This can be device and power failure issues too, which you can test with different board and shorter USB cable.

dkalliv, 2021-09-11:
Maybe I should have mentioned, that I can use the library without any problem with gmail or two local e-mail providers, all using different IMAP servers. I've tested this with two different ESP32 modules and different cables getting the same result.

mobizt, 2021-09-11:
I don't have any info for help. You should follow my request above for debugging.

dkalliv, 2021-09-11:
Here you go...

```yaml
Decoding stack results
0x400d5c24: ESP_Mail_Client::setHeader(IMAPSession*, char*, esp_mail_message_header_t&, int) at C:\Users\dkal\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp line 4666
0x400d691e: ESP_Mail_Client::handleHeader(IMAPSession*, char*, int, int&, esp_mail_message_header_t&, int&, int&) at C:\Users\dkal\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp line 4591
0x400e1071: ESP_Mail_Client::handleIMAPResponse(IMAPSession*, int, bool) at C:\Users\dkal\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp line 5834
0x400e34d5: ESP_Mail_Client::readMail(IMAPSession*, bool) at C:\Users\dkal\Documents\Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_Client.cpp line 357
0x400d2aed: setup() at C:\Users\dkal\Documents\Arduino\Read_Single_Email1/Read_Single_Email1.ino line 215
0x400e9c83: loopTask(void*) at C:\Users\dkal\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.0\cores\esp32\main.cpp line 38
```

mobizt, 2021-09-11:
And how about core debug?
You should  enable Core Debug Level to verbose.

When you want someone to help please offer all of your info and effort to make it easy for other that does not known your system, config and environment.

mobizt, 2021-09-11:
It seems that your device free heap is not enough.

The memory allocation was failed in just c++ string buffer concatenate in setHeader function.

closed 2021-09-11 by mobizt