mobizt/ESP-Mail-Client #94: Example "Read_single_Email" can't receive attachment

zjmeng2006, 2021-09-14:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
I tested example “Send_Attachment_File” and send email to mail box (zjmeng2006@outlook.com ). I found send is ok，but when I used example “Read_single_Email” to receive ，email subject can display ok，but the attached files not be receive.

**Share code snippet to reproduce the issue:**
```yaml
    /* Set the session config */
    session.server.host_name = IMAP_HOST;
    session.server.port = IMAP_PORT;
    session.login.email = AUTHOR_EMAIL;
    session.login.password = AUTHOR_PASSWORD;

    /* Setup the configuration for searching or fetching operation and its result */
    IMAP_Config config;

    /* Message UID to fetch or read e.g. 100 */
    config.fetch.uid = "";

    /* Set seen flag */
    //config.fetch.set_seen = true;

    /* Search criteria */
    config.search.criteria = "UID SEARCH ALL";

    /* Also search the unseen message */
    config.search.unseen_msg = true;

    /* Set the storage to save the downloaded files and attachments */
    config.storage.saved_path = "/email_data";

    /** The file storage type e.g.
     * esp_mail_file_storage_type_none,
     * esp_mail_file_storage_type_flash, and 
     * esp_mail_file_storage_type_sd 
    */
    config.storage.type = esp_mail_file_storage_type_none;

    /** Set to download heades, text and html messaeges, 
     * attachments and inline images respectively.
    */
    config.download.header = true;
    config.download.text = true;
    config.download.html = true;
    config.download.attachment = true;
    config.download.inlineImg = true;

    /** Set to enable the results i.e. html and text messaeges 
     * which the content stored in the IMAPSession object is limited
     * by the option config.limit.msg_size.
     * The whole message can be download through config.download.text
     * or config.download.html which not depends on these enable options.
    */
    config.enable.html = true;
    config.enable.text = true;

    /* Set to enable the sort the result by message UID in the ascending order */
    config.enable.recent_sort = true;

    /* Set to report the download progress via the default serial port */
    config.enable.download_status = true;

    /* Header fields parsing is case insensitive by default to avoid uppercase header in some server e.g. iCloud
    , to allow case sensitive parse, uncomment below line*/
    //config.enable.header_case_sesitive = true;

    /* Set the limit of number of messages in the search results */
    config.limit.search = 5;

    /** Set the maximum size of message stored in 
     * IMAPSession object in byte
    */
    config.limit.msg_size = 512;

    /** Set the maximum attachments and inline images files size
     * that can be downloaded in byte. 
     * The file which its size is largger than this limit may be saved 
     * as truncated file.
    */
    config.limit.attachment_size = 1024 * 1024 * 5;

    /* Connect to server with the session and config */
    if (!imap.connect(&session, &config))
        return;

    /*  {Optional] */
    printAllMailboxesInfo(imap);

    /* Open or select the mailbox folder to read or search the message */
    if (!imap.selectFolder("INBOX"))
        return;

    /*  {Optional] */
    printSelectedMailboxInfo(imap);
    //if (MailClient.setFlag(&imap, 24, "\\Seen", false))
    //    Serial.println("Setting FLAG success");
    //else
    //    Serial.println("Error, setting FLAG");
    /* Read or search the Email and close the session */
    MailClient.readMail(&imap);

```

**Additional information and things you've tried:**
Connecting to IMAP server...
> C: ESP Mail Client v1.4.2
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
IMAP server connected
> C: server connected

Check the capability...
> C: check the capability

Logging in...
> C: send imap command, LOGIN

Reading the list of mailboxes...
> C: send imap command, LIST

Available folders: Notes, Drafts, &W1hoYw-, Outbox, Junk, Inbox, Sent, Deleted

Selecting the INBOX folder...
> C: open the mailbox folder

Info of the selected folder
Total Messages: 1
Predicted next UID: 41
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent

Searching messages...
> C: search messages
> C: 100%

Search limit:5
Found 1 messages
Show 1 messages

Fetching message 1, UID: 38
> C: send imap command, FETCH
> C: fetch message header
> C: Free Heap: 231592

Logging out...
> C: send imap command, LOGOUT

Message fetch cmpleted
> C: message fetch completed
> C: cleaning SSL connection

################################
Messsage Number: 1
Messsage UID: 38
Messsage ID:  <SG2PR02MB44430233B5F997CA75C61B6CC9D99@SG2PR02MB4443.apcprd02.prod.outlook.com>
Accept Language: 
Content Language: 
From: ESP Mail<zjmeng2006@outlook.com>
From Charset: 
To: user1<zjmeng2006@outlook.com>
To Charset: 
CC: 
CC Charset: 
Date: Mon, 13 Sep 2021 08:04:52 +0000
Subject:  Test sending Email with attachments and inline images from SD card and Flash
Subject Charset:

mobizt, 2021-09-14:
The config that you posted above is not right for reading message.

Here is what wrong

```ini
/* Message UID to fetch or read e.g. 100 */
config.fetch.uid = "";

/* Set seen flag */
//config.fetch.set_seen = true;

/* Search criteria */
config.search.criteria = "UID SEARCH ALL";
```

It should be
```ini
/* Message UID to fetch or read e.g. 100 */
config.fetch.uid = "xxxxx"; //should assign the UID of message to read

/* Set seen flag */
//config.fetch.set_seen = true;

/* Search criteria */
config.search.criteria = ""; //should be empty string
```

With the latest update, you can get the latest message UID, after mailbox was selected from

```cpp
int uid = imap.getUID(imap.selectedFolder().msgCount());
```
Then after the mailbox was selected

```ini
config.fetch.uid = imap.getUID(imap.selectedFolder().msgCount()).c_str();
```

mobizt, 2021-09-14:
For the issue, I'm checking.

mobizt, 2021-09-14:
The bug is now fixed in the latest library [v1.5.1](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.1), please update.

zjmeng2006, 2021-09-14:
Thanks！I tested v1.5.1 "Read_single_Email", it show can find and download the attachment files now. 

But when I use “config.storage.type = esp_mail_file_storage_type_sd;“ and insert sd card ，run again , I can't find any files be download into sd card.

PS: I test "Send_Attachment_File” again ,I find sd card be created two files (“bin1.dat” and “orange.png” )，so sd card connect is ok。

mobizt, 2021-09-14:
You should post the debug message.

zjmeng2006, 2021-09-14:
Read_single_Email
config.storage.type = esp_mail_file_storage_type_sd;

Connecting to IMAP server...
> C: ESP Mail Client v1.5.1
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
IMAP server connected
> C: server connected

Check the capability...
> C: check the capability

Logging in...
> C: send imap command, LOGIN

Reading the list of mailboxes...
> C: send imap command, LIST

Available folders: Notes, Drafts, &W1hoYw-, Outbox, Junk, Inbox, Sent, Deleted

Selecting the INBOX folder...
> C: open the mailbox folder

Info of the selected folder
Total Messages: 2
Predicted next UID: 47
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent
Get UID...
> C: Get UID...
UID is 45
> C: UID is 45

Fetching message 1, UID: 45
> C: send imap command, FETCH
> C: fetch message header
> C: fetch body part header, 1
> C: fetch body sub part header, 1.1
> C: fetch body sub part header, 1.2
> C: fetch body sub part header, 1.2.1
> C: fetch body sub part header, 1.2.2
> C: fetch body sub part header, 1.2.3
> C: fetch body sub part header, 1.2.4
> C: fetch body sub part header, 1.3
> C: fetch body part header, 2
> C: fetch body part header, 3
> C: fetch body part header, 4

Attachments (4)
orange.png
green.png
bin1.dat
bin2.dat

Downloading messages...
> C: download plain TEXT message
download /email_data/45/msg.txt, 100%
> C: download HTML message
download /email_data/45/msg.html, 100%

Downloading attachments...
> C: download attachment
download orange.png, 100%
> C: download attachment
download green.png, 100%
> C: download attachment
download bin1.dat, 100%
> C: download attachment
download bin2.dat, 100%

Saving message header to file...
> C: Free Heap: 224900

Logging out...
> C: send imap command, LOGOUT

Message fetch cmpleted
> C: message fetch completed
> C: cleaning SSL connection

################################
Messsage Number: 1
Messsage UID: 45
Messsage ID:  <SG2PR02MB4443FD9A5AED5D53ADD996FAC9DA9@SG2PR02MB4443.apcprd02.prod.outlook.com>
Accept Language: 
Content Language: 
From: ESP Mail<zjmeng2006@outlook.com>
From Charset: 
To: user1<zjmeng2006@outlook.com>
To Charset: 
CC: 
CC Charset: 
Date: Tue, 14 Sep 2021 08:16:32 +0000
Subject:  Test sending Email with attachments and inline images from SD card and Flash
Subject Charset: 
Text Message: This message contains 2 inline images and 2 attachment files.
The inline images were not shown in the plain text message.
Text Message Charset: utf-8
Text Message Transfer Encoding: base64
HTML Message: <meta http-equiv="Content-Type" content="text/html; charset=utf-8"><s=pan style="color:#ff0000;">This message contains 2 inline images and 2 at=tachment files.</span><br><br>[image] [image]
HTML Message Charset: utf-8
HTML Message Transfer Encoding: quoted-printable

Attachment: 4 file(s)
****************************
1. Filename: orange.png, Name: orange.png, Size: 292, MIME: image/png, Type: inline, Creation Date: 
2. Filename: green.png, Name: green.png, Size: 292, MIME: image/png, Type: inline, Creation Date: 
3. Filename: bin1.dat, Name: bin1.dat, Size: 2056, MIME: application/octet-stream, Type: attachment, Creation Date: 
4. Filename: bin2.dat, Name: bin2.dat, Size: 2060, MIME: application/octet-stream, Type: attachment, Creation Date:

mobizt, 2021-09-14:
I don't know are you understand the session configuration, but this is the problem

```ini
    /* Set the storage to save the downloaded files and attachments */
    config.storage.saved_path = "/email_data";

    /** The file storage type e.g.
     * esp_mail_file_storage_type_none,
     * esp_mail_file_storage_type_flash, and 
     * esp_mail_file_storage_type_sd 
    */
    config.storage.type = esp_mail_file_storage_type_none;
```

closed 2021-09-14 by mobizt

mobizt, 2021-09-14:
You said true, still unable to save attachments to SD media in v1.5.1.

Now I fix this issue in [v1.5.2](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.2), please update.

zjmeng2006, 2021-09-15:
Thanks，v1.5.2 work!