mobizt/ESP-Mail-Client #95: How to only receive the unseen msg？

zjmeng2006, 2021-09-16:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
I test example "Read_Single_Email" （V1.5.2），and set config like this:
    config.fetch.set_seen = true;
    config.search.criteria = "";
    config.search.unseen_msg = true;
and the mail box (outlook.com) has a unseen msg email . When I run first time Read_Single_Email , it can receive the email and be set seen. But when I run second time (reset esp32), that msg（be set seen） was received again.
I want to only receive the unseen msg , I not sure is a bug or setting error... Thanks.

**Share code snippet to reproduce the issue:**
```yaml
void setup()
{

    Serial.begin(115200);

#if defined(ARDUINO_ARCH_SAMD)
    while (!Serial)
        ;
    Serial.println();
    Serial.println("**** Custom built WiFiNINA firmware need to be installed.****\nTo install firmware, read the instruction here, https://github.com/mobizt/ESP-Mail-Client#install-custom-built-wifinina-firmware");

#endif

    Serial.println();

    Serial.print("Connecting to AP");

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(200);
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    /** Enable the debug via Serial port 
     * none debug or 0
     * basic debug or 1
     * 
     * Debug port can be changed via ESP_MAIL_DEFAULT_DEBUG_PORT in ESP_Mail_FS.h
    */
    imap.debug(1);

    /* Set the callback function to get the reading results */
    imap.callback(imapCallback);

    /** In case the SD card/adapter was used for the file storagge, the SPI pins can be configure from
     * MailClient.sdBegin function which may be different for ESP32 and ESP8266
     * For ESP32, assign all of SPI pins
     * MailClient.sdBegin(14,2,15,13)
     * Which SCK = 14, MISO = 2, MOSI = 15 and SS = 13
[... 514 lines cut ...]
Messsage Number: 1
Messsage UID: 47
Messsage ID:  <SG2PR02MB444338D44C48090AC491438CC9DB9@SG2PR02MB4443.apcprd02.prod.outlook.com>
Accept Language: 
Content Language: 
From: ESP Mail<zjmeng2006@outlook.com>
From Charset: 
To: user1<zjmeng2006@outlook.com>
To Charset: 
CC: 
CC Charset: 
Date: Wed, 15 Sep 2021 01:37:15 +0000
Subject:  Test sending Email with parallel attachments
Subject Charset: 
Text Message: This message contains image and audio file which will play on the Mail client in parallel or simultaneously (depends on the client supports).
Text Message Charset: us-ascii
Text Message Transfer Encoding: base64
HTML Message: <meta http-equiv="Content-Type" content="text/html; charset=utf-8"><s=pan style="color:#ff0000;">This message contains image and audio file whi=ch will play on the Mail client in parallel or simultaneously (depends on t=he client supports).
HTML Message Charset: utf-8
HTML Message Transfer Encoding: quoted-printable

Attachment: 2 file(s)
****************************
1. Filename: haun.png, Name: haun.png, Size: 7235, MIME: image/png, Type: attachment, Creation Date: 
2. Filename: mu_law.wav, Name: mu_law.wav, Size: 23966, MIME: audio/basic, Type: attachment, Creation Date:

mobizt, 2021-09-16:
The library aims to not alter or modify user data when reading the Email since the library created.

The library uses IMAP's **EXAMINE** command to open the mailbox (in read only mode), then **\Seen** flag will not set to the fetched message.

This is normal behavior of  the most mail servers e.g. Gmail, Outlook that response to **EXAMINE** command and complied to rfc standard unless iCloud which I'm not sure as I see the iCloud response to **EXAMINE** command as READ-WRITE instead of READ-ONLY.

In case of mailbox and message modifications (rename, copy, move, expunge), the mailbox will open using **SELECT** command for READ-WRITE access.

In most mail server with this library, when you open the message to read, the message is still in unread status, you can set the \Seen flag to this message, see [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/IMAP/Set_Flags/Set_Flags.ino) to set its status as read.

To search unread messages, just set search criteria as

`UID SEARCH UNSEEN` for messages that do not have the \Seen flag set.

or

`UID SEARCH NEW` which is similar to `UID SEARCH RECENT UNSEEN` for messages that have the \Recent flag set but not the \Seen flag.

zjmeng2006, 2021-09-16:
Thanks! I try to add "UID SEARCH UNSEEN" and set the \Seen flag(UID=47) and add loop to read mail box , but still receive  the unseen msg(UID=47) every time.

**==========CODE============**
/* Setup the configuration for searching or fetching operation and its result */
    //IMAP_Config config;

    /* Set seen flag */
    config.fetch.set_seen = true;

    /* Search criteria */
    config.search.criteria = "UID SEARCH UNSEEN";

    /* Also search the unseen message */
    config.search.unseen_msg = true;

    /* Set the storage to save the downloaded files and attachments */
    config.storage.saved_path = "/email_data";

    /** The file storage type e.g.
     * esp_mail_file_storage_type_none,
     * esp_mail_file_storage_type_flash, and 
     * esp_mail_file_storage_type_sd 
    */
    config.storage.type = esp_mail_file_storage_type_sd;

    /** Set to download heades, text and html messaeges, 
     * attachments and inline images respectively.
    */
    config.download.header = true;
    config.download.text = true;
    config.download.html = true;
    config.download.attachment = true;
    config.download.inlineImg = true;

    /** Set to enable the results i.e. html and text messaeges 
     * which the content stored in the IMAPSession object is limited
     * by the option config.limit.msg_size.
     * The whole message can be download through config.download.text
     * or config.download.html which not depends on these enable options.
    */
    config.enable.html = true;
    config.enable.text = true;

    /* Set to enable the sort the result by message UID in the ascending order */
    config.enable.recent_sort = true;

    /* Set to report the download progress via the default serial port */
    config.enable.download_status = true;

    /* Header fields parsing is case insensitive by default to avoid uppercase header in some server e.g. iCloud
    , to allow case sensitive parse, uncomment below line*/
    //config.enable.header_case_sesitive = true;

    /* Set the limit of number of messages in the search results */
    config.limit.search = 5;

    /** Set the maximum size of message stored in 
     * IMAPSession object in byte
    */
    config.limit.msg_size = 512;

    /** Set the maximum attachments and inline images files size
     * that can be downloaded in byte. 
     * The file which its size is largger than this limit may be saved 
     * as truncated file.
    */
    config.limit.attachment_size = 1024 * 1024 * 5;

    /* Connect to server with the session and config */
    if (!imap.connect(&session, &config))
[... 345 lines cut ...]
Messsage Number: 2
Messsage UID: 47
Messsage ID:  <SG2PR02MB444338D44C48090AC491438CC9DB9@SG2PR02MB4443.apcprd02.prod.outlook.com>
Accept Language: 
Content Language: 
From: ESP Mail<zjmeng2006@outlook.com>
From Charset: 
To: user1<zjmeng2006@outlook.com>
To Charset: 
CC: 
CC Charset: 
Date: Wed, 15 Sep 2021 01:37:15 +0000
Subject:  Test sending Email with parallel attachments
Subject Charset: 
Text Message: This message contains image and audio file which will play on the Mail client in parallel or simultaneously (depends on the client supports).
Text Message Charset: us-ascii
Text Message Transfer Encoding: base64
HTML Message: <meta http-equiv="Content-Type" content="text/html; charset=utf-8"><s=pan style="color:#ff0000;">This message contains image and audio file whi=ch will play on the Mail client in parallel or simultaneously (depends on t=he client supports).
HTML Message Charset: utf-8
HTML Message Transfer Encoding: quoted-printable

Attachment: 2 file(s)
****************************
1. Filename: haun.png, Name: haun.png, Size: 7235, MIME: image/png, Type: attachment, Creation Date: 
2. Filename: mu_law.wav, Name: mu_law.wav, Size: 23966, MIME: audio/basic, Type: attachment, Creation Date:

mobizt, 2021-09-16:
I extensively test and did not found the issue.

Please make sure, you open the right mailbox for search as the mailbox (folder) that you set the \Seen flag.

I recommend to update the library to latest [v1.5.3](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.3) to use the feature I mention below.

If you want to get the flags for any message, use **imap.getFlags(xxx)** where xxx is the message number or order of message stored in a mailbox, not a message UID.

```cpp
String msgFlags = imap.getFlag(1);
```

And the printing function updated in the new version examples

https://github.com/mobizt/ESP-Mail-Client/blob/fdc2807c9ffe57bd02f53674e0e840aa23f94741/examples/IMAP/Search_Emails/Search_Emails.ino#L337-L409

You can set read to message in other app e.g. mail app in Windows 10 by right click and select **Mark as read** or **Mark as unread**, and see the changes in the search result.

mobizt, 2021-09-16:
If you still get the issue of searching result, this is the internal implementation issue of your working mail server which you can try or prove with other mail server or create new folder (mailbox) and test it with app to set the read status or other flags.

closed 2021-09-16 by mobizt

zjmeng2006, 2021-09-17:
Thanks! But I test v1.5.3 and add "String msgFlags = imap.getFlag(1);" ，then display compile error：
exit status 1
'class IMAPSession' has no member named 'getFlag'

**==========test v1.5.3 , but no add  "String msgFlags = imap.getFlag(1);" debug info：===========**
Connecting to IMAP server...
> C: ESP Mail Client v1.5.3
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
IMAP server connected
> C: server connected

Check the capability...
> C: check the capability

Logging in...
> C: send imap command, LOGIN

Reading the list of mailboxes...
> C: send imap command, LIST

Available folders: Notes, Drafts, &W1hoYw-, Outbox, Junk, Inbox, Sent, Deleted

Selecting the Inbox folder...
> C: open the mailbox folder

Info of the selected folder
Total Messages: 5
Predicted next UID: 57
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent
Get UID...
> C: get UID...
UID is 55
> C: UID is 55

Searching messages...
> C: search messages

Search limit:5
No message found for search criteria
Logging out...
> C: send imap command, LOGOUT

Message fetch cmpleted
> C: message fetch completed
> C: cleaning SSL connection

mobizt, 2021-09-17:
It's getFlags instead of getFlag, you can check the [document](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#get-message-flags-in-selected-or-opened-mailbox).

zjmeng2006, 2021-09-17:
OK，compile pass ,but debug info display "Error, connection closed
> E: connection closed" .

**============CODE==============**
void printMessages(std::vector<IMAP_MSG_Item> &msgItems, bool headerOnly)
{

    for (size_t i = 0; i < msgItems.size(); i++)
    {

        /* Iterate to get each message data through the message item data */
        IMAP_MSG_Item msg = msgItems[i];

        Serial.println("****************************");
        ESP_MAIL_PRINTF("Number: %d\n", msg.msgNo);
        
        ESP_MAIL_PRINTF("UID: %d\n", msg.UID);
        ESP_MAIL_PRINTF("Messsage-ID: %s\n", msg.ID);

        ESP_MAIL_PRINTF("Flags: %s\n", msg.flags);

        **ESP_MAIL_PRINTF("GetFlags: %s\n", imap.getFlags(msg.msgNo));**

**=============Debug Info===============**
Connecting to IMAP server...
> C: ESP Mail Client v1.5.3
> C: connect to IMAP server
> C: host > outlook.office365.com
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
IMAP server connected
> C: server connected

Check the capability...
> C: check the capability

Logging in...
> C: send imap command, LOGIN

Reading the list of mailboxes...
> C: send imap command, LIST

Available folders: Notes, Drafts, &W1hoYw-, Outbox, Junk, Inbox, Sent, Deleted

Selecting the Inbox folder...
> C: open the mailbox folder

Info of the selected folder
Total Messages: 3
Predicted next UID: 57
Flags: \Seen, \Answered, \Flagged, \Deleted, \Draft, $MDNSent
Get UID...
> C: get UID...
UID is 47
> C: UID is 47

Searching messages...
> C: search messages
> C: 100%

Search limit:5
Found 1 messages
Show 1 messages

Fetching message 1, UID: 47
> C: send imap command, FETCH
> C: fetch message header
Get Flags...
> C: get flags...
> C: Free Heap: 231452
Logging out...
> C: send imap command, LOGOUT

Message fetch cmpleted
> C: message fetch completed
> C: cleaning SSL connection

****************************
Number: 3
UID: 47
Messsage-ID: <SG2PR02MB444338D44C48090AC491438CC9DB9@SG2PR02MB4443.apcprd02.prod.outlook.com>
**Flags: 
Get Flags...
> C: get flags...
Error, connection closed
> E: connection closed
GetFlags:**
Attachment: yes
From: ESP Mail<zjmeng2006@outlook.com>
To: user1<zjmeng2006@outlook.com>
Date: Wed, 15 Sep 2021 01:37:15 +0000
Subject: Test sending Email with parallel attachments
Return-Path: zjmeng2006@outlook.com

mobizt, 2021-09-17:
You should making understand the objective of the getFlags function.

The getFlags returns flags string of message in the selected or open mailbox.

What do you expected from that function with no mailbox selected or open and no message to read?

In the examples of IMAP, you don't need this getFlags function as it is already returns the flags in the header information of each message item search/fetch info as you see in the printMessages function

mobizt, 2021-09-17:
To use getFlages, you must log in, open or select mailbox, choose the message no to get flags from.

mobizt, 2021-09-17:
You should use other mail provider as this is the mail server issue instead of library.

All data returned are from server directly.  Mail IMAP server may comply to rfc standards but incomplete.

mobizt, 2021-09-17:
To make you understand why I said **"What do you expected from that function with no mailbox selected or open and no message to read?"**

The error is because you're trying to get flags while session was closed  (no mailbox open or logged out already) since you call [readMail](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#reading-email-through-imap-server) which the second argument of this function is for closing the session which is true by default.

To read mail and keep the session open (stay logged in), just assign the second argument to false like this.

`MailClient.readMail(&imap, false); `

You can read email again and again without to re-log in or sign-in 

You can read, search, copy, delete message without error as you are authenticated and the mailbox is already open.

If the session close, then mailbox closed and log out, of course functions that require the mailbox access will return error.

**Why I set this second argument of readMail to true by default?** 

This is because the user may use this read mail or send mail function few time then closing the session every time will release the memory after used automatically.