mobizt/ESP-Mail-Client #96: "Outside Heap area error" when I give command to send email again through POST request from Webserver

pedros89, 2021-09-22:
**Build tool used:**

- [ X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

**Description of problem: **
I made an HTML page hosted on ESP32 Webserver. On the HTML page there are 5 inputs fields generated dinamically where the user fills in 5 email. When user clicks on submit button a POST request is made to the server and with EEPROM library I save the email on ESP32 memory and I use ESP-Mail-Client to send a welcome mail to all emails. I made a function to send the emails. I have used the `Send_Text.ino` example. **But it work only once.** If on the client side they click the button again to send another email there is heap error. If I reset the ESP32 it works but not two times or more in a row. See image below for the interface I have circled the button that submit the post request.
[image]

The first time I press the button the email is sent but the second time I press it I get this error. I am sure that is not related to EEPROM saving email as that prints out ok. 

```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "IDF/components/heap/heap_caps.c", line 304, function: heap_caps_free
abort() was called at PC 0x401956c3 on core 1 
```

**Share code snippet to reproduce the issue:**
```yaml

//function that give true if email was sent correctly or false if not. I pass to the function two strings: emailSubject and emailText
//I copied this funciton from your example
bool sendEmail(String emailSubject, String emailText){

  smtp.debug(1);   //put to 0 to remove email debug messages
  smtp.callback(smtpCallback);
  
  ESP_Mail_Session session;
 
  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = "example.com"; 
  
  getEmail(); //with this function I get the email1, email2 etc. from EEPROM
  
  SMTP_Message message;
  message.sender.name = "Name";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = emailSubject.c_str();     //it needs to be a constant char so I convert it to c_str()
  if (email1!=""){message.addRecipient("user1", email1.c_str());}
  if (email2!=""){message.addRecipient("user1", email2.c_str());}
  if (email3!=""){message.addRecipient("user1", email3.c_str());}
  if (email4!=""){message.addRecipient("user1", email4.c_str());}
  if (email5!=""){message.addRecipient("user1", email5.c_str());}
  message.addBcc(AUTHOR_EMAIL); 
  message.text.content = emailText.c_str();
  
  message.text.charSet = "us-ascii";
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_high;

  session.certificate.cert_data = rootCACert;
  session.certificate.verify = true;
  if (!smtp.connect(&session)){    
    return false;
  }
  if (!MailClient.sendMail(&smtp, &message)){
    Serial.println("Error sending Email, " + smtp.errorReason());  
    return false;
  } else {
 
    Serial.println("FREE HEAP");
    Serial.print(MailClient.getFreeHeap());
    Serial.print("Bytes");

    return true;
  }
}

//also this funtion is copied
void smtpCallback(SMTP_Status status)
{
  /* Print the current status */
  Serial.println(status.info());

  /* Print the sending result */
  if (status.success())
  {
    Serial.println("----------------");
    ESP_MAIL_PRINTF("Message sent success: %d\n", status.completedCount());
    ESP_MAIL_PRINTF("Message sent failled: %d\n", status.failedCount());
    Serial.println("----------------\n");
    struct tm dt;

    for (size_t i = 0; i < smtp.sendingResult.size(); i++)
    {
      /* Get the result item */
      SMTP_Result result = smtp.sendingResult.getItem(i);
      time_t ts = (time_t)result.timestamp;
      localtime_r(&ts, &dt);

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %d/%d/%d %d:%d:%d\n", dt.tm_year + 1900, dt.tm_mon + 1, dt.tm_mday, dt.tm_hour, dt.tm_min, dt.tm_sec);
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients);
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject);
    }
    Serial.println("----------------\n"); 
    smtp.sendingResult.clear();    //I have added this line to clear results 
  }
}

void handleRequest(){
if (server.hasArg("email1")||server.hasArg("email2")||server.hasArg("email3")||server.hasArg("email4")||server.hasArg("email5")) {
    Serial.println("email request arrived");
    handleEmail();
  }
}

//this funciton save email in EEPROM and then send email
void handleEmail(){
  for(int i=1; i<6; i++){  
  Serial.println("received email by request");
  Serial.println(server.arg("email"+String(i)+"")); //if the POST value is empty it will receive an empty string
  String email0=server.arg("email"+String(i)+"");
  wrStrToMem(email0, 65, (75+i+((i-1)*65)));    //this function pass to the EEPROM the email saved in the right addresses
  }

//if wifi is connected and email is send successful respond with success message to the client
if((checkConnection()==true)&&(sendEmail(emailSubject, emailText)==true)){
    String response_success="all good, email sent, if you have not received it please make sure it is not in the spam folder or that you typed the address correctly";
    server.send(200, "text/html", response_success);
}

```

**Additional information and things you've tried:**

I have added `smtp.sendingResult.clear();  `
but nothing changed

I beleive I need to cancel / flush some parameters that saturate my heap.
I tried adding 
`smtp.closeSession();`
`message.clear();`
I also wanted to use the `empty()`  method but as `MailClient.empty();`  or  `smtp.empty();` but it didn't work

I start to think that there is incompability between `#include <WebServer.h>`  and   `#include <ESP_Mail_Client.h>`

mobizt, 2021-09-22:
This is not library issue unless your declare local variable e.g. session and also smtp (I did not see in your post) and it should existed for use in call back function also.

Since the session and smtp objects pass to MailClient.sendMail as pointer, you should define them global.

mobizt, 2021-09-22:
Send email in loop is ok.

This is the examples that used IMAP and SMTP session objects, both objects defined global, sending the email in loop at some interval with sending result callback. The IMAP session object used for polling the message in selected mailbox for incoming message.
https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Reply_Text/Reply_Text.ino

mobizt, 2021-09-22:
The library was intensively test for all usage conditions since the older ESP32Mail-Client library.

pedros89, 2021-09-22:
This parameters are defined globally and 

```
SMTPSession smtp;
void smtpCallback(SMTP_Status status); //function callback
```

and this param are defined locally
smtp.debug(1);   //put to 0 to remove email debug messages
  smtp.callback(smtpCallback);
  
  ```
ESP_Mail_Session session;
 
  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = "teis.it"; 
```

Should I define everything globally?

mobizt, 2021-09-22:
If that object used as pointer to object in any function, you should maintain its life for usage scope, this is the fundamental.

As you see, in examples, loop is empty then these object only used in setup.

mobizt, 2021-09-22:
Define the ESP_Mail_Session local or global is ok.

This example shows sending email in loop with session reused.

```cpp
SMTPSession smtp;

void smtpCallback(SMTP_Status status);

unsigned long sendMillis = 0;

int count = 0;

ESP_Mail_Session session;

void setup()
{

  Serial.begin(115200);

  Serial.println();

  Serial.print("Connecting to AP");

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED)
  {
    Serial.print(".");
    delay(200);
  }

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  smtp.debug(1);
  smtp.callback(smtpCallback);

  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = "mydomain.net";

  if (!smtp.connect(&session))
    return;
}

void sendMail(bool closeSession)
{
  
  SMTP_Message message;

  message.sender.name = "ESP Mail";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = "Test sending plain text Email";
  message.addRecipient("Someone", "####@#####_dot_com");

  String textMsg = "This is simple plain text message";
  message.text.content = textMsg.c_str();

  message.text.charSet = "us-ascii";

  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;

  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;

  message.addHeader("Message-ID: <abcde.fghij@gmail.com>");

  

  if (!MailClient.sendMail(&smtp, &message, closeSession))
    Serial.println("Error sending Email, " + smtp.errorReason());

  ESP_MAIL_PRINTF("Free Heap: %d\n", MailClient.getFreeHeap());
}

void loop()
{

  if (millis() - sendMillis > 20 * 1000 || sendMillis == 0)
  {
    sendMillis = millis();

    count++;

    if (count % 10 == 0 && count > 1)
      sendMail(true);
    else
      sendMail(false);
  }
}

void smtpCallback(SMTP_Status status)
{

  Serial.println(status.info());

  if (status.success())
  {
    Serial.println("----------------");
    ESP_MAIL_PRINTF("Message sent success: %d\n", status.completedCount());
    ESP_MAIL_PRINTF("Message sent failled: %d\n", status.failedCount());
    Serial.println("----------------\n");
    struct tm dt;

    for (size_t i = 0; i < smtp.sendingResult.size(); i++)
    {

      SMTP_Result result = smtp.sendingResult.getItem(i);
      time_t ts = (time_t)result.timestamp;
      localtime_r(&ts, &dt);

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %d/%d/%d %d:%d:%d\n", dt.tm_year + 1900, dt.tm_mon + 1, dt.tm_mday, dt.tm_hour, dt.tm_min, dt.tm_sec);
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients);
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject);
    }
    Serial.println("----------------\n");

    smtp.sendingResult.clear();
  }
}

```

mobizt, 2021-09-22:
If you run my above code and the free heap is ok. No matter you define session as local or global or you close the session every time after sending.

pedros89, 2021-09-22:
Thank you [mobizt](https://github.com/mobizt). I will give it a try. Becasue I have just tried to define globally:
```
ESP_Mail_Session session;
SMTP_Message message;
```
But still I have the same problem when I send the fist email is sent out ok, but when I call that funciton sendEmail() again I still have the heap error.

mobizt, 2021-09-22:
There is no compatibilities issue between WebServer.h and the library as it used customized version of WiFiClientSecure.h.

If you think this is the compatibilities issue, replace the email code with simple https request code using WiFiClientSecure to test (to get some server response).

If you still get the issue of low free heap, you should debug it first to make sure this is not your code memory allocation issue., and create the clear, clean and minimum example code that can reproduce the issue and open the issue at esp32 repo.

mobizt, 2021-09-22:
You should use esp exception decoder Arduino add on to decode the traces to check where is the last line of successful execution.

It is more helpful to see where your code stop. The assertion error which raised within the code for checking something can't tell anything correctly without checking the last successful code before it assertion error.

pedros89, 2021-09-22:
I really appreciate your help. I have tried to adapt your code you posted above to mine, but the problem this time is that when the ESP32 is conneced to the SMTP server, my Webserver is not responding and giving to the client and empty white page. It has been an issue that I am not able to sort out and I have decided for a turnaroud because I am running on my ESP32 a quite heavy server (with icons, images, bootstrap).
If you think that there is no compatibility issue maybe you can try to include in the library an example where the user has a simple webserver with an HTML input field and the user insert the email and submit it to the ESP and the ESP send out the email every time. This is basically what I was trying to do.
I will make a POST request to a PHP script located at [www.mywebsite.com/sendEmail.php](www.mywebsite.com/sendEmail.php) to do the job.
This is the code I will use, the PHP script will receive the POST and send out the email as easy as that.

```
#include <HTTPClient.h>
#include <WiFiClientSecure.h>

String email="myemail@hello.com", emailObj="test email", emailText="some random text";
void sendEmailPhp(String email, String emailObj, String emailText)
  { 
  WiFiClientSecure *client = new WiFiClientSecure;
  if(client) {
    client -> setCACert(rootCACert);
    { // Add a scoping block for HTTPClient https to make sure it is destroyed before WiFiClientSecure *client is 
    HTTPClient https; 
    Serial.print("[HTTPS] begin...\n");
      if (https.begin(*client, "https://www.mywebsite.com/sendEmail.php")) {       
        /*start POST request*/
        https.addHeader("Content-Type", "application/x-www-form-urlencoded");       //Specify content-type header, remember that for GET is "text/plain"       
        int httpResponseCode = https.POST("mail="+email+"&obj="+emailObj+"&text="+emailText+"");      //Send the actual POST request        
        if(httpResponseCode>0){
           String response = https.getString();     //Get the response to the request
           Serial.println(httpResponseCode);        //Print return code
           Serial.println(response);                //Print request answer
         }
         else{
            Serial.print("Error on sending POST: ");
            Serial.println(httpResponseCode);         
         }
        /*end POST request*/        
        https.end();     
      } 
      else {
        Serial.printf("[HTTPS] Unable to connect\n");
      }
    } // End extra scoping block
    delete client;
  } 
  else {
    Serial.println("Unable to create client");
  } 
}
```

mobizt, 2021-09-22:
I mean using Webserver together with WiFiClientSecure instead.

You should debug and use esp exception decoder to decode the stack traces when you integrate the libraries and found some issue.

The assertion error raised by the assertion statement by programmer to checking something, not the exception raised by system or cpu then you don't know why dev used this statement but there is something that is not ok as he expected.

Then you need to debug by yourself first, to see your code flow. Use stack trace decoder in case of error to see where it stopped.

This is importance as to ensure it stop while execute email code or web server code or somewhere inside the core library.

closed 2021-09-22 by mobizt

mobizt, 2021-09-22:
For the webserver, you can use send_P in case web page data stored in PROGMEM or use streamFile in case data stored in file system (SD or SPIFFS or LittleFS).

https://github.com/espressif/arduino-esp32/blob/317be68cefe4678e49f712e263da8064b75c7d68/libraries/WebServer/src/WebServer.h#L123-L124

https://github.com/espressif/arduino-esp32/blob/317be68cefe4678e49f712e263da8064b75c7d68/libraries/WebServer/src/WebServer.h#L140

With send_P and streamFile, the memory required for data buffer was much smaller instead of buffer the whole content as it send as a chunked encoded data if client sent HTTP/1.1 request.

mobizt, 2021-09-23:
This is the issue in your other parts of code about trying to access or use the memory (heap) of some variable that disposed or the array in the index that out of bound.

As you see the error that at that point, it required the **pointer to some memory address which is not zero or NULL** or code is **trying to call free for deallocate the disposed object**.

I'm sure this is the error in your sketch for memory access or array index addressing issue.