mobizt/ESP-Mail-Client #111: Build error when Debug Level is set to SSL+HTTP_CLIENT

adrcunha, 2021-11-06:
**Build tool used:**

- [X] Arduino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [X] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

None for the minimum sketch.

**Description of problem:**

Build error when Debug Level is set to SSL+HTTP_CLIENT.

```
arduino/libraries/ESP_Mail_Client/src/wcs/esp8266/ESP8266_WCS.cpp: In member function 'virtual int ESP8266_WCS::connect(const char*, uint16_t)':
arduino/libraries/ESP_Mail_Client/src/wcs/esp8266/ESP8266_WCS.cpp:59:30: error: 'DEBUG_ESP_PORT' was not declared in this scope; did you mean 'DEBUG_ESP_SSL'?
   59 | #define DEBUG_BSSL(fmt, ...) DEBUG_ESP_PORT.printf_P((PGM_P)PSTR("BSSL:" fmt), ##__VA_ARGS__)
      |                              ^~~~~~~~~~~~~~
arduino/libraries/ESP_Mail_Client/src/wcs/esp8266/ESP8266_WCS.cpp:79:5: note: in expansion of macro 'DEBUG_BSSL'
   79 |     DEBUG_BSSL("connect: Name loopup failure\n");
      |     ^~~~~~~~~~
arduino/libraries/ESP_Mail_Client/src/wcs/esp8266/ESP8266_WCS.cpp:59:30: error: 'DEBUG_ESP_PORT' was not declared in this scope; did you mean 'DEBUG_ESP_SSL'?
   59 | #define DEBUG_BSSL(fmt, ...) DEBUG_ESP_PORT.printf_P((PGM_P)PSTR("BSSL:" fmt), ##__VA_ARGS__)
      |                              ^~~~~~~~~~~~~~
arduino/libraries/ESP_Mail_Client/src/wcs/esp8266/ESP8266_WCS.cpp:85:5: note: in expansion of macro 'DEBUG_BSSL'
   85 |     DEBUG_BSSL("connect: Unable to connect TCP socket\n");
      |     ^~~~~~~~~~
```

**Code snippet to reproduce the issue:**
```yaml
#include <ESP_Mail_Client.h>
void setup() {}
void loop() {}
```

**Additional information and things you've tried:**
N/A

mobizt, 2021-11-07:
Thanks, this is already known.

This is the user responsibility as the debug port should select in the IDE when they want to debug.

closed 2021-11-07 by mobizt

mobizt, 2021-11-07:
I provide the better commit https://github.com/mobizt/ESP-Mail-Client/commit/379c793e5e76871bc9a7d332c8083a63ba1aa9a5