mobizt/ESP-Mail-Client #114: Deleting Messages does no longer work

dkalliv, 2021-11-07:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
After updating from library version 1.5.2 to 1.5.5 I noticed, that I can no longer delete messages. In order to ensure the problem was not in my code I reproduced the problem with a slightly adapted version of the Delete_Messages.ino sample sketch (see below).

**Share code snippet to reproduce the issue:**
```yaml
/**
 * This example showed how to delete messages from the opened mailbox folder.
 * 
 * Email: suwatchai@outlook.com
 * 
 * Github: https://github.com/mobizt/ESP-Mail-Client
 * 
 * Copyright (c) 2021 mobizt
 *
*/

/** To receive Email using Gmail, IMAP option should be enabled. https://support.google.com/mail/answer/7126229?hl=en
 * and also https://accounts.google.com/b/0/DisplayUnlockCaptcha
 * 
*/

/** For ESP8266, with BearSSL WiFi Client 
 * The memory reserved for completed valid SSL response from IMAP is 16 kbytes which
 * may cause your device out of memory reset in case the memory 
 * allocation error.
*/

#include <Arduino.h>
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <ESP_Mail_Client.h>

//To use only IMAP functions, you can exclude the SMTP from compilation, see ESP_Mail_FS.h.

#define WIFI_SSID "xxxxx"
#define WIFI_PASSWORD "xxxxxxxxx"

/* The imap host name e.g. imap.gmail.com for GMail or outlook.office365.com for Outlook */
#define IMAP_HOST "xxxx.xxxx.xxxx.xxx"

/** The imap port e.g. 
 * 143  or esp_mail_imap_port_143
 * 993 or esp_mail_imap_port_993
*/
#define IMAP_PORT 993

/* The log in credentials */
#define AUTHOR_EMAIL "xx@xxxxxxxx.xx"
#define AUTHOR_PASSWORD "xxxxxxxxxx"

/* Print the list of mailbox folders */
void printAllMailboxesInfo(IMAPSession &imap);

/* Print the selected folder info */
void printSelectedMailboxInfo(IMAPSession &imap);

/* The IMAP Session object used for Email reading */
IMAPSession imap;

void setup()
{

    Serial.begin(115200);

#if defined(ARDUINO_ARCH_SAMD)
    while (!Serial)
        ;
    Serial.println();
    Serial.println("**** Custom built WiFiNINA firmware need to be installed.****\nTo install firmware, read the instruction here, https://github.com/mobizt/ESP-Mail-Client#install-custom-built-wifinina-firmware");

#endif

    Serial.println();

    Serial.print("Connecting to AP");

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(200);
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    /** Enable the debug via Serial port 
     * none debug or 0
     * basic debug or 1
     * 
     * Debug port can be changed via ESP_MAIL_DEFAULT_DEBUG_PORT in ESP_Mail_FS.h
    */
    imap.debug(333);

    /* Declare the session config data */
    ESP_Mail_Session session;

    /* Set the session config */
    session.server.host_name = IMAP_HOST;
    session.server.port = IMAP_PORT;
    session.login.email = AUTHOR_EMAIL;
    session.login.password = AUTHOR_PASSWORD;

    /* Setup the configuration for searching or fetching operation and its result */
    IMAP_Config config;

    /* Connect to server with the session and config */
    if (!imap.connect(&session, &config))
        return;

    /*  {Optional} */
    printAllMailboxesInfo(imap);

    /* Open or select the mailbox folder to read or search the message */
    if (!imap.selectFolder("INBOX",false))
        return;

    /* Define the MessageList class to add the message to delete */
    MessageList toDelete;

   /* Add message uid to delete to the list */

    toDelete.add(136);
    /* Delete all messages permanently by assign the second param to true*/
    imap.deleteMessages(&toDelete, true);

    ESP_MAIL_PRINTF("Free Heap: %d\n", MailClient.getFreeHeap());
}

void loop()
{
}

void printAllMailboxesInfo(IMAPSession &imap)
{
    /* Declare the folder collection class to get the list of mailbox folders */
    FoldersCollection folders;

    /* Get the mailbox folders */
    if (imap.getFolders(folders))
    {
        for (size_t i = 0; i < folders.size(); i++)
        {
            /* Iterate each folder info using the  folder info item data */
            FolderInfo folderInfo = folders.info(i);
            ESP_MAIL_PRINTF("%s%s%s", i == 0 ? "\nAvailable folders: " : ", ", folderInfo.name, i == folders.size() - 1 ? "\n" : "");
        }
    }
}

```

And here is the serial output:
```yaml

Connecting to AP...........
WiFi connected.
IP address: 
xxx.xxx.xxx.xxx

> C: ESP Mail Client v1.5.5
> C: connect to IMAP server
> C: host > imap.swizzonic-mail.ch
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
> C: server connected
> C: check the capability
$ CAPABILITY
* OK swizzonic-mail.ch CommuniGate Pro IMAP Server 7.1.0 ready
* CAPABILITY IMAP4 IMAP4REV1 ACL NAMESPACE UIDPLUS IDLE LITERAL+ QUOTA ID MULTIAPPEND LISTEXT CHILDREN BINARY ESEARCH LOGIN-REFERRALS UNSELECT SASL-IR LIST-EXTENDED SPECIAL-USE CREATE-SPECIAL-USE XLIST AUTH=LOGIN AUTH=PLAIN
$ OK completed
> C: send imap command, LOGIN
$ LOGIN xxx@xxxxxxx.xx xxxxxxxxxxxxx
$ OK completed
> C: send imap command, LIST
$ LIST "" *
* LIST (\HasNoChildren \UnMarked) "/" INBOX
* LIST (\HasNoChildren \UnMarked) "/" Trash
$ OK completed

Available folders: INBOX, Trash
> C: open the mailbox folder
$ SELECT "INBOX"
* FLAGS (\Answered \Flagged \Deleted \Seen \Draft $MDNSent $Hidden $Media $Forwarded Junk $Label1 $Label2 $Label3)
* OK [PERMANENTFLAGS (\Answered \Flagged \Deleted \Seen \Draft $MDNSent $Hidden $Media $Forwarded Junk $Label1 $Label2 $Label3)] limited
* 4 EXISTS
* 0 RECENT
* OK [UIDNEXT 137] predicted next UID
* OK [UIDVALIDITY 402276210] UIDs valid
* OK [UNSEEN 2] message 2 is first unseen
$ OK [READ-WRITE] SELECT completed
> C: open the mailbox folder
$ SELECT ""
$ NO mailbox does not exist
> E: mailbox does not exist
Free Heap: 174351
```

**Additional information and things you've tried:**
Deleting messages still works in version 1.5.3. I assume the problem is related to the introduction of MBSTRING. After some investigation I noticed that _currentFolder seems to be empty when IMAPSession::deleteMessages is called. Deleting messages seems to work, after changing _currentFolder to std::string in ESP_Mail_Client.h (line 2834).

mobizt, 2021-11-07:
Thanks for reporting this issue.

I will check and update soon.

mobizt, 2021-11-08:
The MB_String is ok, the error is due to the folder closed in case of different selection mode and the current folder name will be cleared, then any folder name const chars that refers to its own will be empty too.

The issues were fixed in the [v1.5.6](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.6). Please update.

closed 2021-11-08 by mobizt

dkalliv, 2021-11-08:
Works great now, thank you!