mobizt/ESP-Mail-Client #116: SelectedFolderInfo is no longer updated with every call of selectFolder

dkalliv, 2021-11-10:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
In library version 1.5.6 SelectedFolderInfo is no longer updated with every call of selectFolder (this was working with version 1.5.5). In the example below msgCount() and nextUID() remain the same, even though, a new message was received after the first iteration.

**Share code snippet to reproduce the issue:**
```yaml
/**
 * This example will search all Emails in the opened mailbox folder.
 * 
 * Created by K. Suwatchai (Mobizt)
 * 
 * Email: suwatchai@outlook.com
 * 
 * Github: https://github.com/mobizt/ESP-Mail-Client
 * 
 * Copyright (c) 2021 mobizt
 *
*/

/** To receive Email using Gmail, IMAP option should be enabled. https://support.google.com/mail/answer/7126229?hl=en
 * and also https://accounts.google.com/b/0/DisplayUnlockCaptcha
 * 
*/

/** For ESP8266, with BearSSL WiFi Client 
 * The memory reserved for completed valid SSL response from IMAP is 16 kbytes which
 * may cause your device out of memory reset in case the memory 
 * allocation error.
*/

#include <Arduino.h>
#if defined(ESP32)
#include <WiFi.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#endif
#include <ESP_Mail_Client.h>

//To use only IMAP functions, you can exclude the SMTP from compilation, see ESP_Mail_FS.h.

#define WIFI_SSID "xxxxxx"
#define WIFI_PASSWORD "xxxxxxxxx"

/* The imap host name e.g. imap.gmail.com for GMail or outlook.office365.com for Outlook */
#define IMAP_HOST "xxxxx.xxxxx.xxxx.xxx"

/** The imap port e.g. 
 * 143  or esp_mail_imap_port_143
 * 993 or esp_mail_imap_port_993
*/
#define IMAP_PORT 993

/* The log in credentials */
#define AUTHOR_EMAIL "xxx@xxxxxxxxx.xx"
#define AUTHOR_PASSWORD "xxxxxxxxx"

/* Print the selected folder info */
void printSelectedMailboxInfo(SelectedFolderInfo sFolder);

/* The IMAP Session object used for Email reading */
IMAPSession imap;

void setup()
{

    Serial.begin(115200);

#if defined(ARDUINO_ARCH_SAMD)
    while (!Serial)
        ;
    Serial.println();
    Serial.println("**** Custom built WiFiNINA firmware need to be installed.****\nTo install firmware, read the instruction here, https://github.com/mobizt/ESP-Mail-Client#install-custom-built-wifinina-firmware");

#endif

    Serial.println();

    Serial.print("Connecting to AP");

    WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(200);
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    Serial.println();

    imap.debug(333);

    /* Declare the session config data */
    ESP_Mail_Session session;

    /* Set the session config */
    session.server.host_name = IMAP_HOST;
    session.server.port = IMAP_PORT;
    session.login.email = AUTHOR_EMAIL;
    session.login.password = AUTHOR_PASSWORD;

    /* Setup the configuration for searching or fetching operation and its result */
    IMAP_Config config;

    /* Message UID to fetch or read */
    config.fetch.uid = "";

    config.search.criteria = "UID SEARCH ALL";

    /* Also search the unseen message */
    config.search.unseen_msg = true;

    /* Set the storage to save the downloaded files and attachments */
    config.storage.saved_path = "/email_data";

    config.storage.type = esp_mail_file_storage_type_flash;

    /** Set to download heades, text and html messaeges, 
     * attachments and inline images respectively.
    */
    config.download.header = true;
    config.download.text = true;
    config.download.html = true;
    config.download.attachment = true;
    config.download.inlineImg = true;

    config.enable.html = true;
    config.enable.text = true;

    /* Set to enable the sort the result by message UID in the ascending order */
    config.enable.recent_sort = true;

    /* Set to report the download progress via the default serial port */
    config.enable.download_status = true;

    /* Set the limit of number of messages in the search results */
    config.limit.search = 40;

    /** Set the maximum size of message stored in 
     * IMAPSession object in byte
    */
    config.limit.msg_size = 512;

    config.limit.attachment_size = 1024 * 1024 * 5;

    /* Connect to server with the session and config */
    if (!imap.connect(&session, &config))
        return;

    /* Open or select the mailbox folder to read or search the message */
    if (!imap.selectFolder("INBOX"))
        return;

}

void loop()
{ 
  if (!imap.selectFolder("INBOX"))
        return;
  printSelectedMailboxInfo(imap.selectedFolder());
  delay(10000);
}

void printSelectedMailboxInfo(SelectedFolderInfo sFolder)
{
    /* Show the mailbox info */
    ESP_MAIL_PRINTF("\nInfo of the selected folder\nTotal Messages: %d\n", sFolder.msgCount());
    ESP_MAIL_PRINTF("Predicted next UID: %d\n", sFolder.nextUID());
    for (size_t i = 0; i < sFolder.flagCount(); i++)
        ESP_MAIL_PRINTF("%s%s%s", i == 0 ? "Flags: " : ", ", sFolder.flag(i).c_str(), i == sFolder.flagCount() - 1 ? "\n" : "");
}

```
Here is the serial output:

```yaml
Connecting to AP...........
WiFi connected.
IP address: 
xxx.xxx.xxx.xxx

> C: ESP Mail Client v1.5.6
> C: connect to IMAP server
> C: host > xxxx.xxxxxxxxxxxx.xxx.xx
> C: port > 993
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
> C: server connected
> C: check the capability
$ CAPABILITY
* OK swizzonic-mail.ch CommuniGate Pro IMAP Server 7.1.0 ready
* CAPABILITY IMAP4 IMAP4REV1 ACL NAMESPACE UIDPLUS IDLE LITERAL+ QUOTA ID MULTIAPPEND LISTEXT CHILDREN BINARY ESEARCH LOGIN-REFERRALS UNSELECT SASL-IR LIST-EXTENDED SPECIAL-USE CREATE-SPECIAL-USE XLIST AUTH=LOGIN AUTH=PLAIN
$ OK completed
> C: send imap command, LOGIN
$ LOGIN xxx@xxxxxxx.xx xxxxxxxxxxxxxx
$ OK completed
> C: open the mailbox folder
$ EXAMINE "INBOX"
* FLAGS (\Answered \Flagged \Deleted \Seen \Draft $MDNSent $Hidden $Media $Forwarded Junk $Label1 $Label2 $Label3)
* OK [PERMANENTFLAGS (\Answered \Flagged \Deleted \Seen \Draft $MDNSent $Hidden $Media $Forwarded Junk $Label1 $Label2 $Label3)] limited
* 37 EXISTS
* 0 RECENT
* OK [UIDNEXT 1934] predicted next UID
* OK [UIDVALIDITY 402270099] UIDs valid
* OK [UNSEEN 3] message 3 is first unseen
$ OK [READ-ONLY] EXAMINE completed

Info of the selected folder
Total Messages: 37
Predicted next UID: 1934
Flags: \Answered, \Flagged, \Deleted, \Seen, \Draft, $MDNSent, $Hidden, $Media, $Forwarded, Junk, $Label1, $Label2, $Label3

Info of the selected folder
Total Messages: 37
Predicted next UID: 1934
Flags: \Answered, \Flagged, \Deleted, \Seen, \Draft, $MDNSent, $Hidden, $Media, $Forwarded, Junk, $Label1, $Label2, $Label3

Info of the selected folder
Total Messages: 37
Predicted next UID: 1934
Flags: \Answered, \Flagged, \Deleted, \Seen, \Draft, $MDNSent, $Hidden, $Media, $Forwarded, Junk, $Label1, $Label2, $Label3
```

**Additional information and things you've tried:**
When calling closeFolder() and empty() before selectFolder() then SelectedFolderInfo seems to be updated as expected.

mobizt, 2021-11-10:
Thanks, the library was updated to [v1.5.7](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v1.5.7).

You can re-select or re-open the same mailbox with the same readOnly mode but [not too often than 5 seconds](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#select-or-open-the-mailbox-folder-to-search-or-fetch-the-message-inside) otherwise the function will immediately exit with true.

closed 2021-11-10 by mobizt

dkalliv, 2021-11-11:
This works great, thank you.