mobizt/ESP-Mail-Client #117: authentication failed with various servers

aldolo69, 2021-11-12:
**Build tool used:**

- [X ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ X] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
"authentication failed" with gmail.com and other email server. used only demo code with correct login data
the same imap setup works in other mail client.

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

mobizt, 2021-11-13:
https://github.com/mobizt/ESP-Mail-Client/blob/602efd199488a008b182e05b7f0173c68e71fe23/examples/SMTP/Send_Text/Send_Text.ino#L32-L37

closed 2021-11-13 by mobizt

mobizt, 2021-11-13:
You should learn about Google security first.

https://support.google.com/accounts/answer/6010255?hl=en