mobizt/ESP-Mail-Client #142: I had to remove all the F() macros to get it to work.

koolwood, 2022-01-26:
**Build tool used:**

- [x ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [x ] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

In "Send_Text.ino" example I had to remove all the F() macros to get it to work.  Once I did that is sent the email just fine.
Those are the only changes I made to the example code other network and email requirements.
I am not sure it is a bug because I could have something in the IDE set wrong but for not it works.

**Description of problem:**

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

//  session.time.ntp_server = F("pool.ntp.org,time.nist.gov");
  session.time.ntp_server = "pool.ntp.org,time.nist.gov";
```

**Additional information and things you've tried:**

koolwood, 2022-01-26:
I also had to do the same thing in Send_Attachment_File example.

mobizt, 2022-01-26:
This is the esp8266 core issue about flash string pointer which you may need to update the SDK too.

I will fix this on library side too.

closed 2022-01-26 by mobizt

koolwood, 2022-01-26:
Thanks for the response.  I think the F() macro works with the String class because I have used that in other programs.For now I have something that works.  
Thanks for the library.