mobizt/ESP-Mail-Client #160: startTLS

fiveoften, 2022-03-23:
**Build tool used:**

- [ x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**
Lolin(wemos) D1 R2 & Mini
- [x ] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

none

**Description of problem:**

we have the problem that we dont get a startTLS connection. he refuse TLS on Port 143.
Port: 993 SSL works great.

16:41:13.426 -> IMAP server connected
16:41:13.426 -> >  server connected
16:41:13.426 ->
16:41:13.426 -> Check the capability...
16:41:13.426 -> >  check the capability
16:41:13.426 -> * OK [CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ STARTTLS LOGINDISABLED AUTH=CRAM-MD5] Dovecot (Debian) ready.
16:41:13.426 -> * CAPABILITY IMAP4rev1 SASL-IR LOGIN-REFERRALS ID ENABLE IDLE LITERAL+ STARTTLS LOGINDISABLED AUTH=CRAM-MD5
16:41:13.426 -> $ OK Pre-login capabilities listed, post-login capabilities have more.
16:41:13.426 ->
16:41:13.426 -> Send command, STARTTLS
16:41:13.426 -> >  send STARTTLS command
16:41:13.426 -> STARTTLS BAD Error in IMAP command received by server.
16:41:43.265 -> Error, connection closed
16:41:43.265 -> > E: connection closed

**Additional information and things you've tried:**

SSL works great.

our webserver is setup after this tutorial
https://workaround.org/ispmail/bullseye/

mobizt, 2022-03-23:
Your server is not actually support STARTTLS command or not comply with rfc2595.

After checking STARTTLS from response from CAPABILITY command, if found, the STARTTLS command can begin.
Then the server should be reply with response like this
**$ OK Begin TLS negotiation now**

You should tick out the timestamp in Serial Monitor and post the serial debug message using Github formatting

mobizt, 2022-03-23:
I'm investigating this issue.

mobizt, 2022-03-23:
The library update [v2.1.2](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v2.1.2) is available with this issue fixed.

Please update.

closed 2022-03-23 by mobizt