mobizt/ESP-Mail-Client #173: NTP server timeout - Error, NTP server time synching timed out

marmou-fritzing, 2022-05-12:
**Build tool used:**

- [ ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [X] Other
PROGRAMINO IDE

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based
LilyGo TTGO T-call & PMU: http://www.lilygo.cn/claprod_view.aspx?TypeId=21&Id=1285&FId=t28:21:28

**Other Libraries That are used:**

#include <Arduino.h>
#include <string.h>
#include <FS.h>
#include <SD.h>
#include <SPIFFS.h>
#include <Update.h>
#include <SPI.h>
#include <Wire.h>
#include <OneWire.h>
#include <DallasTemperature.h>
#include <WiFi.h>
#include <HTTPClient.h> // HTTP client I use for loging values to cloud.
#include <ESP32Ping.h>
#include <ESP_Mail_Client.h>
#include <EEPROM.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>
#include <axp20x.h> // Power management
#include <PCF8574.h> // Port expander

**Description of problem:**
After last actualisation always happens, that NTP server is not available or don't answer so quick, as is necessary. Here is report of Email client debug and callback:
On older library happens the same (what I remember) only one time.

Connecting to SMTP server...
> C: ESP Mail Client v2.2.2
> C: Wait for NTP server time synching
Error, NTP server time synching timed out
> E: NTP server time synching timed out
> C: Connect to SMTP server
> C: Host > smtp.gmail.com
> C: Port > 465
> C: starting socket
> C: connecting to Server
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: performing the SSL/TLS handshake
> C: verifying peer X.509 certificate
> C: SMTP server connected
SMTP server connected, wait for greeting...
< S: 220 smtp.gmail.com ESMTP lv15-20020a170906bc8f00b006f3ef214dfcsm1904077ejb.98 - gsmtp
Sending greeting response...
> C: Send SMTP command, HELO
< S: 250-smtp.gmail.com at your service, [85.71.29.47]
< S: 250-SIZE 35882577
< S: 250-8BITMIME
< S: 250-AUTH LOGIN PLAIN XOAUTH2 PLAIN-CLIENTTOKEN OAUTHBEARER XOAUTH
< S: 250-ENHANCEDSTATUSCODES
< S: 250-PIPELINING
< S: 250-CHUNKING
< S: 250 SMTPUTF8
Logging in...
[... 104 lines cut ...]
                     if(smtp.closeSession() == true){
                     }else{
                     }
                     PoradiSekvenceOdesliEmailWiFi = 0;
                     VyresHlaseniWIFIEmailem(4);
                     PocetOpakovaniPrikazuWiFi++;
                  }else{
                     PoradiSekvenceOdesliEmailWiFi = 3;
                     VyresHlaseniWIFIEmailem(2);
                  }
               }
            }else if(PoradiSekvenceOdesliEmailWiFi == 3){
               // ------------------- ČIŠTĚNÍ ---------------------
               smtp.sendingResult.clear();
               message.clear();
               OdesliEmail                   = false; // Sending finished, cycle already will not continue
               OdeselPrikazDoWIFI            = false; // Unlocking WIFI next commands
               PoradiSekvenceOdesliEmailWiFi = 0;
            }
```

**Additional information and things you've tried:**
Only notice: Because sending of e-mail takes some time and device is in this time busy and I need to be subroutine for e-mail works simoutaneously with other parts of program, sequence is separated to 4 steps, guided by "PoradiSekvenceOdesliEmailWiFi" variable.
Variables (objects) session and message are defined like a global variables.
I tried find on internet any methode how immediately after start device and connected it to the internet, synchronise time. I'v found nothing. :-( I don'T know what it happens automatically and whether is it necessary.

mobizt, 2022-05-12:
If your device time does not set or sync with NTP server before calling `smtp.connect(&session))`, the library will sync or set the device time with NTP server internally and if the time was not sync within 10 seconds, the error NTP time sync timed out will print out.

This is already written in the example comment.

https://github.com/mobizt/ESP-Mail-Client/blob/68b90789f522db7170eb3ace7129efce3d388949/examples/SMTP/Send_Text/Send_Text.ino#L209-L221

closed 2022-05-12 by mobizt

mobizt, 2022-05-12:
The device time should be valid by manually set or sync with NTP server or RTC chip or library internal NTP server synching.

This valid time is required for creating `Date` header of message to avoid the receiver considered your message as spam.

marmou-fritzing, 2022-05-12:
I understand. Thank You. I try set it manually immediatelly after start of device with helpful of smtp.setSystemTime(WiFi.getTime());. I report result.

mobizt, 2022-05-12:
Do you know what is WiFiNINA?

WiFiNINA is the WiFi library and firmware for SMD21 and RPI2040 devices with U-block ESP32 based WiFi module built-in, not for ESP32 based boards.

For ESP32 and ESP8266, there are many resources about NTP time in youtube and google search. It is simple and easy by just call

`configTime(gmtOffsetInSeconds, daylightOffsetInSeconds, "pool.ntp.org", "time.nist.gov");`

Then you need to wait the value returns from function `time(nullptr)` to be valid timestampt (uint32_t or tiime_t).

I assume that you understand what is timestamp, if not search google.

mobizt, 2022-05-12:
https://github.com/mobizt/ESP-Mail-Client#samd21

marmou-fritzing, 2022-05-12:
Yes, I understand and I don't wanted use WIFININA. :-) I only reported, that function getTime() for ESP32 not found and You confirmed me, that is neccssary use any other method.
Thank you and have a nice day!