mobizt/ESP-Mail-Client #174: SD_Card_Mounting

hall1235, 2022-05-13:
Hello Mobizt

I used esp32 and tried to send a file .csv from SD card via Email but I could not send the file.

Please Help.

I do not understand the function name SD_Card_Mounting its did not work.

My code

#include <Arduino.h>

#include <WiFi.h>
#include <SD.h>

#include <ESP_Mail_Client.h>
#include <extras/SDHelper.h>

#define WIFI_SSID             "my_wifi"
#define WIFI_PASSWORD         "password"

#define SMTP_HOST             "smtp.gmail.com"

#define SMTP_PORT             587

#define AUTHOR_EMAIL          "myemail@gmail.com"
#define AUTHOR_PASSWORD       "mypassword"

SMTPSession smtp;

void smtpCallback(SMTP_Status status);

#define sd_cs                             12
#define sd_sck                           14
#define sd_miso                         27
#define sd_mosi                         13
#define sd_freq                           80000000

#define SPI_SCK_PIN   14
#define SPI_MISO_PIN  27
#define SPI_MOSI_PIN  13
#define SPI_CS_PIN    12
#define ESP32

bool sd_initial_succress = false;
bool flag_sdcard = false;

//SPIClass spiSD( HSPI );

//SPIClass spi;

void setup()
{

  Serial.begin(115200);
  Serial.print("Connecting Internet");

  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(200);
  }

  Serial.println();
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  //  spiSD.begin( sd_sck, sd_miso, sd_mosi, sd_cs );
  //  if ( SD.begin( sd_cs, spiSD ) ) flag_sdcard = true;
  //  flag_sdcard ? Serial.println( "SD initial" ) : Serial.println( "SD fail" );

  Serial.println("In");
  sd_initial_succress = SD_Card_Mounting();
  sd_initial_succress ? Serial.println( "SD initial" ) : Serial.println( "SD fail" );
  delay( portMAX_DELAY );

  smtp.debug(1);
  smtp.callback(smtpCallback);
  ESP_Mail_Session session;

  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = F("mydomain.net");

  session.time.ntp_server = F("pool.ntp.org,time.nist.gov");
  session.time.gmt_offset = 3;
  session.time.day_light_offset = 0;

  SMTP_Message message;

  message.enable.chunking = true;

  message.sender.name = F("ESP Mail");
  message.sender.email = AUTHOR_EMAIL;

  message.subject = F("Test SD card");

  message.addRecipient(F("user1"), F("receiver@hotmail.com"));

  String htmlMsg = "<span style=\"color:#ff0000;\">This message contains 1 inline image and 1 attachment file.</span><br/><br/>[image]";
  message.html.content = htmlMsg;
  message.html.charSet = F("utf-8");

  message.html.transfer_encoding = Content_Transfer_Encoding::enc_qp;
  message.text.content = F("This message contains 1 inline image and 1 attachment file.\r\nThe inline images were not shown in the plain text message.");
  message.text.charSet = F("utf-8");
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_base64;
  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_normal;

  SMTP_Attachment att[1];
  int attIndex = 0;

  att[attIndex].descr.filename = F("dataloger_20220422.csv");
  att[attIndex].descr.mime = F("application/octet-stream"); //csv
  att[attIndex].file.path = F("/m_202204.csv");
  att[attIndex].file.storage_type = esp_mail_file_storage_type_sd;
  att[attIndex].descr.transfer_encoding = Content_Transfer_Encoding::enc_base64;

  message.addAttachment(att[attIndex]);

  if (!smtp.connect(&session))
    return;

  if (!MailClient.sendMail(&smtp, &message, true))
    Serial.println("Error sending Email, " + smtp.errorReason());

  ESP_MAIL_PRINTF("Free Heap: %d\n", MailClient.getFreeHeap());
}

void loop()
{
  delay(portMAX_DELAY);
}

void smtpCallback(SMTP_Status status)
{
  /* Print the current status */
  Serial.println(status.info());

  /* Print the sending result */
  if (status.success())
  {
    Serial.println("----------------");
    ESP_MAIL_PRINTF("Message sent success: %d\n", status.completedCount());
    ESP_MAIL_PRINTF("Message sent failled: %d\n", status.failedCount());
    Serial.println("----------------\n");
    struct tm dt;

    for (size_t i = 0; i < smtp.sendingResult.size(); i++)
    {
      /* Get the result item */
      SMTP_Result result = smtp.sendingResult.getItem(i);
      time_t ts = (time_t)result.timestamp;
      localtime_r(&ts, &dt);

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %d/%d/%d %d:%d:%d\n", dt.tm_year + 1900, dt.tm_mon + 1, dt.tm_mday, dt.tm_hour, dt.tm_min, dt.tm_sec);
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients.c_str());
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject.c_str());
    }
    Serial.println("----------------\n");

    //You need to clear sending result as the memory usage will grow up.
    smtp.sendingResult.clear();
  }
}

mobizt, 2022-05-13:
Did you read the comment in examples?

It states that you should edit your SD card config in file sdHelper.h.

closed 2022-05-13 by mobizt

mobizt, 2022-05-13:
https://github.com/mobizt/ESP-Mail-Client/blob/68b90789f522db7170eb3ace7129efce3d388949/examples/SMTP/Send_Attachment_File/Send_Attachment_File.ino#L16-L18

mobizt, 2022-05-13:
You should change this SPI pins config
https://github.com/mobizt/ESP-Mail-Client/blob/68b90789f522db7170eb3ace7129efce3d388949/src/extras/SDHelper.h#L17-L20

The SPI clock frequency is set to 4 MHz by default.

The  SD_Card_Mounting() function is in that file.

hall1235, 2022-05-18:
Thank you so much Mobizt