mobizt/ESP-Mail-Client #180: Soft WDT reset after "Wait for NTP server time synching"

mmbaguette, 2022-06-04:
-->

**Build tool used:**
Arduino

**Board used (ESP32/ESP8266/Arudino):**
ESP8266

**Other Libraries That are used:**
```cpp
#include <ESP8266WiFi.h>
#include <ESPAsyncUDP.h>
#include <Pinger.h>
#include <SoftwareSerial.h>
#include <LiquidCrystal.h>
#include <ESP_EEPROM.h>
#include <mdns.h>
#include <ESP_Mail_Client.h>
```
**Description of problem:**
Soft WDT reset when sending email:
```
Sending email"\<�? " to someone@gmail.com. Company: Unhackable Smart Home®
Connecting to SMTP server...
> C: ESP Mail Client v2.3.1
> C: Wait for NTP server time synching

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Soft WDT reset
...
```

**Share code snippet to reproduce the issue:**
```cpp
void send_alert(const char *name, const char *IP_str)
{
  SMTP_Message message;
  message.sender.name = "Unhackable Smart Home\u00AE";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = std::string(name) + " has arrived!";
  message.addRecipient("Home", RECIPIENT_EMAIL);
  Serial.printf("Sending email\"%s\" to %s.\n" , message.subject, RECIPIENT_EMAIL);

  std::string htmlMsg = std::string("<h1>") + name + " joined the network!</h1><br><p>Device " + name + " (" + IP_str + ")</p>";
  message.html.content = htmlMsg;
  message.text.charSet = "us-ascii";
  message.html.transfer_encoding = Content_Transfer_Encoding::enc_7bit;

  if (!smtp.connect(&session)) // connect
    return;                    // return if not connected

  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
}
```

mobizt, 2022-06-04:
This is not related to the library as it is intensively tested.

The NTP time synching is process that connect to NTP server using UDP protocol by calling `configTime` function of ESP8266 core SDK and waits for NTP server response for 10 seconds, the wdt was fed during this before exiting the waiting loop when timed out.

If your device crashed before the timed out message showing, you may have the problem with your device power supply or out of memory to establish the UDP connection.

You should debug your code first.

You should set the debug option and port in your IDE.

mobizt, 2022-06-04:
You can test with examples and it should always work.

To include the third party library with this library, you should debug code by yourself.

closed 2022-06-04 by mobizt

mmbaguette, 2022-06-04:
Thanks for your response.
Do know if my function call has anything to do with this issue?
```cpp
#define IP2STR(a) (a)[0], (a)[1], (a)[2], (a)[3]
#define IPSTR "%d.%d.%d.%d"

char IP_str[MAX_LCD];
sprintf(IP_str, IPSTR, IP2STR(joinedIP));
Serial.printf("%s (%s) has arrived!\n", deviceName, IP_str);

if (millis() >= lastEmailJoinAlert + emailJoinAlertInterval) { //don't spam email
  send_alert(deviceName, IP_str);
  lastEmailJoinAlert = millis();
}
```
Revised function:
```cpp
void send_alert(const char *name, const char *IP_str)
{
  char subject[30];
  sprintf(subject, "%s has arrived!", name);
  SMTP_Message message;
  message.sender.name = "Unhackable Smart Homes\u00AE";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = subject;
  message.addRecipient("Home", RECIPIENT_EMAIL);
  Serial.printf("Sending email \"%s\" to %s.\n" , subject, RECIPIENT_EMAIL);

  std::string htmlMsg = std::string("<h1>") + name + " joined the network!</h1><br><p>Device " + name + " (" + IP_str + ")</p>";
  message.html.content = htmlMsg;
  message.text.charSet = "us-ascii";
  message.html.transfer_encoding = Content_Transfer_Encoding::enc_7bit;

  if (!smtp.connect(&session)) // connect
    return;                    // return if not connected

  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
}
```
I don't think it's related to running out of memory.
`Global variables use 32920 bytes (40%) of dynamic memory, leaving 49000 bytes for local variables. Maximum is 81920 bytes.`

mobizt, 2022-06-04:
I cannot help to debug your code.

Why you trust the compile time information instead of runtime checking?

Are you new to embedded programming?

mobizt, 2022-06-04:
I recommend you to update the ESP8266 core SDK to the latest version.

And set the memory option in the IDE, see [this](https://github.com/mobizt/ESP-Mail-Client#memory-options-for-esp8266) for how to.

mmbaguette, 2022-06-04:
I updated the SDK using those instructions. I still get the same issue. Now I just have to debug my code. Thanks for your help.

mmbaguette, 2022-06-05:
Fix: I was sending the email inside a callback function. Running it asynchronously, like in the `loop()` function, solves the issue.