mobizt/ESP-Mail-Client #181: Unable to connect to gmail since they disabled "less secure apps" on May 30, 2022

mtnbkr88, 2022-06-09:
**Build tool used:**

- [X ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

None

**Description of problem:**
It appears as of May 30, 2022 gmail has turned off "less secure apps" which has broken connecting to the gmail server on port 465. I'm getting this error:
< S: 535-5.7.8 Username and Password not accepted. Learn more at
< S: 535 5.7.8  https://support.google.com/mail/?p=BadCredentials je7-20020a170903264700b001676f87473fsm4313619plb.302 - gsmtp
> E: authentication failed
My username and password are correct. I can log in to gmail.com with them using a browser no problem. I'm using code provided by https://microcontrollerslab.com/esp32-send-emails-through-smtp-server-arduino-ide/. How can I get ESP-Mail-Client working now that gmail has disabled "less secure apps"?

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

#include <Arduino.h>
#include <WiFi.h>
#include <ESP_Mail_Client.h>

#define WIFI_SSID "Your_SSID"
#define WIFI_PASSWORD "Your_Password"

#define SMTP_server "smtp.gmail.com"
#define SMTP_Port 465

#define sender_email "Write_Sender_Email_address"
#define sender_password "Write_Sender_Email_Password"

#define Recipient_email "Write_recipient_Email_address"

SMTPSession smtp;

void setup(){
  Serial.begin(115200);
  Serial.println();
  Serial.print("Connecting...");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED){
    Serial.print(".");
    delay(200);
  }
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  smtp.debug(1);

  ESP_Mail_Session session;

  session.server.host_name = SMTP_server ;
  session.server.port = SMTP_Port;
  session.login.email = sender_email;
  session.login.password = sender_password;
  session.login.user_domain = "";

  /* Declare the message class */
  SMTP_Message message;

  message.sender.name = "ESP32";
  message.sender.email = sender_email;
  message.subject = "ESP32 Testing Email";
  message.addRecipient("Microcontrollerslab",Recipient_email);

   //Send HTML message
  String htmlMsg = "<div style=\"color:#FF0000;\"><h1>Hello Microcontrollerslab!</h1><p>This is an HTML email sent from ESP Board</p></div>";
  message.html.content = htmlMsg.c_str();
  message.html.content = htmlMsg.c_str();
  message.text.charSet = "us-ascii";
  message.html.transfer_encoding = Content_Transfer_Encoding::enc_7bit; 

 /* 
  //Send simple text message
  String textMsg = "Hello Microcontrollerslab! This is a simple text sent from ESP board";
  message.text.content = textMsg.c_str();
  message.text.charSet = "us-ascii";
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit; */

  if (!smtp.connect(&session))
    return;

  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
 
}

void loop(){

}

```

**Additional information and things you've tried:**

mobizt, 2022-06-09:
Did you read the comment in all examples?

This is not the library issue and you should  use discussion instead.

mobizt, 2022-06-09:
https://github.com/mobizt/ESP-Mail-Client/blob/78aa94f0050112f6e456bc3acd6435c7079ab08e/examples/SMTP/Send_Text/Send_Text.ino#L34-L46

closed 2022-06-09 by mobizt