mobizt/ESP-Mail-Client #182: Storages disabled = 'class MB_FS' has no member named 'getFlashFile()'

marmou-fritzing, 2022-06-13:
**Build tool used:**

- [X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [X] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
When I comment in file ESP_Mail_FS.h all storage definition blocks (SPIFFS, SD_MMC), then I got during compilation an message " 'class MB_FS' has no member named 'getFlashFile()' ".
Then I added the condition into ESP32_SSL_Client.cpp file - function 'void ESP32_TCP_Client::setCertFile'.
Bold below.
Then it works.

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

    void ESP32_TCP_Client::setCertFile(const char *certFile, mb_fs_mem_storage_type storageType)
{
   if (!mbfs)
      return;
**_#if defined (SPIFFS) || defined (SD_MMC) || defined (sd_fat_fs)_**
   if (strlen(certFile) > 0)
   {
      MB_String filename = certFile;
      if (filename.length() > 0)
      {
         if (filename[0] != '/')
            filename.prepend('/');
      }
      
      int len = mbfs->open(filename, storageType, mb_fs_open_mode_read);
      if (len > -1)
      {
         if (storageType == mb_fs_mem_storage_type_flash)
         {
            fs::File file = mbfs->getFlashFile();
            wcs->loadCACert(file, len);
            mbfs->close(storageType);
            baseSetCertType(esp_mail_cert_type_file);
         }else if (storageType == mb_fs_mem_storage_type_sd)
         {
            
#if defined(MBFS_ESP32_SDFAT_ENABLED)
            
            if (cert_buf)
               mbfs->delP(&cert_buf);
            
            cert_buf = (char *)mbfs->newP(len);
            if (mbfs->available(storageType))
               mbfs->read(storageType, (uint8_t *)cert_buf, len);
            
            mbfs->close(storageType);
            wcs->setCACert((const char *)cert_buf);
            baseSetCertType(esp_mail_cert_type_file);
            
#elif defined(MBFS_SD_FS)
            fs::File file = mbfs->getSDFile();
            wcs->loadCACert(file, len);
            mbfs->close(storageType);
            baseSetCertType(esp_mail_cert_type_file);
#endif
         }
      }
   }
**_#endif_**
}
```

**Additional information and things you've tried:**

mobizt, 2022-06-13:
Please test with examples first.

Than create the minimum and complete test example that reproduced the issue.

Post the **full compilation error** here.

mobizt, 2022-06-13:
I intensively test with filesystems and don't found any issue.

And I test it again and everything is ok.

closed 2022-06-13 by mobizt