mobizt/ESP-Mail-Client #183: GMAIL STOP SENDING MAY 30 2022

pinomarcelo, 2022-06-13:
**Build tool used:**

- [ ] Arudino
- [X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [X] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

	adafruit/Adafruit BusIO@^1.7.2
	adafruit/Adafruit GFX Library@^1.10.13
	adafruit/Adafruit PCD8544 Nokia 5110 LCD library@^2.0.1
	claws/BH1750@^1.3.0
	adafruit/Adafruit AHTX0@^2.0.1
	dancol90/ESP8266Ping@^1.0
	mathworks/ThingSpeak@^2.0.0
	arduino-libraries/NTPClient@^3.1.0
	mobizt/ESP Mail Client@^2.0.3

**Description of problem:**
Since may 30 2022 gmail disabled the user-pass autentication, and it can't be enabled any more from user settings:
https://support.google.com/accounts/answer/6010255?hl=en-419

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

//MAIL CLIENT
#define SMTP_HOST "smtp.gmail.com"
#define SMTP_PORT 465
#define AUTHOR_EMAIL "ciquibicremoto@gmail.com"
#define AUTHOR_PASSWORD "gmail account password"
#define RECIPIENT_EMAIL "destination"
SMTPSession smtp;

    smtp.debug(1);
    Serial.print(timeClient.getFormattedTime() + "\t" + "Sending Email... \r\n");
    ESP_Mail_Session session;
    session.server.host_name = SMTP_HOST;
    session.server.port = SMTP_PORT;
    session.login.email = AUTHOR_EMAIL;
    session.login.password = AUTHOR_PASSWORD;
    session.login.user_domain = "";
    SMTP_Message message;
    message.sender.name = siteName;
    message.sender.email = AUTHOR_EMAIL;
    message.subject = "Sistema encendido";
    message.addRecipient("Ciquibic", RECIPIENT_EMAIL);
    String textMsg = "Sistema encendido";
    message.text.content = textMsg.c_str();
    message.text.charSet = "us-ascii";
    message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
    message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
    message.response.notify = esp_mail_smtp_notify_success | esp_mail_smtp_notify_failure | esp_mail_smtp_notify_delay;
    // Connect to server with the session config 
    if (!smtp.connect(&session))
      return;
    // Start sending Email and close the session 
    if (!MailClient.sendMail(&smtp, &message))
      Serial.println(timeClient.getFormattedTime() + "\t" + "Error sending Email, " + smtp.errorReason());
    else
      Serial.println(timeClient.getFormattedTime() + "\t" + "Email Sent!");
  } else {
    Serial.print("No network available");
  }

```

**Additional information and things you've tried:**
Response from SMTP google server:

> C: ESP Mail Client v2.4.1
> C: Wait for NTP server time synching
> C: Connect to SMTP server
> C: Host > [smtp.gmail.com](http://smtp.gmail.com/)
> C: Port > 465
> C: SMTP server connected
< S: 220 [smtp.gmail.com](http://smtp.gmail.com/) ESMTP i9-20020a9d53c9000000b0060adcc87e37sm3596821oth.74 - gsmtp
> C: Send SMTP command, HELO
< S: [250-smtp.gmail.com](http://250-smtp.gmail.com/) at your service, [200.16.16.13]
< S: 250-SIZE 35882577
< S: 250-8BITMIME
< S: 250-AUTH LOGIN PLAIN XOAUTH2 PLAIN-CLIENTTOKEN OAUTHBEARER XOAUTH
< S: 250-ENHANCEDSTATUSCODES
< S: 250-PIPELINING
< S: 250-CHUNKING
< S: 250 SMTPUTF8
> C: Send SMTP command, AUTH PLAIN
> C: [ciquibicremoto@gmail.com](mailto:ciquibicremoto@gmail.com)
> C: [ciquibicremoto@gmail.com](mailto:ciquibicremoto@gmail.com) *************
< S: 535-5.7.8 Username and Password not accepted. Learn more at
< S: 535 5.7.8  https://support.google.com/mail/?p=BadCredentials i9-20020a9d53c9000000b0060adcc87e37sm3596821oth.74 - gsmtp
> E: authentication failed

mobizt, 2022-06-13:
https://github.com/mobizt/ESP-Mail-Client/blob/3caf37e7ac768a3367e7b0bb8e9c1a114cb4e127/examples/SMTP/Send_Text/Send_Text.ino#L34-L46

closed 2022-06-13 by mobizt

mobizt, 2022-06-13:
This is not the library issue in any way, you should ask the question in the [discussion ](https://github.com/mobizt/ESP-Mail-Client/discussions)instead.

For the usage, the library provides much more information that you have to read it.