mobizt/ESP-Mail-Client #191: ESP32-CAM send a jpg file in SD card : "SD Storage is not ready."

m3hm3tb4yr4m, 2022-07-19:
I actually use the librairy ESP-Mail-Client to program a ESP32-CAM with Arduino IDE. I save a picture in the sd card. After, I want to send this picture in attachment of the mail. I don't understand why I receive an error like "SD Storage is not ready". The mail is sent without the attachment. When I see the logs :
22:43:00.287 -> Saved file to path: /images/2022.07.19-21.42.57.jpg
22:43:01.307 -> preparing smtp
22:43:01.307 -> Connecting to SMTP server...
22:43:01.307 -> > C: ESP Mail Client v2.4.7
22:43:01.307 -> > C: Wait for NTP server time synching
22:43:01.647 -> > C: Connect to SMTP server
22:43:01.647 -> > C: Host > mail.***.com
22:43:01.647 -> > C: Port > 587
22:43:01.647 -> > C: starting socket
22:43:01.885 -> > C: connecting to Server
22:43:01.953 -> > C: SMTP server connected
22:43:01.953 -> 
22:43:01.953 -> SMTP server connected, wait for greeting...
22:43:02.939 -> < S: 220 mail.***.com ESMTP ready
22:43:02.939 -> 
22:43:02.939 -> Sending greeting response...
22:43:02.939 -> > C: Send SMTP command, HELO
22:43:05.965 -> < S: 250-mail.infomaniak.com
22:43:05.965 -> < S: 250-PIPELINING
22:43:05.965 -> < S: 250-SIZE
22:43:05.965 -> < S: 250-ETRN
22:43:05.965 -> < S: 250-ENHANCEDSTATUSCODES
22:43:05.965 -> < S: 250-8BITMIME
22:43:05.965 -> < S: 250-DSN
22:43:05.965 -> < S: 250-AUTH PLAIN LOGIN
22:43:05.965 -> < S: 250 STARTTLS
22:43:05.965 -> 
22:43:05.965 -> Send command, STARTTLS
22:43:05.965 -> > C: Send STARTTLS command
22:43:07.971 -> < S: 220 2.0.0 Start TLS
22:43:07.971 -> > C: Perform the SSL handshake
22:43:07.971 -> > C: seeding the random number generator
22:43:08.005 -> > C: setting up the SSL/TLS structure
22:43:08.005 -> ! W: Skipping SSL Verification. INSECURE!
22:43:08.005 -> > C: setting hostname for TLS session
22:43:08.005 -> > C: performing the SSL/TLS handshake
22:43:09.909 -> > C: verifying peer X.509 certificate
22:43:09.909 -> 
22:43:09.909 -> Sending greeting response...
22:43:09.909 -> > C: Send SMTP command, HELO
22:43:09.943 -> < S: 250-mail.infomaniak.com
22:43:09.943 -> < S: 250-PIPELINING
22:43:09.943 -> < S: 250-SIZE
22:43:09.977 -> < S: 250-ETRN
22:43:09.977 -> < S: 250-ENHANCEDSTATUSCODES
22:43:09.977 -> < S: 250-8BITMIME
22:43:09.977 -> < S: 250-DSN
22:43:09.977 -> < S: 250 AUTH PLAIN LOGIN
22:43:09.977 -> 
22:43:09.977 -> Logging in...
22:43:09.977 -> > C: Send SMTP command, AUTH PLAIN
22:43:09.977 -> > C: ***@***.**
22:43:09.977 -> > C: ***@***.** ***********
22:43:10.045 -> < S: 235 2.0.0 OK
22:43:10.045 -> 
22:43:10.045 -> Sending Email...
22:43:10.045 -> > C: Send Email
22:43:10.045 -> 
22:43:10.045 -> Sending message header...
22:43:10.045 -> > C: Send message header
22:43:10.113 -> < S: 250 2.1.0 Ok
22:43:10.181 -> < S: 250 2.1.5 Ok
22:43:10.181 -> 
22:43:10.181 -> Sending message body...
22:43:10.181 -> > C: Send message body
22:43:10.249 -> < S: 354 End data with <CR><LF>.<CR><LF>
22:43:10.249 -> 
22:43:10.249 -> Appending message...
22:43:10.249 -> > C: apend message
22:43:10.283 -> 
22:43:10.283 -> Sending inline data...
22:43:10.283 -> > C: Send inline data
22:43:10.283 -> 
22:43:10.283 -> 
22:43:10.283 -> > C: 
22:43:10.283 -> SD Storage is not ready.
22:43:10.283 -> > E: SD Storage is not ready.
22:43:10.283 -> 
22:43:10.283 -> Sending attachments...
22:43:10.283 -> > C: Send attachments
22:43:10.283 -> 
22:43:10.283 -> 
22:43:10.283 -> > C: 
22:43:10.384 -> SD Storage is not ready.
22:43:10.384 -> > E: SD Storage is not ready.
22:43:10.384 -> 
22:43:10.384 -> Finishing the message sending...
22:43:10.418 -> > C: Finish the message sending
22:43:10.792 -> < S: 250 2.0.0 Ok: queued as 4LnW4X51f0zlqwsJ
22:43:10.792 -> 
22:43:10.792 -> Closing the session...
22:43:10.792 -> > C: Terminate the SMTP session
22:43:10.860 -> < S: 221 2.0.0 Bye
22:43:10.860 -> 
22:43:10.860 -> Message sent successfully
22:43:10.860 -> > C: Message sent successfully
22:43:10.860 -> 
22:43:10.860 -> ----------------
22:43:10.860 -> Message sent success: 1
22:43:10.860 -> Message sent failed: 0
22:43:10.860 -> ----------------
22:43:10.860 -> 
22:43:10.860 -> Message No: 1
22:43:10.860 -> Status: success
22:43:10.860 -> Date/Time: Tue Jul 19 21:43:07 2022
22:43:10.860 -> 
22:43:10.860 -> Recipient: ***@***.**
22:43:10.860 -> Subject: Motion detection from ESP32cam
22:43:10.860 -> ----------------

The program is the following:

`void sendPhoto(void)
{
  Serial.println("preparing smtp");
 
  smtp.debug(1);
  smtp.callback(smtpCallback);
  ESP_Mail_Session session;
  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.time.ntp_server = F("pool.ntp.org,time.nist.gov");
  session.time.gmt_offset = 3;
  session.time.day_light_offset = 0;
  SMTP_Message message;
  message.sender.name = F("ESP Mail"); // This witll be used with 'MAIL FROM' command and 'From' header field.
  message.sender.email = AUTHOR_EMAIL; // This witll be used with 'From' header field.
  message.subject = F("Motion detection from ESP32cam");
  message.addRecipient(F("moi"), F("mbd@keemail.me")); // This will be used with RCPT TO command and 'To' header field.
  String textMsg = "Motion detected : ESP32cam.";
  message.text.content = textMsg;
  message.text.charSet = F("us-ascii");
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
  SMTP_Attachment  att; // att[2];
  att.file.storage_type = esp_mail_file_storage_type_sd;
  att.file.path=photo_2_send;
  att.descr.transfer_encoding = Content_Transfer_Encoding::enc_base64;
  att.descr.content_encoding = Content_Transfer_Encoding::enc_base64;
  message.addInlineImage(att);
  message.addAttachment(att);
  if (!smtp.connect(&session))
    return;
  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
}`

mobizt, 2022-07-19:
You must follow examples and read the comments first.

You can save file to SD or your SD card is ready to use but you **must let the library know your filesystem used and its configuration**.

You should changes the macro definitions for your filesystem in ESP_Mail_FS.h as in example comment 
https://github.com/mobizt/ESP-Mail-Client/blob/e669ca2ecdad988071e27aca55a84433d3f4fc1d/examples/SMTP/Send_Attachment_File/Send_Attachment_File.ino#L16-L18

https://github.com/mobizt/ESP-Mail-Client/blob/e669ca2ecdad988071e27aca55a84433d3f4fc1d/src/ESP_Mail_FS.h#L64-L68

Change the SD pins configuration in /extras/SDHelper.h as in the example comment. Typo in the example comment, the file path should be /extras/SDHelper.h in stead of /addons/SDHelper.h 

In case of SD SPI interface

https://github.com/mobizt/ESP-Mail-Client/blob/e669ca2ecdad988071e27aca55a84433d3f4fc1d/src/extras/SDHelper.h#L17-L20

This line initiate the SD configuration in the function `SD_Card_Mounting()`.

https://github.com/mobizt/ESP-Mail-Client/blob/e669ca2ecdad988071e27aca55a84433d3f4fc1d/src/extras/SDHelper.h#L71-L72

Call function `SD_Card_Mounting()` in your code.

https://github.com/mobizt/ESP-Mail-Client/blob/e669ca2ecdad988071e27aca55a84433d3f4fc1d/examples/SMTP/Send_Attachment_File/Send_Attachment_File.ino#L106-L107

closed 2022-07-19 by mobizt

m3hm3tb4yr4m, 2022-10-11:
I'm using a SD card with FAT32 format. 
If I understand, I have to find the file Arduino\libraries\ESP_Mail_Client\src\ESP_Mail_FS.h and modify it with uncommenting lines 41 to 43 and commenting lines 64 to 68. 
For the second file Arduino\libraries\ESP_Mail_Client\src\extras\SDHelper.h, I don't understand what I have to do. 
Can you explain me what I have to do exactly?

Thanks
RegardsM. DONAT-- 
 Envoyé avec Tutanota, la boîte aux lettres sécurisée et sans publicité.

mobizt, 2022-10-11:
@m3hm3tb4yr4m 

Right, if your card type is MMC 

In file [SDHelper.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/extras/SDHelper.h#L96), the second argument of function [MailClient.sdMMCBegin](https://github.com/mobizt/ESP-Mail-Client/tree/master/src#initialize-the-sd_mmc-card-esp32-only) using at line number 96 was set for 4-bit mode

https://github.com/mobizt/ESP-Mail-Client/blob/5c555d26838055dbabb9c72a8545f87ccf1fe483/src/extras/SDHelper.h#L96

If your MMC card interface is 4-bit, you don't have to do anything with that SDHelper.h file.

Unless you connect MMC card in 1-bit mode, the line number 96 should change to

`if (!MailClient.sdMMCBegin("/sdcard", true, true))`

The function `sdMMCBegin` is to let the library knows the type of your SD filesystem which the function arguments are the same as `SD_MMC.begin` in the SD_MMC.h core library.

For SPI, 1-bit mode and 4-bit mode interfaces, please see [this](https://www.instructables.com/Select-SD-Interface-for-ESP32/).

This the [example](https://github.com/espressif/arduino-esp32/blob/master/libraries/SD_MMC/examples/SDMMC_Test/SDMMC_Test.ino) to test your MMC card.

If you card type is SD, I recommend the SdFat instead because it offers the fastest transfer speed when compared to the core SD.h library.

mobizt, 2022-10-11:
@m3hm3tb4yr4m 

You should call `SD_Card_Mounting();` in your code as in the examples.