mobizt/ESP-Mail-Client #201: SMTP unable to connect to server

LeeSpa, 2022-10-19:
**Build tool used:**

- [ x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x ] ESP32-based

**Other Libraries That are used:**

none

**Description of problem:**

I tried example "SMTP - Send_Text "and got following error: ! E: UNKNOWN ERROR CODE (0001)

_Connecting to SMTP server...
> C: ESP Mail Client v2.5.3
> C: Wait for NTP server time synching
> C: Connect to SMTP server
> C: Host > smtp.gmail.com
> C: Port > 465
> C: seeding the random number generator
> C: setting up the SSL/TLS structure
! W: Skipping SSL Verification. INSECURE!
> C: setting hostname for TLS session
> C: perform the SSL/TLS handshake
! E: UNKNOWN ERROR CODE (0001)
> C: cleaning SSL connection
Error, unable to connect to server
> E: unable to connect to server_

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**
used my Google account with app password

mobizt, 2022-10-19:
If your ESP32 Arduino Core SDK version is older than v1.0.5 i.e., v1.0.4, you need to update to the newer version.

Using Arduino IDE's Boards Manager to update.

mobizt, 2022-10-19:
https://github.com/mobizt/ESP-Mail-Client#esp32-and-esp8266-sdks

closed 2022-10-19 by mobizt

LeeSpa, 2022-10-20:
Great Thanks! Its working now.

mobizt, 2022-10-20:
I recommend updating the library to v2.6.0 too, due to another SMTP bug fix.

LeeSpa, 2022-10-20:
Right, I did update also to v2.6.0
Thanks mobizt