mobizt/ESP-Mail-Client #204: conflict with async_tcp

mugginsjm, 2022-11-04:
I have been trying to trigger an email send with attachment (100K) using MQTT
I'm running ESPAsyncWebServer as well as AsyncMqttClient. I don't understand the watchdog error and don't know how to correct it. Can anyone help please. I get:

upload "fridge.txt", 0%
upload "fridge.txt", 5%
upload "fridge.txt", 10%
upload "fridge.txt", 15%
upload "fridge.txt", 20%
E (39166) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (39166) task_wdt:  - async_tcp (CPU 1)

E (39166) task_wdt: Tasks currently running:
E (39166) task_wdt: CPU 0: IDLE
E (39166) task_wdt: CPU 1: loopTask
E (39166) task_wdt: Aborting.
abort() was called at PC 0x401022b4 on core 0

Backtrace:0x40083b49:0x3ffbf09c |<-CORRUPTED

ELF file SHA256: 0000000000000000

Rebooting...

mobizt, 2022-11-04:
The error is about async tcp which not related to this library.

closed 2022-11-04 by mobizt

mobizt, 2022-11-04:
You may need to manage how to use that async task or function properly.

The library uses normal sync tcp method for operation which will not block the system for too long and allow idle time to feed the wdt inside which any waiting loop has time out checking and will not cause wdt overflowing reset.

This is not conflicts but your code is not well manage.

mugginsjm, 2022-11-04:
Ok thanks. Any general hints on how to resolve. I have a cut down sketch with nothing in the loop and just basic asyncwebserver and asyncwifi and it still stalls.