mobizt/ESP-Mail-Client #207: email programma work not good after update arduino esp32 framwork from 2.0.0 to 2.05

Nulista, 2022-11-15:
<Hallo,
!-- Thanks for reporting a bug for this project. READ THIS FIRST:

- Provide as many details as possible. Simply saying "X gives bug" or "X gives error" is not enough!
- Paste logs, configuration sample (platformio.ini file) and code into the backticks (```).
- Read through the template carefully and fill out all missing details.
- Please also search for similar issues in this issue tracker first and read through the ESPHome FAQ.

DO NOT DELETE ANY TEXT from this template! Otherwise the issue may be closed without a comment.
-->

**Build tool used:**

- [*] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**
<Lolin32 wemos!--
Please provide details on which board is used. 
-->
- [ ] ESP8266-based
- [*] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
With arduino esp32 framework version 2.0.0 it works perfectly, but with version 2.05(latest), I get an error and email is not sent.
The wifi router is ± 8years old :// Ps. security settings in router Wpa/Wpa2 - PSK.
Here is the debug text:
//---------------------------------------------------------------------------------------------------------------
[     4][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz

Connecting to AP[    59][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
[   147][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 2 - STA_START
.[   223][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
..............[  2967][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[  2967][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 192.168.1.104, MASK: 255.255.255.0, GW: 192.168.1.1

WiFi connected.
IP address: 
192.168.1.104

Connecting to SMTP server...
> C: Wait for NTP server time synching
[  8056][E][vfs_api.cpp:104] open(): /littlefs/tze.txt does not exist, no permits for creation
[  8119][I][ESP32_SSL_Client.cpp:208] connect_ssl(): WARNING: Skipping SSL Verification. INSECURE!

SMTP server connected, wait for greeting...

Sending greeting response...

Logging in...

Sending Email...

Sending message header...

Sending message body...

Finishing the message sending...

Closing the session...
[ 11335][D][WiFiClient.cpp:546] connected(): Disconnected: RES: 0, ERR: 128

Message sent successfully

----------------
Message sent success: 1
[... 294 lines cut ...]
      /* Get the result item */
      SMTP_Result result = smtp.sendingResult.getItem(i);

      // In case, ESP32, ESP8266 and SAMD device, the timestamp get from result.timestamp should be valid if
      // your device time was synched with NTP server.
      // Other devices may show invalid timestamp as the device time was not set i.e. it will show Jan 1, 1970.
      // You can call smtp.setSystemTime(xxx) to set device time manually. Where xxx is timestamp (seconds since Jan 1, 1970)
      time_t ts = (time_t)result.timestamp;

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %s\n", asctime(localtime(&ts)));
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients.c_str());
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject.c_str());
    }
    Serial.println("----------------\n");

    // You need to clear sending result as the memory usage will grow up.
    smtp.sendingResult.clear();
  }
}

**Additional information and things you've tried:**

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2022-11-15:
What is the error. I don't see any error from your post.

The library was intensively tested and updated recently. 

Of course, it tested for many platforms and supported latest sdk version (2.0.5 for ESP32 Arduino Core).

And I test it again now and everything is ok.

FYI.

The library has its own TCP clients that are dependent of the SSL client of core libraries, and nothing is related to the sdk version.

closed 2022-11-15 by mobizt

Nulista, 2022-11-15:
Thanks for the info.
If I set the priority higher, the email works ok.

mobizt, 2022-11-15:
That is your email server issue