mobizt/ESP-Mail-Client #215: Upgrade to Arduino IDE 2.0  resulted in conflicts with SD library

stardad60, 2022-12-28:
**Build tool used:**

- [X ] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [X ] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
Updated  to Arduino IDE to 2.0 and now struggling with previously working sketches using the ESP_Mail_Client library.  
The first problem was compile errors:

In file included from c:\Users\kh_admin\Documents\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,
                 from c:\Users\kh_admin\Documents\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from c:\Users\kh_admin\Documents\Arduino\libraries\SD\src/SD.h:20,
                 from c:\Users\kh_admin\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_FS.h:102,
                 from c:\Users\kh_admin\Documents\Arduino\libraries\ESP_Mail_Client\src/extras/RFC2047.h:7,
                 from c:\Users\kh_admin\Documents\Arduino\libraries\ESP_Mail_Client\src/ESP_Mail_Client.h:34,
                 from C:\Users\kh_admin\Documents\Arduino\EMailtest4_updated_public\EMailtest4_updated_public.ino:6:
c:\Users\kh_admin\Documents\Arduino\libraries\SD\src/utility/Sd2PinMap.h:524:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
Multiple libraries were found for "SD.h"
  Used: C:\Users\kh_admin\Documents\Arduino\libraries\SD
  Not used: C:\Users\kh_admin\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\SD
exit status 1

Compilation error: exit status 1

There is confusion on what SD library to use…. I tried the solution shown in the Arduino Forum entry https://forum.arduino.cc/t/compilation-error-arduino-2-0-sd-h/1045826 but it did not work for me. 
I tried an alternative solution of deleting the SD library that is referenced as ‘Used ‘ in the error (C:\Users\kh_admin\Documents\Arduino\libraries\SD).
 Deleting that library  directory  allows progress  past that compiler error.

However then a 2nd compile error emerges:

C:\Users\kh_admin\Documents\Arduino\EMailtest4_updated_public\EMailtest4_updated_public.ino: In function 'void smtpCallback(SMTP_Status)':
C:\Users\kh_admin\Documents\Arduino\EMailtest4_updated_public\EMailtest4_updated_public.ino:125:59: error: cannot pass objects of non-trivially-copyable type 'class MB_String' through '...'
       ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients);
                                                           ^
C:\Users\kh_admin\Documents\Arduino\EMailtest4_updated_public\EMailtest4_updated_public.ino:126:54: error: cannot pass objects of non-trivially-copyable type 'class MB_String' through '...'
       ESP_MAIL_PRINTF("Subject: %s\n", result.subject);
                                                      ^

exit status 1

Compilation error: cannot pass objects of non-trivially-copyable type 'class MB_String' through '...'

                                                          
It appears there are things needed in one SD library but that library conflicts elsewhere.…probably a deeper library.   I’m not sure how to solve this properly. I thought about blowing everything away and starting over with totally new installs. I have lots of sketches with lots of libraries  and I’m afraid  such a drastic move may lead to even more library problems. Right now it seems best to focus and trace what the the ESP_MAIL_Client  files don’t like.  Any recommendations?

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP_Mail_Client.h>

#define WIFI_SSID "xxxxxxxxxxxx"
#define WIFI_PASSWORD "xxxxxxxxxxxxx"

/* #define SMTP_HOST "smtp.gmail.com" */
#define SMTP_HOST "smtp.mail.yahoo.com"
/*#define SMTP_PORT 465*/   /*secure channel*/
/*#define SMTP_PORT 587*/   /*plain text , on command secure*/
                                /* both 587 and 465 seem to work with yahoo*/
#define SMTP_PORT 587

/* The sign in credentials */
#define AUTHOR_EMAIL "zzzzzzzzzzzz@att.net"

#define AUTHOR_PASSWORD "zzzzzzzzzzzzz"

/* Recipient's email*/
#define RECIPIENT_EMAIL "zzzzzzzzz@vtext.com"
#define RECIPIENT_EMAIL_TEST "zzzzzzzzzz@att.net"

/* The SMTP Session object used for Email sending */
SMTPSession smtp;

/* Callback function to get the Email sending status */
void smtpCallback(SMTP_Status status);

void setup() {
  Serial.begin(9600);
  Serial.println();
  Serial.print("Connecting to AP");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(200);
  }
  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println();

  /** Enable the debug via Serial port
     none debug or 0
     basic debug or 1
  */
  smtp.debug(1);

  /* Set the callback function to get the sending results */
  smtp.callback(smtpCallback);

  /* Declare the session config data */
  ESP_Mail_Session session;

  /* Set the session config */
  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = "";

  /* Declare the message class */
  SMTP_Message message;

  /* Set the message headers */
  message.sender.name = "ESP8266";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = "ESP8266 Test Email";
  message.addRecipient("kurt", RECIPIENT_EMAIL);

  
  String textMsg = "Hello World! - Sent from ESP8266 board\r\n.\r\n";
  message.text.content = textMsg.c_str();
  message.text.charSet = "us-ascii";
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;

  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
  message.response.notify = esp_mail_smtp_notify_success | esp_mail_smtp_notify_failure | esp_mail_smtp_notify_delay;

  /* Connect to server with the session config */
  if (!smtp.connect(&session))
    return;

  /* Start sending Email and close the session */
  if (!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending Email, " + smtp.errorReason());
}

void loop() {

}

/* Callback function to get the Email sending status */
void smtpCallback(SMTP_Status status) {
  /* Print the current status */
  Serial.println(status.info());

  /* Print the sending result */
  if (status.success()) {
    Serial.println("----------------");
    ESP_MAIL_PRINTF("Message sent success: %d\n", status.completedCount());
    ESP_MAIL_PRINTF("Message sent failled: %d\n", status.failedCount());
    Serial.println("----------------\n");
    struct tm dt;

    for (size_t i = 0; i < smtp.sendingResult.size(); i++) {
      /* Get the result item */
      SMTP_Result result = smtp.sendingResult.getItem(i);
      time_t ts = (time_t)result.timestamp;
      localtime_r(&ts, &dt);

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %d/%d/%d %d:%d:%d\n", dt.tm_year + 1900, dt.tm_mon + 1, dt.tm_mday, dt.tm_hour, dt.tm_min, dt.tm_sec);
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients);
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject);
    }
    Serial.println("----------------\n");
  }
}

```

**Additional information and things you've tried:**

mobizt, 2022-12-28:
The problem is you installed SD library (c:\Users\kh_admin\Documents\Arduino\libraries\SD) that is not compatible with core library SD.h of ESP8266 and ESP32.

This is already stated here
https://github.com/mobizt/ESP-Mail-Client#third-party-sd-library-must-be-removed

closed 2022-12-28 by mobizt

mobizt, 2022-12-28:
If you want to use this SD library (c:\Users\kh_admin\Documents\Arduino\libraries\SD) for other Arduino devices except for ESP8266 and ESP32, you must install under the same folder of your sketch.

mobizt, 2022-12-28:
The global library installation in Arduino platform causes many problems when the same header files existed in many libraries inside the libraries folder.

stardad60, 2022-12-28:
Yes, that explains my first action when I removed the SD directory and the compile went further.  For the 2nd compile error I did find a solution to the  two errors.  These changes matched generic solutions to errors stating "cannot pass objects of non-trivially-copyable type 'class MB_String' through '...'" .   It helped that ESP_MAIL_Client had  examples using the same string convert.  
I replaced
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients);
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject);
with
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients.c_str());
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject.c_str());

 The original code was running for the last couple of years. **Was there an ESP_MAIL_Client change that caused the need for this update or an underlying library?**    I'm not sure if there is any relation to the SD library issue pointed to in the above answers.

mobizt, 2022-12-28:
I think that you never update library so long time which everything has changed so far.

If you check the examples, you will see how.

https://github.com/mobizt/ESP-Mail-Client/blob/f2fe563b5fab47c728130da88670e014265cbfdd/examples/SMTP/Send_Text/Send_Text.ino#L286-L287

mobizt, 2022-12-28:
This was changed since [v2.0.0](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v2.0.0).