mobizt/ESP-Mail-Client #217: Sending image file stored in SPIFFS

NDC-SC, 2023-01-09:
**Build tool used:**

- [ X] Arduino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arduino):**

- [ ] ESP8266-based
- [ X] ESP32-based ESP32 DEVKIT DOIT

**Other Libraries That are used:**

**Description of problem:**

An image data file stored in PROGMEM is successfully attached to a sent email.
An image file located in the data folder within the sketch folder does not get included as an attachment in an email. The problem appears to be that the file in the data folder is not found by LittleFS, because of a LittleFS loading issue
E (5253) esp_littlefs: mount failed,  (-84)
E (5254) esp_littlefs: Failed to initialize LittleFS

The image file is successfully loaded into SPIFFS with the ESP32 Sketch Data Upload plugin (as checked using the built-in SPIFFS library)

Arduino IDE 1.8.19 and ESP32 DEVKIT DOIT with Core SDK version 2.0.6

Is there a work around with the built-in SPIFFS library, as supported by espressif
OR
Must all image files be included as image data files stored in PROGMEM

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

mobizt, 2023-01-09:
The filesystem used in this library can be configurable in [ESP_Mail_FS.h](https://github.com/mobizt/ESP-Mail-Client/blob/master/src/ESP_Mail_FS.h) and already wrote in the example's comments.

In `ESP_Mail_FS.h` at line No. 57, the flash filesystem used in this library will set to LittleFS when you are using ESP32 core v2.0.x otherwise SPIFFS will be used in the library. This is the requirement from some users and SPIFFS has bugs in reading file which I can't remember.

https://github.com/mobizt/ESP-Mail-Client/blob/842f30eff4a590222dd172e512d2750dff47a6f7/src/ESP_Mail_FS.h#L51-L69

If you want to use SPIFFS instead of LittleFS, in ESP_Mail_FS.h, line 60-61 can be replaced by lines 66-67.

If you don't want this config to be overwritten when you update the library, you can create your own configuration file `"Custom_ESP_Mail_FS.h"` in the same src folder as `ESP_Mail_FS.h`.

In your `"Custom_ESP_Mail_FS.h" `, write the following

```cpp
#pragma once

#ifndef Custom_ESP_Mail_FS_H
#define Custom_ESP_Mail_FS_H

#include <SPIFFS.h>
#undef ESP_MAIL_DEFAULT_FLASH_FS // remove the previous definition in ESP_Mail_FS.h
#define ESP_MAIL_DEFAULT_FLASH_FS SPIFFS

#endif

```

Or you can use LittleFS by trying this plugin for file upload instead
https://github.com/lorol/arduino-esp32fs-plugin

As this is not the library issue, I will close the issue and the discussion can be continued.

closed 2023-01-09 by mobizt

mobizt, 2023-01-09:
The attachment files sources are not limit which you can read and store in all filesystems (flash and sd) and PROGMEM.

NDC-SC, 2023-01-09:
I replaced LittleFS with SPIFFS and both the image file loaded in SPIFFS and the image data file in PROGMEM were successfully included in an email.
(If you want to use SPIFFS instead of LittleFS, in ESP_Mail_FS.h, line 60-61 can be replaced by lines 66-67.)

I wonder why image file loaded in SPIFFS with the original ESP32 Sketch Data Upload plugin wasn't recognised by LittleFS
Is the github.com/lorol/arduino-esp32fs-plugin specifically required when using LittleFS ?

mobizt, 2023-01-09:
I have never check that recent ESP32 core v2.0.6 fixes this issue or not. 
If this issue still existed, that fork file upload plugin may require.
I never test this plugin before, but the developer claims it works for LittleFS.

NDC-SC, 2023-01-09:
The original ESP32 Sketch Data Upload plugin and SPIFFS is a good solution.

What was the reason for ESP-Mail_Client using LittleFS and not the built-in SPIFFS for ESP32, which is supported by espressif

mobizt, 2023-01-09:
You should research the topic about LittleFS vs SPIFFS.

SPIFFS has lower performance than LittleFS and it may set to obsolete or not supported by the Core SDK developer (not sure ESP8266 or ESP32 core) and as I post above some user of this library request to use LittleFS by default in [this discussion](https://github.com/mobizt/ESP-Mail-Client/discussions/194) and I accept his request.

Actually, SPIFFS writing bug (not reading as I post above) was found in ESP8266 platform, see [this issue](https://github.com/esp8266/Arduino/issues/8372).

mobizt, 2023-01-09:
I test with this old LitteFS upload plugin from the same developer as plugin I post above, and it works and I'm using it.
https://github.com/lorol/arduino-esp32littlefs-plugin

As you will see on his repo, he recommends to use the universal combined plugin instead.

NDC-SC, 2023-01-09:
I have installed the universal combined plugin and both the image file loaded in SPIFFS and the image data file in PROGMEM were successfully included in an email.
LittleFS and the universal combined plugin works well.

NDC-SC, 2023-01-09:
One further question, what is the difference between addAttachement, addInlineImage and addParallelAttachment ?
Thanks

mobizt, 2023-01-09:
All are related to multipart content type, see rfc1341 section 7

Attachment is normal file attachment.

Inline is for image only that image will be display as part of message and also can set the CID (Content ID) in each message header field for use in the image tag of html message like this `<img src="cid:my_image_cid"/>`

Parallel is for media (sound and video) that all showed or played simultaneously (mail client or reader may not support and show or play this content serially).

NDC-SC, 2023-01-11:
Thanks for your helpful comments and for answering questions.