mobizt/ESP-Mail-Client #219: ESP_mail_client not compatible with latest  esp8266 / Arduino build (version 3.1.1)

gitmikrod, 2023-01-19:
**Build tool used:**

- [X ] Arudino
- [ X] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [x ] ESP8266-based
- [ ] ESP32-based

**Other Libraries That are used:**

**Description of problem:**

email client causes a WDT error when trying to connect to NTP server when using latest esp8266 / arduino build (https://github.com/esp8266/Arduino). 

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

The reason is probably that in the new release there was a change to delay() function:
From release notes (https://github.com/esp8266/Arduino/releases):
  Non-interruptible delay(), wait for the full duration

Easiest way to reproduce error is to use any of the esp_mail_client examples with 3.1.1 esp8266 build.

I've confirmed bug on both Arduino IDE and PlatformIO (using VSCode) (both updated to latest versions). I'm on Windows 10

I used two different boards (WEMOS D1 mini and WEMOS D1 mini Pro) and both show the same problem.

Note, downgrading to version 3.1.0 of the arduino board made everything work (i.e., I managed to send emails).

mobizt, 2023-01-23:
The [library update](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v2.7.3) is available. You can download now.

closed 2023-01-23 by mobizt

gitmikrod, 2023-01-23:
Wow, that's quick! Thank you so much for fixing this and, most of al,l thank you so very much for all the effort and skill you put into the fantastic ESP_mail library!