mobizt/ESP-Mail-Client #221: .pio/libdeps/esp32dev/ESP Mail Client/src/extras/MB_Time.h:373:9: error: ambiguous overload for 'operator+=' (operand types are 'MB_String' and 'int')        s += 0;        ~~^~~~

DanielW0830, 2023-02-01:
**Build tool used:**

- [ ] Arudino
- [X ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [X ] ESP32-based

**Other Libraries That are used:**

when I try to use the latest version of the lib I get arror in MB_String
Older versions work, (I think 1.6.2, but not the newer ones)

**Share code snippet to reproduce the issue:**
```yaml

In file included from .pio/libdeps/esp32dev/ESP Mail Client/src/ESP_Mail_Client.h:43,
                 from src/main.cpp:50:
.pio/libdeps/esp32dev/ESP Mail Client/src/extras/MB_Time.h:373:9: error: ambiguous overload for 'operator+=' (operand types are 'MB_String' and 'int')
       s += 0;
       ~~^~~~
```

**Additional information and things you've tried:**

DanielW0830, 2023-02-01:
It's prob some version of the core I'm using....    dunno....

PACKAGES:
 - framework-arduinoespressif32 @ 3.20006.221224 (2.0.6)
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- RemoteDebug @ 3.0.5
|   |-- AsyncElegantOTA @ 2.2.7
|   |-- WiFi @ 2.0.0
|   |-- SPI @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- InterpolationLib @ 1.0.0
|-- twilio-esp32-client @ 0.1.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- AsyncElegantOTA @ 2.2.7
|-- ElegantOTA @ 2.2.9
|   |-- Update @ 2.0.0
|   |-- WebServer @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- esp32_https_server @ 1.0.0
|-- ESP Mail Client @ 2.7.6
|   |-- SPI @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- Ethernet @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|   |-- LittleFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SD @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |   |-- SPI @ 2.0.0
|   |-- SD_MMC @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- SPIFFS @ 2.0.0
|   |   |-- FS @ 2.0.0
|   |-- FS @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- ESP32 Async UDP @ 2.0.0
|-- Preferences @ 2.0.0
|-- ArduinoOTA @ 2.0.0
|   |-- Update @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ESPmDNS @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- HTTPClient @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- WiFiClientSecure @ 2.0.0
|   |   |-- WiFi @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- Ticker @ 2.0.0
|-- Update @ 2.0.0
|-- WiFi @ 2.0.0
|-- Wire @ 2.0.0

mobizt, 2023-02-01:
The library developed in VScode and PlatformIO and everything works normally in PlatformIO and Arduino IDE.

This error is not related to library unless your PlatformIO installation and update issue.

Please read this and follow when you updated your PlatformIO from older 2.0.0 to the newer version.
https://github.com/mobizt/ESP-Mail-Client#library-instalation

closed 2023-02-01 by mobizt

mobizt, 2023-02-01:
In summary, you need to clear the old installation of this library that may be installed globally in your PC temp directory since PlatformIO older version.

In PlatformIO newer version, the library will be installed locally to the project folder.

When the compiler (linker) searches for the header file and see multiples files with the same name from both local storage and global storage, it may choose the wrong older one to compile and causes the compilation error. 

MB_String.h is the wrong file links in this case as this old version file remains in your PlatformIO global temp folder.