mobizt/ESP-Mail-Client #223: esp32 reboot

workpage2, 2023-02-07:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
Hello.
I tried your example code and it worked great. However, when I pasted it into a real project, esp32 crashes on the line: `smtp.connect(&session)`
I pasted the lines from the example without modification.
I have a very complex project. I use web, ws, ntp, mqtt, modbus tcp, snmp, http ssl connection (not always). Help me understand where the problem is.

`Connecting to SMTP server...`
`> C: Wait for NTP server time synching`
`abort() was called at PC 0x40086b13 on core 0`

`ELF file SHA256: 0000000000000000`

`Backtrace: 0x40088a48:0x3ffbf550 0x40088cc5:0x3ffbf570 0x40086b13:0x3ffbf590 0x40086b39:0x3ffbf5b0 0x40086c8d:0x3ffbf5e0 0x40147a3b:0x3ffbf600 0x40140c89:0x3ffbf8c0 0x40140b79:0x3ffbf910 0x4008acfd:0x3ffbf940 0x40089961:0x3ffbf970 0x40084fb0:0x3ffbf990 0x400842ea:0x3ffbf9c0 0x40087075:0x3ffbf9e0 0x401c8e7b:0x3ffbd3c0 0x401b265b:0x3ffbd3e0 0x4008b485:0x3ffbd400 0x40089cd6:0x3ffbd420
`

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**

workpage2, 2023-02-07:
I tried sending email immediately after wifi initialization, but it failed.

mobizt, 2023-02-07:
I can't help in case of third-party libraries issue.

There are many possibilities of reboot from wdt reset e.g., memory leaks, use of deleted object, low free heap and stack overflown and the blocking code that cause wdt counterter overflown.

You should debug your code your self.

closed 2023-02-07 by mobizt

workpage2, 2023-02-07:
The error might look like this:

`Wifi started.`
`Connecting to SMTP server...`
`> C: Wait for NTP server time synching`
`Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)`
`Debug exception reason: Stack canary watchpoint triggered (loopTask) `
`Core 1 register dump:`
`PC      : 0x4008c93b  PS      : 0x00060c36  A0      : 0x4008c336  A1      : 0x3ffafff0  `
`A2      : 0x3ffb0060  A3      : 0x3ff6153c  A4      : 0x00060c20  A5      : 0x3ffb0810  `
`A6      : 0x00002aaa  A7      : 0x3ffb0180  A8      : 0x801845c9  A9      : 0x4008c424  `
`A10     : 0x00000042  A11     : 0x00000004  A12     : 0x40087010  A13     : 0x3ffdcb88  `
`A14     : 0x00000aaa  A15     : 0x3ffb0160  SAR     : 0x00000012  EXCCAUSE: 0x00000001  `
`EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  `

`ELF file SHA256: 0000000000000000`

`Backtrace: 0x4008c93b:0x3ffafff0 0x4008c333:0x00000000`

`Rebooting...`

mobizt, 2023-02-07:
You should debug your code first nobody can use that information you post to debug because the elf file is not available for anyone.

Decode the stack trace by esp exception decoder from the link of my above reply and see how the code stop at that line.

workpage2, 2023-02-07:
Decoding stack results
0x401857ee: mbedtls_mpi_exp_mod at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/bignum.c line 2115
0x40178960: mbedtls_rsa_public at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 784
0x401799f9: mbedtls_rsa_rsassa_pkcs1_v15_verify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 2401
0x40179a9f: mbedtls_rsa_pkcs1_verify at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/rsa.c line 2458
0x40188399: rsa_verify_wrap at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c line 115
0x401772cb: mbedtls_pk_verify_restartable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/pk.c line 308
0x40190370: ssl_parse_server_key_exchange at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c line 2904
0x40190ac3: mbedtls_ssl_handshake_client_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_cli.c line 3840
0x4017ae79: mbedtls_ssl_handshake_step at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c line 8117
0x4017aeb9: mbedtls_ssl_handshake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c line 8141
0x4012be76: ESP32_SSL_Client::connect_ssl(ESP32_SSL_Client::ssl_context_t*, char const*, char const*, char const*, char const*, char const*, char const*, bool) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src\wcs\esp32\ESP32_SSL_Client.cpp line 365
0x4012c6bf: ESP32_WCS::connect(char const*, unsigned short, char const*, char const*, char const*) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src\wcs\esp32\ESP32_WCS.cpp line 219
0x4012c755: ESP32_WCS::connect(char const*, unsigned short) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src\wcs\esp32\ESP32_WCS.cpp line 146
0x4012c1db: ESP32_TCP_Client::connect(bool, bool) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src\wcs\esp32\ESP32_TCP_Client.cpp line 306
0x4012964e: SMTPSession::connect(bool&) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_SMTP.h line 3892
0x40129796: SMTPSession::handleConnection(esp_mail_session_config_t*, bool&) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_SMTP.h line 3766
0x401297c9: SMTPSession::connect(esp_mail_session_config_t*) at C:\Users\Workshop\Documents\Arduino\libraries\ESP-Mail-Client-master\src/ESP_Mail_SMTP.h line 3717
0x400e9a2a: setup() at Z:\work\esp32_ver3\ver1.0\esp32_fs/esp32_fs.ino line 4225
0x401311e6: loopTask(void*) at C:\Users\Workshop\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 18
0x40089cd6: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

workpage2, 2023-02-07:
any ideas.

mobizt, 2023-02-07:
Your device is out of memory which it fails when mbedTLS trying to allocate the memory.

You need to have 100K+ free heap to use with this email library and other libraries that connect to server via secure protocols (https) which required approx. 50k for mbedTLS engine to work during TLS handshake.

You should reduce your memory usage used by other libraries and your own code.

mobizt, 2023-02-07:
You should enable debug options in Arduino IDE for WiFi, SSL and OOM (out of memory) which you will see the information for better understand.

workpage2, 2023-02-07:
I don't use attachments in emails. I am planning to use the email library for sending only. Can I reduce the amount of memory used due to this?

workpage2, 2023-02-07:
`Serial.println(ESP.getFreeHeap());`
I usually use this article before sending email.
Available 256631.

mobizt, 2023-02-07:
It's not true. The heap was reduced by dynamic allocation that you should check your device free heap at anywhere in your code to know its condition.

It should check before these two functions call.
https://github.com/mobizt/ESP-Mail-Client/blob/ac39ffe3f18da4fb5a5be4d778f5ca4539b26c87/examples/SMTP/Send_Text/Send_Text.ino#L264-L269

If you use other library that run FreeRTOS task, the memory was much reduced.

Again, you should enable debug options in IDE which make you better understand the situation where and how it fails to allocate memory.

mobizt, 2023-02-07:
As you see from the issue template what information you should provide when open the issue.

I can't verify or debug your code especially with other third libraries project.

The code that you can post here should keep small without third party libraries that work on WiFi too e.g., MQTT, web server, SSL client etc.

workpage2, 2023-02-07:
I will try to create this situation in a small file and find out which library is causing the problem. I'll come back.

workpage2, 2023-02-07:
I figured out the problem. After including the email library, some parts of the code stopped compiling.
I fixed it, but the result is different. Apparently this caused an overflow.

This stopped working:
`String a="o";`
`char b="k";`
`a=a+b;`

I fixed it this way:
`String a="o";`
`char b="k";`
`a=a+(String)b;`

The result is not "ok", but "o107".

This option works:
`String a="o";`
`String temp;`
`char b="k";`
`temp=b;`
`a=a+temp;`

I don't know how this is related to the email library, but this solved the problem. Thanks to all.

mobizt, 2023-02-07:
It not's related to the library but the syntax you use causes unexpected result on run time and leads to exception.

`char b="k"; `

Which you can't assign one byte  8 bit signed interger (int8_t or char) to the const char* or pointer to the char array.

it should be

`char b = 'k';` // 'k' assign ascii code (1 byte) of k

mobizt, 2023-02-07:
It required the basic Arduino C/C++ programming knowledge to work with string.

mobizt, 2023-02-07:
I will delete this issue as because of user programming error which is not related to the library.