mobizt/ESP-Mail-Client #228: Client issuing a STARTTLS command to a server that does not support TLS

fsnuffer, 2023-02-16:
**Build tool used:**

- [ X] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32):**

- [ ] ESP8266-based
- [ X] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
I send two e-mails back to back.  One to a server that does not support TLS and the second to one that does.  The first round of e-mails go through without a problem.  When I try to send the same e-mails again, when the client connects to the Non-TLS server the client issues a STARTTLS even though the HELO header does not indicate indicate it supports TLS.  Below are snippets of the debug logs.  

**Share code snippet to reproduce the issue:**
```yaml
SMTPSession smtp;  is a global variable.
ESP_Mail_Session session; is defined within the function so it should be deleted when the function completes.

```

**Additional information and things you've tried:**
Here is the failed connect.  The server did not include 250-STARTTLS in its reply.

22:04:37.223 -> Sending greeting response...
22:04:37.223 -> > C: Send SMTP command, HELO
22:04:37.291 -> < S: 250-mscle5.******** Hello [10.1.1.125]
22:04:37.291 -> < S: 250-SIZE
22:04:37.291 -> < S: 250-DSN
22:04:37.291 -> < S: 250-ENHANCEDSTATUSCODES
22:04:37.291 -> < S: 250-AUTH
22:04:37.291 -> < S: 250-8BITMIME
22:04:37.291 -> < S: 250 NOOP
22:04:37.291 -> 
22:04:37.291 -> Send command, STARTTLS
22:04:37.291 -> > C: Send STARTTLS command
22:04:37.327 -> < S: 502 command not implemented or not allowed to be used by 10.1.1.219
22:04:37.327 -> Error, SMTP server greeting failed
22:04:37.327 -> > E: SMTP server greeting failed
22:04:37.327 -> > C: cleaning SSL connection

[SMTP Debug Logs.txt](https://github.com/mobizt/ESP-Mail-Client/files/10751866/SMTP.Debug.Logs.txt)

mobizt, 2023-02-16:
This is not library issue.

Library is well managed and will not send `STARTTLS` command without server supports by checking server capability from greeting response or user enable it trough the session object `session.secure.startTLS =true;`

Which in this case, no STARTTLS capability found from server ressonse or your SMTP server assigned was not support STARTTLS.

```yaml
22:04:37.291 -> < S: 250-mscle5.******** Hello [10.1.1.125]
22:04:37.291 -> < S: 250-SIZE
22:04:37.291 -> < S: 250-DSN
22:04:37.291 -> < S: 250-ENHANCEDSTATUSCODES
22:04:37.291 -> < S: 250-AUTH
22:04:37.291 -> < S: 250-8BITMIME
22:04:37.291 -> < S: 250 NOOP
```

I don't see your code how you configure the session, unless you should remove this line from your code.
`session.secure.startTLS =true;`

And you are assign private IP to the  option `session.login.user_domain` which is not valid.

https://github.com/mobizt/ESP-Mail-Client/blob/7a8ce48dd8152ea1dd2c3e4513e24be07921e960/examples/SMTP/Send_Text/Send_Text.ino#L139-L147

For the proper usages, please follow the examples that library provided in the current version you use.
Examples may update because of library updates.

mobizt, 2023-02-16:
I think now I understand what you mean after reading the debug logs when `SMTPSession` was used for different servers and some internal config was not clear.

I will investigate and inform you update.

mobizt, 2023-02-16:
Thanks for reporting this issue.

The new version [2.7.11](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v2.7.11) is available with bugs fixed and more functions added.

Please try.

closed 2023-02-16 by mobizt

fsnuffer, 2023-02-16:
Verified fixed.  Thanks.