mobizt/ESP-Mail-Client #247: NTP Server Timed Out

QnBarb, 2023-03-24:
**Build tool used:**

- [x] Arudino
- [x] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

**Description of problem:**
This sketch, to send email via SMTP (Gmail) was working the day before, but now this is the error I get. Please advise. I'm new to this, so if this is the wrong place to ask, please point me in the right direction. I appreciate your help.

#### Connecting to SMTP server...
> C: ESP Mail Client v3.1.3
> C: wait for NTP server time synching
#### Error, NTP server time synching timed out
! E: NTP server time synching timed out

**Share code snippet to reproduce the issue:**
  #include <Arduino.h>
#if defined(ESP32)
  #include <WiFi.h>
#endif
#include <ESP_Mail_Client.h>
#define WIFI_SSID "****"
#define WIFI_PASSWORD "****"
#define SMTP_HOST "smtp.gmail.com"
#define SMTP_PORT 465
#define AUTHOR_EMAIL "****"
#define AUTHOR_PASSWORD "****"
#define RECIPIENT_EMAIL "****"
SMTPSession smtp;
SMTP_Message message;
ESP_Mail_Session session;
String alertMsg;
void smtpCallback(SMTP_Status status);

void setup() {
  Serial.begin(115200);
  Wire.begin(I2C_DEV_ADDR); //Initiate I2C protocol for slave mode, notice the address 0x55
  Wire.onReceive(receiveEvent); //Receiving data from I2C bus
  Serial.println("Connecting to internet");
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  while(WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(200);
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP Address: ");
  Serial.println(WiFi.localIP());
  Serial.println();
  smtp.debug(1);
  smtp.callback(smtpCallback);
  ESP_Mail_Session session;
  session.server.host_name = SMTP_HOST;
  session.server.port = SMTP_PORT;
  session.login.email = AUTHOR_EMAIL;
  session.login.password = AUTHOR_PASSWORD;
  session.login.user_domain = "";
  SMTP_Message message;
  message.sender.name = "****";
  message.sender.email = AUTHOR_EMAIL;
  message.subject = "****";
  message.addRecipient("****", RECIPIENT_EMAIL);
  message.text.content = alertMsg.c_str();
  message.text.charSet = "us-ascii";
  message.text.transfer_encoding = Content_Transfer_Encoding::enc_7bit;
  message.priority = esp_mail_smtp_priority::esp_mail_smtp_priority_low;
  message.response.notify = esp_mail_smtp_notify_success | esp_mail_smtp_notify_failure | esp_mail_smtp_notify_delay;
  if(!smtp.connect(&session))
    return;
  if(!MailClient.sendMail(&smtp, &message))
    Serial.println("Error sending email, " + smtp.errorReason());
}

void loop() {}

void smtpCallback(SMTP_Status status){
  Serial.println(status.info());
  if (status.success()){
    Serial.println("----------------");
    ESP_MAIL_PRINTF("Message sent success: %d\n", status.completedCount());
    ESP_MAIL_PRINTF("Message sent failled: %d\n", status.failedCount());
    Serial.println("----------------\n");
    struct tm dt;
    for (size_t i = 0; i < smtp.sendingResult.size(); i++){
      /* Get the result item */
      SMTP_Result result = smtp.sendingResult.getItem(i);
      time_t ts = (time_t)result.timestamp;
      localtime_r(&ts, &dt);

      ESP_MAIL_PRINTF("Message No: %d\n", i + 1);
      ESP_MAIL_PRINTF("Status: %s\n", result.completed ? "success" : "failed");
      ESP_MAIL_PRINTF("Date/Time: %d/%d/%d %d:%d:%d\n", dt.tm_year + 1900, dt.tm_mon + 1, dt.tm_mday, dt.tm_hour, dt.tm_min, dt.tm_sec);
      ESP_MAIL_PRINTF("Recipient: %s\n", result.recipients.c_str());
      ESP_MAIL_PRINTF("Subject: %s\n", result.subject.c_str());
    }
    Serial.println("----------------\n");
  }
}

PASTE .cpp / .ino code here

```

**Additional information and things you've tried:**
I've tried different board, created new email address, changed the SMTP port, reinstalled libraries, reboot, using PlatformIO and Arduino IDE.

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

QnBarb, 2023-03-24:
The sketch for sending email via SMTP (Gmail) was working before yesterday, but now I'm getting error of 'NTP Server Timed Out..' Do I need to update the library? Thank you so much in advance.

mobizt, 2023-03-24:
It's not library issue. 
The library works fine. 
The problem is your internet. Device can't connect to NTP server.

closed 2023-03-24 by mobizt

QnBarb, 2023-03-24:
Thank you. Have a great day.

github-coder-lab, 2023-03-27:
I have the same problem. With V2.7.11 it works with V3.1.3 there is always an NTP timeout

mobizt, 2023-03-27:
@github-coder-lab 

If you install the library only in Library Manager, you can update library using Library Manager as usual. 

If you install library by using zip file, the problem will be happened because the folder will be created using the zip filename while when library install or update via Library Manager, the library will be installed under the folder that defined by `name` filed of `library.properties` file.

When there are old version library and new version library under the libraries folder the problem will occur at compile time or runtime which old source file are compiled together of new source file which is only the problem when you are using Arduino IDE.

mobizt, 2023-03-27:
@QnBarb 

I don't what you mean about trigger.

To keep session open, see [this example](https://github.com/mobizt/ESP-Mail-Client/blob/master/examples/SMTP/Send_Reuse_Session/Send_Reuse_Session.ino).