mobizt/ESP-Mail-Client #249: random but repeatedly problems/errors with smtp blocking eMail functionality

ChristofSchmid, 2023-03-29:
**Build tool used:**

- [x] Arudino
- [ ] Platformio (Visual Studio Code plugin)
- [ ] Platformio CLI
- [ ] Other

**Board used (ESP32/ESP8266/Arudino):**

- [ ] ESP8266-based
- [x] ESP32-based

**Other Libraries That are used:**

#include <esp_task_wdt.h>  
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
#include <ESP_Mail_Client.h>
#include <CircularBuffer.h>

#include <time.h>
#include <RadioLib.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

//Add filesystem
#include "FS.h"
#include <LITTLEFS.h>

**Description of problem:**
First of all, congrats and sthanks to this comprehensive lib. I am very sorry to bug you here, but I cannot solve the problems encouterd myself. Maybe you coild guide me in the right direction.

I have developed an irrigation system and use ESP-Mail-Client to send critical information from the soil sensors attached.
I started with v2.7 and moved on to the actual build.
1. Issue
when I close the smtp session after MailClient.sendMail using the param 3 flag true, one mail can be sent. Then I receive an error message "> C: cleaning SSL connection". The next use of .sendMail is successful and so on.
2. Issue (version of logs and code below)
keeping the smtp open works well for a limited time. Then I usually receive an error "! E: SSL - Memory allocation failed"
In the logs you can see that after 2+ hours, the .sendMail does nothing anymore 

**Share code snippet to reproduce the issue:**
```yaml
PASTE .cpp / .ino code here

***** Sketch initial code ******

/*********
  Rui Santos
  Complete project details at https://RandomNerdTutorials.com/esp32-websocket-server-arduino/
  The above copyright notice and this permission notice shall be included in all
  copies or substantial portions of the Software.
*********/

// Import required libraries
#include <esp_task_wdt.h>  
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>
#include <ESP_Mail_Client.h>
#include <CircularBuffer.h>

#include <time.h>
#include <RadioLib.h>
#include <Adafruit_GFX.h>
#include <Adafruit_SSD1306.h>

//Add filesystem
[... 1327 lines cut ...]
#### Current: 115572, Min: 114524, Max: 197760, Diff_1: -82188, Diff_58: 1048
#### Message sent failed: 2
#### Heap Info
#### Current: 117904, Min: 114524, Max: 197760, Diff_1: -79856, Diff_59: 2332
#### -------------------------
#### Heap Info
#### Current: 117896, Min: 114524, Max: 197760, Diff_1: -79864, Diff_60: -8
#### Status: failed
#### Heap Info
#### Current: 117916, Min: 114524, Max: 197760, Diff_1: -79844, Diff_61: 20
#### Date/Time: Wed Mar 29 19:14:38 202
#### Heap Info
#### Current: 117872, Min: 114524, Max: 197760, Diff_1: -79888, Diff_62: -44
#### Recipient: christof.schmid@aaa-immobilien.ch
#### Heap Info
#### Current: 117856, Min: 114524, Max: 197760, Diff_1: -79904, Diff_63: -16
#### Subject: Irrigation notification
#### Heap Info
#### Current: 117872, Min: 114524, Max: 197760, Diff_1: -79888, Diff_64: 16
Error sending Email, connection closed
Sending Humidity Message failed

**** and so on until ESP-Mail-Client does not even show a single Message on serial...

<!-- LEAVE THIS LINE AS-IS AND DON'T DELETE IT, OTHERWISE THE ISSUE WILL BE CLOSED AUTOMATICALLY. -->

mobizt, 2023-03-30:
Where is "**** next attempt ***" printing on your code?

mobizt, 2023-03-30:
Now the library update [v3.1.5](https://github.com/mobizt/ESP-Mail-Client/releases/tag/v3.1.5) with issues fixed is available.
Please update.

closed 2023-03-30 by mobizt

ChristofSchmid, 2023-03-30:
Hi Suwatchai

Thanks a lot for the fix.
I had the new build running all day. 

When leaving the smtp connection open, after a while there is a disconnect and reconnect which I expected. This caused 2 of 40 messages not sent. ( I guess google disconnected when the send process was in progress or so)

When using the MailClient.sendMail with the param 3 flag true (close option) the lib works flawless, no message skipped.

----------------
Message sent success: 44
Message send failed: 0
----------------

Thanks again
Christof